paroj/xpad #1: only expose connected devices

paroj, 2015-06-30:
only expose connected devices with Xbox360 Wireless Receiver. For This we need to figure out a good way [how to determine the xpad_nr](http://www.spinics.net/lists/linux-input/msg29493.html).
For status see *connected_only* branch.

computerquip, 2015-07-01:
The bug is caused by discontinuing listening to the wireless adapter, which is done here: https://github.com/paroj/xpad/blob/connected_only/xpad.c#L687

Even if a device isn't present physically connected to the adapter, this is how the adapter tells you whether a controller is connected or not. You must always be listening.

computerquip, 2015-07-01:
Also, the xpad_nr can be determined by holding a bit array held in global state. Each bit represents a connected controller. When a new controller connects, iterate through the bit array until you see one that isn't set. Take the position of that bit, and use the result of position modulo 4 (4 being the static number of LEDs). Then set the bit to indicate the controller is using that position. When the controller is disconnected, be sure and unset the bit. 

You can probably just dynamically add onto the bit array... or just hold a 32-bit integer and say the limit is 32 controllers.

computerquip, 2015-07-01:
The adapter has a hard limit for 4 controllers. On top of this, there are only 4 LED positions. The original XUSB driver also only supported 4 controllers. I remember why I only supported 4 controllers with Xpad5

paroj, 2015-07-01:
yeah.. I probably should just go with `intf->cur_altsetting->desc.bInterfaceNumber` for enumerating the wireless controllers.

But this is the usecase I am thinking about and which is covered by the current atomic counter approach:
1. attach a wired x360 pad (LED1)
2. connect a wireless x360 pad (LED2)
the 5th controller is identified by LED1 again

However the counter just keeps incrementing after suspend/ resume. (this is the actual issue with it)

But this is unrelated to only exposing connected devices and I should treat it as a different issue.

paroj, 2015-07-01:
thanks for the hint regarding listening to the adapter. will try it as soon as I have some time

computerquip, 2015-07-01:
Really, you can just hold some global state that can track when controllers are connected. 

Using the interface number would break when used in combination with wired controllers.

paroj, 2015-07-03:
works :+1:

closed 2015-07-03 by paroj