paroj/xpad #2: do not drop force feedback packets

paroj, 2015-06-30:
while the `URB ffff88020d047d80 submitted while active` issue should be fixed in master, FFB packets might still be dropped. Consider [this patch for SteamOS](https://github.com/ValveSoftware/steamos_kernel/pull/14).

Basaltq, 2015-07-26:
Related? http://steamcommunity.com/app/233130/discussions/0/611702631210587760/#c611702631214382829

paroj, 2015-07-26:
yes, related. but the linked patch is a better solution.

paroj, 2015-09-05:
according to dtor one should not use a queue. makes sense.
http://www.spinics.net/lists/linux-input/msg40708.html

paroj, 2015-09-05:
and this is a test for reproducing the race:
```
while true; do
        for i in $(seq 0 5); do
                echo $i > /sys/class/leds/xpad0/subsystem/xpad0/brightness
        done
done
```

paroj, 2015-09-27:
re-submitting logic is now in master: 906c1f1aeef2c2969f1e5c42ea954e9572cb26c4.
verified that LED commands get correctly re-submitted.
Still need a test for FF commands. Otherwise done.

paroj, 2015-10-01:
actually this particular problem is fixed. the next one is in #10

closed 2015-10-01 by paroj