paroj/xpad #5: x360: need stable and consistent enumeration of pads

paroj, 2015-07-01:
split off from#1: right now there is only a atomic counter which has 2 issues:
1. is not stable: keeps incrementing after suspend/ resume so the led nr changes
2. is not consistent: there is no easy way to find out which led belongs to which input device

however it does handle the following use case:
1. attach a wired x360 pad (LED1)
2. connect a wireless x360 pad (LED2)
the 5th controller is identified by LED1 again

Need to ping [the guys mentioned here about it](http://www.spinics.net/lists/linux-input/msg29493.html).

paroj, 2015-07-03:
fix first part by also decrementing the counter.

computerquip, 2015-07-07:
So, with a global increment, we have the problem of filling in slots. Take this scenario:

1) Controller A connects. LED 1 lights up. 
2) Controller B connects. LED 2 lights up. 
3) Controller A disconnects. LED 1 turns off. 
4) Controller A reconnects. LED 3 turns on.

The behavior we want is:

1) Controller A connects. LED 1 lights up. 
2) Controller B connects. LED 2 lights up. 
3) Controller A disconnects. LED 1 turns off. 
4) Controller A reconnects. LED 1 turns on. 
5) Controller C connects. LED 3 turns on. 
6) Controller B disconnects. Controller C disconnects. Controller B reconnects. LED 2 lights up. 

I found this easier to solve in xpad5 than I originally thought it would be. Simply create a global bit array, where each bit represents a possible controller. 0 is inactive, 1 is active. Then iterate through the bit array each time a controller reconnects to find an open position. 

(In the case of Xpad5, I hold state on each possible controller in the xusb module. This is analogous to the bit array I'm speaking of here.)

paroj, 2015-07-07:
you are right, it is quite broken right now. 

what I dont like about a bit array, is that you artifically limit the number of possible pads. Though 32 pads should be enough for the start and I guess this is easiest to implement. (modulo race-free bit counting)

Then again, there was talk about [using the USB interface number](http://www.spinics.net/lists/linux-input/msg29527.html) - at least for a constent naming of the LEDs. However I am not sure how to access/ use this..

paroj, 2015-07-07:
implemented the bitmask approach: https://github.com/paroj/xpad/commit/bd240e4661550d54b0a47dca2a707c141c6315f7

computerquip, 2015-07-08:
controller->num_controller = (interface->cur_altsetting->desc.bInterfaceNumber + 1) / 2;

This is what I did in the original driver which uses the interface number of the wireless adapter.

I think I stopped using this because it doesn't work in conjunction with wired controllers or other wireless adapters.

paroj, 2015-10-01:
split out led part to #11

paroj, 2015-10-01:
[dtor suggested using idr/ ida](http://www.spinics.net/lists/linux-input/msg40240.html) instead of custom bitmask appraoch

paroj, 2015-10-01:
fixed in master

closed 2015-10-01 by paroj