paroj/xpad #12: New version of  Logic3 Rock Candy Gamepad not supported

Fenisu, 2015-10-02:
Hi,
Appearently the Rock Candy Gamepad for XBOX 360 is supported. They released a new version, "for XBOX One". I tried to add it to the list, compile it and load it, plug in the gamepad and now it gets recognised of course but it does not work.

The message is:
> input: Rock Candy Gamepad for Xbox One as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/input/input139
> xpad 1-3:1.0: xpad_irq_in - urb shutting down with status: -2

This is the lsusb of the device: http://pastebin.com/pp354U2x

If I can do anything to help to get this working, please let me know.

paroj, 2015-10-03:
it would rather help to see how you changed the code

Fenisu, 2015-10-03:
I just added it to the list: 
http://pastebin.com/WEM6JHE5

paroj, 2015-10-04:
the error message means that it could not properly initialize the controller.
you could try reverting this change, as it modified the initialization for xpadone: https://github.com/paroj/xpad/commit/f3cf5f3981be72f541aaddc6b9e81781e88c21f7

Fenisu, 2015-10-04:
I tried with no luck. Exactly same behaviour. Does this mean that the controller works differently? or that there is something missing on the xbox one implementation? (I tried the controller on Windows 10, just in case, and it gets recognized and lights up)

paroj, 2015-10-04:
yes, probably the xpad initialization packet is different to what window sends. (incomplete and/ or different values)
Although the official gamepad works, the Rock Candy Gamepad seems to be more picky.

Unfortunately the only way to find out is to use a USB sniffer on Windows and look in the command stream for differences..

Fenisu, 2015-10-05:
Well, I made some usb sniffing on Windows with freeusbanalyzer and this is what I got from just connecting the device:
http://pastebin.com/eZsfjHpg

First time I am doing this kind of stuff, I will try to decipher it. Too bad I don't own an XBOX 360 controller to compare and see what packets are those that are important. Any suggestions would be appreciated.

paroj, 2015-10-15:
working initialization sequence https://github.com/Fenisu/xbox/commit/6c4842e84ffbee3d2f5956ee5f7f4d7eae13b50c

paroj, 2015-12-24:
I have implemented the extended initialization sequence in the [xone2015 branch](https://github.com/paroj/xpad/tree/xone2015).

Fenisu, 2015-12-25:
Yes, but it will have to be in ~10 days.
Maybe interested in:
https://github.com/ValveSoftware/steamos_kernel/pull/36

The changes in last steamos_xpad plus that diff makes my controller work.

paroj, 2015-12-25:
this is strange as the steamos xpad still uses only one packet to initialize the controller which did not work for you here. Also why do you set `MAP_TRIGGERS_TO_BUTTONS`?

Fenisu, 2015-12-26:
Well, I'm rather new to all this gamepads stuff and after recognizing the controller by the kernel and making available the event/js, I use steam to try it. I realized that while the controller is supported as is by one game,  most games don't recognize it at all for some reason, so I have to map the controller using xboxdrv as if it were an xbox360 controller, and then I failed to get those triggers working as apparently they are not buttons but throttles.

paroj, 2015-12-26:
I see - the problem are actually the games that were written with only the xbox360 controller in mind. However the kernel driver is the wrong place to remap buttons; if the devs decide to fix xbox one controller support, all xbox one controller should behave the same.

If the games are using SDL2, you should rather use `SDL_GAMECONTROLLERCONFIG`.

Also I think that your Rock Candy controller issue is fixed in master now, however I let the xone2015 branch there so you can test both.

Fenisu, 2015-12-27:
I see, well, taht makes much more sense, I didnät not even know that those were actually throttles on xbox one. I will change that diff for steamos_xpad.

Good to know, as soon as I get back home I try both branches. Thanks for your hard work.

Fenisu, 2016-01-11:
I just tried many new xpad modules. Both master and xone2015 fail to work properly with my controller. This is what happens:
```
[102850.672064] usb 2-2: new full-speed USB device number 19 using xhci_hcd
[102850.847650] xpad 2-2:1.0: xpad_prepare_next_out_packet - found pending output packet 2: 0x01
[102850.847720] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input53
[102850.965757] xpad 2-2:1.0: xpad_irq_in - urb shutting down with status: -2
[102851.189193] xpad 2-2:1.0: xpad_irq_in - urb shutting down with status: -2
[102851.329311] xpad 2-2:1.0: xpad_irq_in - urb shutting down with status: -2
[102867.369435] xpad 2-2:1.0: xpad_irq_in - urb shutting down with status: -2
```
Then I tried the new xpad from Valve. Valve pushed some changes for their kernel on SteamOS brewmaster beta. With those changes, the controller is fully functional: https://github.com/ValveSoftware/steamos_kernel/commit/76e4b04b93b40db186d0c4bbbd1824f5d98c76a9

Thank you for your time.

dantob, 2016-02-02:
pull this commit, should be all that's missing now the xone init is sorted? https://github.com/ValveSoftware/steamos_kernel/commit/96053cb77390d18cc76cebead210481ea173bee5

closed 2022-10-22 by Fenisu