paroj/xpad #13: Afterglow Prismatic Wired Xbox One Controller Not Working + Fix

aronschatz, 2015-10-22:
Long story: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1507318

Short story: I'm reviewing a new Afterglow Prismatic Controller for Xbox One from PDP and it didn't work out of the box due to the vendor and product ids not being in the xpad.c file. I've added them in and it works fine.

Here's the diff:

128d127
<       { 0x0e6f, 0x0139, "Afterglow Prismatic Wired Controller", 0, XTYPE_XBOXONE },
303d301
<       XPAD_XBOXONE_VENDOR(0x0e6f),            /* 0x0e6f X-Box One controllers */

paroj, 2015-10-22:
can you please make a pull request for this?

aronschatz, 2015-10-22:
Sure. The second thing I'm noticing is that Steam hates this controller for some reason and I can't seem to figure out how to map the buttons correctly. I believe that's a steam issue, though. jstest works fine.

Or is there something else I should modify before doing a pull request to try to see if it can be fixed in the xpad driver?

aronschatz, 2015-11-18:
Took me awhile, but the pull request has been made. Steam works fine after being able to once again map buttons.

closed 2015-12-19 by paroj