paroj/xpad #14: XBox One Wireless Controller Support

Queuecumber, 2015-10-31:
I figured I'd open an issue for this, I have the wireless adapter "for windows 10", I haven't done much debugging with the xpad driver but I figured support for this is one of your goals. I can help debug this and/or provide any information you might need.

wjbuys, 2015-11-01:
I also just got one, and would be willing to help with debugging.

In case it's useful, [here](https://dl.dropboxusercontent.com/u/153816/xbox-one-wireless-adapter.pcapng.gz)'s a Wireshark dump of the USB traffic when plugging in and removing the adapter.

paroj, 2015-11-01:
@wjbuys did you dump running linux? If so dumping using windows 10 where the handshake actually succeeds would be more vaulable..

Queuecumber, 2015-11-01:
If I can get ahold of a windows 10 box I will see if I can get the dump for you

wjbuys, 2015-11-01:
@paroj yep, that was on Linux. I don't have a Windows 10 box, will see if I can coax it out of a VM.

wjbuys, 2015-11-02:
I've captured the exchange between the wireless adapter and a Windows 10 VM [here](https://dl.dropboxusercontent.com/u/153816/xbox-one-wireless-adapter-virtualbox.pcapng.gz). I couldn't actually get it to pair with a controller though.

Queuecumber, 2015-11-03:
@wjbuys that's weird, was it giving an error or something?

wjbuys, 2015-11-03:
@Queuecumber nope, it just went something like:

* Set up Virtualbox device filter to allow MS devices by vendor ID.
* Plug in adapter. Adapter light is off. Adapter shows up as network adapter in Device Manager.
* Switch on controller. Controller light blinks slowly.
* Long-press adapter pair button. Adapter light blinks.
* Long-press controller pair button. Controller light blinks rapidly, adapter light goes solid.
* ~5 seconds pass, and controller light blinks slowly again (no connection).

This might be a Virtualbox issue (or just me, I can't Windows and I've never used the controller with an actual XBox), so getting a real dump would help a lot.

Chouzinho, 2015-12-03:
I had a dual-boot box with Ubuntu 14.04 and Windows 10 Pro 64-bit. I have the Xbox One wireless adapter for Windows 10, which is correctly paired with a Xbox One controller and I would like to help but I haven't previous experience. What tools should I have to use in order to dum usb traffic in Windows 10?

steverweber, 2015-12-14:
mac driver with xbox one support under active dev... might have some good code to share :)
https://github.com/360Controller/360Controller/pull/151
https://github.com/360Controller/360Controller/issues/160#issuecomment-159952216

Queuecumber, 2015-12-14:
Here are links to the USB traffic captured from their issue 160

https://drive.google.com/file/d/0B381VMfcesrlaG5fTjl3WjZuc2c/view
https://drive.google.com/file/d/0B381VMfcesrlNWZwRkpoTXJBaDA/view
https://drive.google.com/file/d/0B381VMfcesrlRmlKN0tPWGVfdFU/view

mooware, 2016-01-20:
So, is anyone working on this? I'd like to help, if I can.

I did a USB capture in Windows 10 with two controllers, but I have no idea how to interpret it:
https://dl.dropboxusercontent.com/u/267889/xone.pcap

Almamu, 2016-01-22:
I can help implementing this, I do have a Windows 10 box with an XBOX One Wireless adapter + a linux box in which I can test. I just need some more info to start reversing the protocol, does Wireshark detect the wireless dongle as a normal network adapter under Windows 10 (if so, can I just do a capture on it or do I need a special driver or something)?

mooware, 2016-01-22:
I used usbpcap with Wireshark.

drejjmit, 2016-01-28:
Almamu Yes. Under both W10 and W7 it's detected as a Network adapter.

Almamu, 2016-01-29:
@wjbuys Make sure that at least USB 2.0 support is enabled in the VM, I had the same problems until I activated it.

Tadly, 2016-02-11:
Trying to get that thing work with KVM (using libvirt) and USB passthrough to a win10 guest which (I'm assuming) is due to it not being properly recognized by the host (not sure on that though. I'm new to kvm)

What I'm trying to say is:
If there's anything I can help with, I'll do my best to do so.

What I've noticed so far:

1. Adapter does show up on the windows guest as Network adapter
2. When booting the guest, I get a `device descriptor read/64, error -71`

Pairing stick and controller does not work.
Same behaviour as for @wjbuys with virtualbox

Queuecumber, 2016-02-11:
I think at this point we really need someone who can understand the wireshark dumps. Thanks to the mac folks we actually have a couple of dumps floating around (I think). Not that more dumps wouldn't be extremely useful just that the big missing piece is someone who knows how to implement this. I tried to look at it myself but this isn't exactly my area of expertise

steverweber, 2016-02-11:
I read somewhere that the data is ... encrypted. I hope this is just a rumor.

Queuecumber, 2016-02-11:
I really have no idea, but since the USB isn't I don't see why they would do that for the wireless

LiranV, 2016-02-11:
@Queuecumber Probably because the xbox360 wireless adapter had many copies that were cheaper than the original.

steverweber, 2016-02-11:
if they are using encryption, they might be breaking a law...

The controller was only compatible with win10 for some time (felt like ms was trying to force more users to move to win10)... but then they added support for windows 7 near the end of 2015.... http://controllermax.com/forums/showthread.php?t=151076

if they are using encryption then the key might need to be pulled from the driver or somehow bruteforced.

this might spell the last day i buy a ms products.

Queuecumber, 2016-02-16:
Another set of captures courtesy of the mac people 
 
https://drive.google.com/file/d/0B3-WQPnZBRZhRmlNZjltU1pWTms/view

mvittiglio, 2016-02-17:
@paroj Will try to get a dump running Linux when I get home this weekend.

dodslaser, 2016-02-19:
Bought an adapter yesterday and just hooked it up to my win10 laptop. The adapter loads three drivers; `FW_ACC_00U.bin` (firmware I assume), `xboxgip.sys` (no clue) and `mt7612US.sys` (driver for the mediatek mt7612 wireless chip). So the controller must be communicating over 802.11b/g/n/ac right? And the controllers connect to the adapter like a hotspot? I don't have a linux box handy right now, but maybe sniffing the authentication with normal wifi tools, like aircrack-ng, might tell us something.

Almamu, 2016-02-20:
@dodslaser Looks like xboxgip.sys is the driver of the xbox controllers:
http://batcmd.com/windows/10/services/xboxgip/ 

Remember that XBOX pads use wireless communication and not bluetooth so it makes sense to have that driver loaded for the wireless adapter.

I don't think we could use different adapters for this but it's worth trying, although i'm not knowledged in wifi adapters and linux drivers at all... I'm still trying to figure out how to interpret the usbcap captures with wireshark to understand the way windows and the adapter talk to each other and adding support for it on this driver...

drejjmit, 2016-02-20:
The adapter has a physical pairing button. I can move the adapter to another system and it's still paired. I think you need the adapter for this to work.

dodslaser, 2016-02-20:
@drejjmit As long as what the pairing button does can't be done by software.

Almamu, 2016-02-20:
@drejjmit It could be possible (as long as it uses standard 802.11n/b/g/whatever, but you'll probably need to do the pairing every time you want to use your gamepad + full protocol emulation (wireless communications) is needed. I don't think Its impossible, after all the button should do some packet broadcasting for the pad to read and answer

Almamu, 2016-02-20:
Anyway the purpose of this issue is to add support for microsoft's wireles adapter, not make our own adapter. I think we should focus on understanding the os <-> adapter communication to get the pads working with an official adapter. We can try to emulate the wireless adapter at a later time

dodslaser, 2016-02-20:
@Almamu Agreed, lets focus on geeing it to communicate at all first. I was just considering the possibilities. As for scanning with aircrack-ng, I couldn't find anything on the 2.4GHz band. I'll scan 5GHz once I find my other USB WiFi adapter.

steverweber, 2016-02-20:
The controllers also get firmware updates. So the worrying is real.

Queuecumber, 2016-02-20:
We don't even know for sure that it's encrypted yet do we? As far as I know no one with any experience in this area has looked at dumps from the wireless adapter yet.

steverweber, 2016-02-20:
Agreed sorry abort previous posts.

Tadly, 2016-02-20:
@Queuecumber ah crap, sorry...
wanted to share that thread for a while now and kept forgetting about it :/

http://controllermax.com/forums/showthread.php?t=151076&s=0794035b822a0fbaa8e480cb1f8c8676&p=1014946&viewfull=1#post1014946

That's all I know so far...

xdarklight, 2016-02-21:
mt76 is the open source driver for the MT76x2 family (which includes the MT7612).
It's firmware looks suspiciously similar to what FW_ACC_00U.bin contains: https://github.com/openwrt/mt76/tree/master/firmware (I've compared it with mt7662_firmware_e3_v1.9.bin)

rpcastagna, 2016-02-21:
From looking at the dumps that have been uploaded, I don't think the protocol is too complicated. They're apparently using standard headers on the packets (at least Wireshark seems to recognize them) and the patterns seem relatively consistent. They're definitely not encrypted, and it wouldn't make sense from the engineer's perspective to make them encrypted. 

I'm going to play around with these and see what I can do. It'd be really helpful if anyone could make a more extensive dump and annotate it: start recording, plug in without touching anything and see the traffic it produces just from that initial handshake, then systematically press each button once, waiting a second in between, then upload it along with the order that you hit all the buttons in. 

For any other devs reading along, the dumps uploaded so far seem to have a pretty simple pattern:

1. Adapter polls for device
2. Controller says it's alive
3. Adapter requests basic configuration information 
4. Controller says it's a usb 1.1 device that's not powered by the wire and supports remote wakeup
5. Adapter requests more config information
6. Controller identifies itself as a proprietary interface -- presumably Microsoft specific -- and lists 8 endpoints that probably map to the buttons
7. Controller sends extra packet that's not standards compliant but is probably used as part of the configuration
8. Adapter and controller handshake to confirm they understand each other
9. Adapter and controller ping back and forth with each other very frequently, with the controller including status information that probably includes buttons currently pressed

paroj, 2016-02-21:
please try to keep the discussion here focused on getting the controller to work using the adapter.
I have opened a new issue regarding WiFi protocol #25. Will clean up the comments here now.

Queuecumber, 2016-04-17:
Just to bring this to everyone's attention: https://github.com/360Controller/360Controller/issues/156#issuecomment-211083411

One of the Mac guys has figured something out, nothing definitive yet though. When he releases his code I'll take a look at how he did it and see what happens

stephen304, 2016-04-30:
Looks like he's been committing his code to his fork, if anybody here wants to start tinkering with it: https://github.com/haiduc32/360Controller/commits/master

haiduc32, 2016-05-01:
also please check my latest comment on the mac thread.

BadFlo, 2016-06-29:
Any news on this ?

haiduc32, 2016-07-09:
@developers - I managed to crack the most important parts of the protocol to the extent that now I have my xbox one controller sending all input commands through the wireless to my mac. you can find the class that is of interest: https://github.com/haiduc32/360Controller/blob/master/WirelessGamingReceiver/OneWirelessGamingReceiver.cpp
you can find all the commands in there. most of it is initializatioin, then pairing (or smth). some messages are in ProcessMessage as they are out of sync.
some limited information on the messages mostly from ProcessMessage can be found in here: https://docs.google.com/document/d/1AyrhC3gnVAmZyjlCGeJmPauCopRkLN56NyjHiuj8b14/edit?usp=sharing
if anyone is serious about working on that let me know if you need help on the protocol and we can setup a skype chat.

artik, 2016-07-13:
haiduc32 : well done !

cgutman, 2016-07-17:
@haiduc32 the Wireless adapter itself is simply a MediaTek MT7216US NIC. The proprietary part is Microsoft's protocol that runs across it (called Xbox GIP apparently). In Microsoft's stack, they seem to use a stock MediaTek provided NIC driver for the adapter itself and their own protocol driver that sits above it. Microsoft then puts a fairly standard HID stack on top of GIP. There are some quirks needed since the Xbox controllers don't send HID report descriptors (to my knowledge), so one would need to be provided by the OS.

I assume you're driving the whole thing from one driver (the NIC itself and GIP that runs atop it). Maybe it would make your life easier to split out the portion dedicated to the MediaTek NIC and implement the GIP driver separately? That may help even more if the new Bluetooth controller uses the same protocol.

I wrote some thoughts on the best way to structure this code in a thread here: http://thread.gmane.org/gmane.linux.kernel.input/50585

mooware, 2016-09-12:
A quick note regarding the new Xbox One S controller with Bluetooth support: It already works with Linux (I'm using it with my Steam Link). I suppose this change was all it took: 1405c6a61dfcc7bb8609242aa44948caefdd6c11. I also had to update the controller firmware (which is done through the "Xbox Accessories" Windows 10 app; no idea if there's a way to do it without Windows).

TimoVerbrugghe, 2016-09-25:
@mooware: you mean the Xbox One S Controller works with Linux over Bluetooth or using cable only?

mooware, 2016-09-25:
@TimoVerbrugghe: Bluetooth. Only tried it with the Steam Link, don't know about regular Linux distributions. I found this commit which seems to contain the relevant changes in the Linux kernel of the Steam Link: https://github.com/ValveSoftware/steamlink-sdk/commit/d6afa93787217033a0f0f5db5f557b252c36260e

Tazerbot, 2016-09-25:
I have gotten a cap file for the xbox one wireless controller and adapter. Includes pairing of the controller as well as a single press of every button. Let me know if there is anything else you guys need, would love to see this up and running!

Toniob, 2016-09-26:
@mooware @TimoVerbrugghe Currently, the controller is working with a bluetooth connection under a stock distribution (Debian unstable at last). The pairing is ok but to be able to connect the controller, the following sysfs parameter has to be changed : /sys/module/bluetooth/parameters/disable_ertm (from N to Y). The following command will do the trick :
`$ echo 1 > /sys/module/bluetooth/parameters/disable_ertm`
After that, everything is fine, the controller is recognised and works really well (except the vibration which is not working). No issue using a cable connection.

braxion, 2016-10-08:
@cgutman and @haiduc32: Hi guys, is there any progress at the development for the xbox one wireless adapter? Thanx in advance!

ottk3, 2016-10-15:
@mooware did you use the "Microsoft Wireless Adapter" or another BT connection?

mooware, 2016-10-18:
@ottk3 The official wireless adapter discussed in this issue doesn't use Bluetooth, if that's what you meant. I was referring specifically to the new Xbox One S controller, which supports Bluetooth and thus doesn't require the wireless adapter.

TheKrane, 2016-11-03:
Any progress here? Would like to help out if I can somehow.

paroj, 2016-11-03:
someone needs to figure out which of the USB commands used here are really needed:
https://github.com/haiduc32/360Controller/blob/master/WirelessGamingReceiver/OneWirelessGamingReceiver.cpp

a first step would be porting the sequence to this driver and see whether we can receive input too.

Note however that the Bluetooth version already works (reportedly) - so if you just want a working gamepad, you might want to consider one of these.

haiduc32, 2016-11-04:
well.. let me put it this way.. if you skip a step in the initialization faze, the next steps will just return you unexpected values, so the chance of making it smaller is pretty slim. In fact I did skip a quite a few message types (after initialization faze), and it seemed to work. The messages looked like keep-alive, and maybe something to do with keyboard and sound, that were out of scope for me.

devkid, 2016-11-05:
The wireless adapter is actually just a 5 GHz 802.11 wifi dongle. You can sniff the traffic with airodump-ng. I have done a little "reverse engineering" here: https://github.com/paroj/xpad/issues/25#issuecomment-258642789

The chip inside has another USB vendor and product ID, but I think it should be possible to force-bind the appropriate driver to the device. The rest can be done with a link-layer protocol in kernel space or directly from userspace.

Haapavuo, 2016-11-08:
Okay. So now it seems that we have enough information to communicate with the controller since we know the essential parts of the protocol. 

Now we just need someone who can make a linux driver (or two drivers) to support the protocol that is used.

Zsuark, 2016-11-28:
Hi everyone,

I'm using the xpad driver under linux (Linux Mint 18) - still no joy having the XBox One S controller supported. It connects temporarily, but the controller isn't recognised in steam or jstest. (Yes I've tried `echo 1 > /sys/module/bluetooth/parameters/disable_ertm` - which helps but no button presses or stick movements register.)

If there is anything I can do to help - please let me know.

Thanks

R

zommuter, 2016-11-29:
@devkid So does that mean one could also use _any_ wifi dongle instead of the Microsoft one provided the driver is written accordingly?

devkid, 2016-11-30:
@zommuter Yup. With the builtin WiFi in my laptop I managed to get it to semi-functional state in which I received input data from the controller, but it was very unstable, probably because I did it by injecting raw frames and the Intel driver doesn't seem to be very happy about that. My new 5 GHz WiFi router will probably arrive in a few days and I hope that I'll be able to implement it there with the nl80211 protocol instead of frame injection.

If someone has enough time, on could try to get [this](https://github.com/ulli-kroll/mt7612u) driver working by adding the correct USB product and vendor IDs to the driver code. (It bugged out on my system, but might have been a problem with NetworkManager and didn't have any time to continue working on this.)

@Zsuark I think you are on the wrong thread, this issue is for the WiFi variant.

[70 more replies]

closed 2020-02-02 by paroj