paroj/xpad #19: No input with xbox one controller

LiranV, 2015-12-23:
Hi,

When I connect my controller while running current Arch kernel, xpad is crashing.
When I install xpad from git using dkms, my system hangs when I connect the controller.
So I'm running Arch with 4.4.0-rc6-mainline kernel.
With this setup the controller is detected, dmesg:

[ 1340.701682] usb 6-1.5.4: new full-speed USB device number 9 using ehci-pci
[ 1340.801626] input: Microsoft X-Box One pad (Covert Forces) as /devices/pci0000:00/0000:00:1d.0/usb6/6-1/6-1.5/6-1.5.4/6-1.5.4:1.0/input/input33

But when I use jstest it shows the buttons and axis but none of them react when I press the buttons on the controller.

My controller is not the Covert Forces version but it is the regular black version with 3.5mm jack.

Any idea how I can sort this out?

socketbind, 2015-12-24:
Similar problem on Ubuntu 15.10. The system just crashes when using the xpad version shipped with the kernel. Using the driver in this repository fixes the crash, but it is impossible to get any sort of input from the controller. I'm not familiar with kernel module development but I'm happy to help out with testing/capturing/etc, although I will not have access to a Windows PC until January.

It seems that the 360Controller guys have found a solution here:
https://github.com/360Controller/360Controller/pull/120

paroj, 2015-12-24:
seems that newer controllers need an extended initialization sequence instead of a single packet. related #12 

so far we have these two sequences:

```
Drew Mills (360Controller):
{ 0x04, 0x20, 0x01, 0x00 };
{ 0x01, 0x20, 0x01, 0x09, 0x00, 0x04, 0x20, 0x3a, 0x00, 0x00, 0x00, 0x9c, 0x00 };
{ 0x01, 0x20, 0x02, 0x09, 0x00, 0x04, 0x20, 0xd6, 0x00, 0x00, 0x00, 0x00, 0x00 };
{ 0x05, 0x20, 0x02, 0x09, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x55, 0x53 };
{ 0x05, 0x20, 0x03, 0x01, 0x00 };
{ 0x0a, 0x20, 0x04, 0x03, 0x00, 0x01, 0x14 };
{ 0x06, 0x20, 0x01, 0x02, 0x01, 0x00 }; # crash packet?

Ignacio Quezada: (https://github.com/Fenisu/xbox)
( 0x04, 0x20, 0x01, 0x00)
( 0x01, 0x20, 0x01, 0x09, 0x00, 0x04, 0x20, 0x3a, 0x00, 0x00, 0x00, 0x80, 0x00)
( 0x01, 0x20, 0x01, 0x09, 0x00, 0x04, 0x20, 0xba, 0x00, 0x00, 0x00, 0x00, 0x00)
( 0x05, 0x20, 0x02, 0x09, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x55, 0x53)
( 0x05, 0x20, 0x03, 0x01, 0x00)
( 0x0a, 0x20, 0x04, 0x03, 0x00, 0x01, 0x14)
```

will add one of them to xpad as I have time

Toniob, 2015-12-24:
I've got the same issue with a former xbone one controller which has just been updated (using xbox accessories under windows 10). The pad is detected (the /dev/input/ device is created) but the xbox logo stays off - instead of on before the firmware update - and when I test it with jstest, it is recognized but nothing works.

paroj, 2015-12-24:
@Toniob can you give me the dmesg line where your controller is identified? (or the output of lsusb)

paroj, 2015-12-24:
at all:
I have implemented the extended initialization sequence in the [xone2015 branch](https://github.com/paroj/xpad/tree/xone2015).  Can you test and report back?

LiranV, 2015-12-24:
@paroj I'm going to test it now.

socketbind, 2015-12-24:
Going to give a shot too. I believe we all have the same firmware/hardware version.

socketbind, 2015-12-24:
I can confirm that the controller works again.

```
[ 2930.089607] usb 1-2: new full-speed USB device number 3 using xhci_hcd
[ 2930.218910] usb 1-2: New USB device found, idVendor=045e, idProduct=02dd
[ 2930.218914] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2930.218915] usb 1-2: Product: Controller
[ 2930.218917] usb 1-2: Manufacturer: Microsoft
[ 2930.218918] usb 1-2: SerialNumber: 7EED835E32E8
[ 2930.756604] input: Microsoft X-Box One pad (Covert Forces) as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/input/input19
[ 2930.756856] usbcore: registered new interface driver xpad
[ 2930.757069] xpad 1-2:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x01
[ 2930.761062] xpad 1-2:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x01
[ 2930.765066] xpad 1-2:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x05
[ 2930.769068] xpad 1-2:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x05
[ 2930.773080] xpad 1-2:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x0a
[ 2930.782497] xpad 1-2:1.0: xpad_irq_in - urb shutting down with status: -2
[ 2931.799516] xpad 1-2:1.0: xpad_irq_in - urb shutting down with status: -2
[ 2940.108737] xpad 1-2:1.0: xpad_irq_in - urb shutting down with status: -2
```

LiranV, 2015-12-24:
Yes working for me too.
Thank you!
```
[  158.389855] usb 6-1.5.4: new full-speed USB device number 7 using ehci-pci
[  158.517551] input: Microsoft X-Box One pad (Covert Forces) as /devices/pci0000:00/0000:00:1d.0/usb6/6-1/6-1.5/6-1.5.4/6-1.5.4:1.0/input/input28
[  158.517639] usbcore: registered new interface driver xpad
[  158.522850] xpad 6-1.5.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x01
[  158.526845] xpad 6-1.5.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x01
[  158.530846] xpad 6-1.5.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x05
[  158.534847] xpad 6-1.5.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x05
[  158.538845] xpad 6-1.5.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0: 0x0a
[  158.542007] xpad 6-1.5.4:1.0: xpad_irq_in - urb shutting down with status: -2
```

socketbind, 2015-12-24:
Thank you @paroj

paroj, 2015-12-25:
while we got this one resolved, #20 needs to be solved too before we can send this upstream. So maybe someone here can help out. You will find instructions in #20.

closed 2015-12-25 by paroj