paroj/xpad #20: xone: find minimal initialization sequence

paroj, 2015-12-25:
[currently we use a 6 packet initialization sequence](https://github.com/paroj/xpad/blob/xone2015/xpad.c#L742-L754) for xbox one pads.

However it originates from a dump of a single controller on Windows. We need to figure out their meaning and test whether all of them are really needed.

paroj, 2015-12-25:
somebody with a xbox one pad (FW 2015) should uncomment the lines one by one until the controller stops working and report back the needed combination. Do not forget to plug the controller in and out between each try.

socketbind, 2015-12-25:
It seems that only the following initialization packets are needed:
```
struct xpad_output_packet xpad_one_init[] = {
	/* start controller w/o input */
	{{ 0x04, 0x20, 0x01, 0x00 }, 4, true},
	/* this was the only packet needed for pre 2015 firmware*/
	{{ 0x05, 0x20, 0x03, 0x01 /* rumble enable? */, 0x00}, 5, true} // required
};
```

The controller lights up and all buttons/axes work. Removing the first packet will make the controller unresponsive and not send any input. Could somebody please verify this for me with their own hardware?

I used the following testing sequence:
1. Unplug controller
2. rmmod xpad
3. Comment out an element in xpad_one_init
4. sudo dkms remove -m xpad -v 0.4 --all
5. sudo dkms install -m xpad -v 0.4
6. modprobe xpad
7. Plug in the controller and verify with jstest

LiranV, 2015-12-25:
Just tested it too with the same two packets @socketbind used and it works.

paroj, 2015-12-25:
could you do one more test with
```c++
static const struct xpad_output_packet xpad_one_init[] = {
	{{ 0x05, 0x20, 0x03, 0x01 /* rumble enable? */, 0x00}, 5, true},
	{{ 0x05, 0x20, 0x03, 0x01 /* rumble enable? */, 0x00}, 5, true},
};
```
i.e. verify that we really need that specific packet and not an arbitrary initialization packet

socketbind, 2015-12-25:
I replaced the array with the one you provided and it also seems to fix the issue.

paroj, 2015-12-25:
sorry for the noise - I chose the packets poorly as `0x4 & 0x5 == 1`. Could you try:

```c++
static const struct xpad_output_packet xpad_one_init[] = {
	{{ 0x01, 0x20, 0x01, 0x09, 0x00, 0x04, 0x20, 0x3a, 0x00, 0x00, 0x00, 0x80, 0x00}, 13, true},
	/* this was the only packet needed for pre 2015 firmware*/
	{{ 0x05, 0x20, 0x03, 0x01 /* rumble enable? */, 0x00}, 5, true},
};
```

LiranV, 2015-12-25:
@paroj This works for me.
Tested with jstest.

socketbind, 2015-12-25:
Works great with my controller too.

paroj, 2015-12-25:
hmm - so far this does not clarify much.

could you test again using the master branch (as of now). Initially there was a bug such that the first packet was not submitted - maybe that was all..

socketbind, 2015-12-25:
Recompiled it, making it sure that it contains the "back to start.." commit and the controller is working properly with the master branch.

socketbind, 2015-12-25:
I'm still curious though which commit fixed the actual issue because when I reverted the array to its original state (with only the original init sequence) the controller stopped working. I think I was using the xone2015 branch but I accidentally managed to wipe it.

paroj, 2015-12-25:
I force push changes to keep the number of commits low, therefore the broken version wiped from history.

However you this is the fix we actually needed here: c27e6369098b3f5ebbd11f14c0524aa5f9a8c20a

dantob, 2015-12-26:
the version is master is working, but controller is once again detected as special forces. You left out this part 

-	{ 0x045e, 0x02dd, "Microsoft X-Box One pad (Covert Forces)", 0, XTYPE_XBOXONE },
+	{ 0x045e, 0x02dd, "Microsoft X-Box One pad (Firmware 2015)", 0, XTYPE_XBOXONE },

closed 2015-12-26 by paroj