paroj/xpad #21: GameMaker Studio games not detecting Xbox360 wireless gamepads

ghallberg, 2016-01-12:
So, for some reason the Xbox360 Wireless controller only works with games made in GameMaker Studio (Nuclear Throne, Cook Serve Delicious, Risk of Rain etc) if used with xboxdrv.

According to this discussion it's a simple fix, just renaming the controller: http://4archive.org/board/g/thread/206346

paroj, 2016-01-12:
probably the controller just needs to be added to the SDL2 gamecontrollerdb:
https://hg.libsdl.org/SDL/file/0052384f717f/src/joystick/SDL_gamecontrollerdb.h

we can not arbitrarily change the device names as other existing software might expect the name to be as it is.

ghallberg, 2016-01-13:
I understand you can't change the device name, and I don't really have a good solution for this, however I don't think GM:S (http://www.yoyogames.com/studio) uses SDL2?

When looking through the file you linked I don't see any of the related device names mentioned either.

ghallberg, 2016-01-13:
I'm sorry for posting this here, since it's actually a bug in the closed source GameMaker: Studio. I'm going to try to report the bug to them as well, however they aren't very responsive it seems.

I understand that this is not a simple fix but I wanted to highlight it since the cause seems to be clear, and quite a few popular games are affected.

ghallberg, 2016-01-14:
I reported this to the GameMaker people, and they seem to be responding quite well. Hopefully that will work out and no work will be needed on your part :)

ghallberg, 2016-01-25:
I'll just close this issue, there's not much to do from your end. Thanks for the civil response to my frivolous issue.

closed 2016-01-25 by ghallberg