paroj/xpad #22: Unification of xpad.c driver from here and other sources

btegs, 2016-01-23:
I addition to the xpad.c file offered here, there is one for SteamOS via https://github.com/ValveSoftware/steamos_kernel/blob/brewmaster-4.1/drivers/input/joystick/xpad.c and another that this guy wants pushed into Linux 4.5: http://lkml.iu.edu/hypermail/linux/kernel/1601.2/04543.html

Seeing that the xpad.c file offered here, in the SteamOS repo, and what is trying to be pushed into 4.5, is there a way that there can be some unification so there is one bleeding edge xpad driver across the board? Each file has their own little differences and it is confusing to know which is the best.

What would be awesome is if the xpad.c driver offered here includes the patches from the other ones so it is less confusing. Thanks.

paroj, 2016-01-23:
there is only one official xpad driver - the one in the upstream kernel, basically what is being pushed here:
http://lkml.iu.edu/hypermail/linux/kernel/1601.2/04543.html

The xpad driver offered here is basically the same as in Linux 4.5. The differences you see is just because I forgot to push some changes. The flow of code is:
* I apply/ test changes in this repository
* I send it for review to dtor (the linux-input maintainer)
* We fix any issues we find (iterate)
* dtor forwards the changes into the official kernel

as you can see SteamOS does not appear there. Any changes you see originating from SteamOS are cherry-picked by me into this repostory and then bubble through the above pipeline.

btegs, 2016-01-23:
Thanks for the clarification. The SteamOS kernel has setup for the Xbox One Elite controller, which is missing from this one. Any plans on bringing in the code from the SteamOS xpad.c?

paroj, 2016-01-24:
It was already on my TODO list. Now tracked in #23.

dantob, 2016-02-02:
misc usb ids from steamos kernel
https://gitlab.com/dan.g.tob/xpad/commit/bb659b909341c35376725c6d4548f6c2bfe208db
and
https://github.com/ValveSoftware/steamos_kernel/commit/96053cb77390d18cc76cebead210481ea173bee5

MattSturgeon, 2016-03-14:
Is it likely that @ValveSoftware will rebase off the 4.5 xpad, or are there still too many differences?

closed 2016-05-01 by paroj