paroj/xpad #23: xone: add support for Elite Controller from steamos_kernel

paroj, 2016-01-24:
cherry-pick this commit https://github.com/ValveSoftware/steamos_kernel/commit/76e4b04b93b40db186d0c4bbbd1824f5d98c76a9

btegs, 2016-04-19:
So... do we have an update on this? By you not supporting the Xbox One Elite controller, it is keeping it out of the Linux kernel as they tend to use this version instead of the more up to date SteamOS one. Please merge this code in as it is almost 3 months late.

Schroedingers-Cat, 2016-04-24:
So what is the status of this one?

dantob, 2016-04-26:
Looks like it should work; minus the back paddles; if you just add the ID. Give it a try.

> patch -Np1 -i add-xbox-one-elite-id.txt
[add-xbox-one-elite-id.txt](https://github.com/paroj/xpad/files/235643/add-xbox-one-elite-id.txt)

Schroedingers-Cat, 2016-04-26:
I applied the patch to the cloned git repository. But when I'm trying to add xpad to the kernel modules, I get this error:

```bash
sudo dkms install -m xpad -v 0.4

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....(bad exit status: 2)
....(bad exit status: 2)-21-generic KVERSION=4.4.0-21-generic
ERROR (dkms apport): binary package for xpad: 0.4 not found
Error! Bad return status for module build on kernel: 4.4.0-21-generic (x86_64)
Consult /var/lib/dkms/xpad/0.4/build/make.log for more information.
user@T430:/usr/src/xpad-0.4$ 
```

It's happening with and without the patch.

dantob, 2016-04-26:
Looks like your using Ubuntu xenial.

Make sure you install at least:
build-essential linux-headers-generic

Not sure if you'll need anything else. Post /var/lib/dkms/xpad/0.4/build/make.log if it's still not building

Schroedingers-Cat, 2016-04-26:
Yes, I'm using Ubuntu Gnome 16.04. With 15.10, the OS would freeze immediately when plugging in the Xbox One Elite controller. This seems fixed with 16.04 (I read it was a problem with the kernel), so things have improved a bit :)

The logfile is:
```log
DKMS make.log for xpad-0.4 for kernel 4.4.0-21-generic (x86_64)
Di 26. Apr 16:28:02 CEST 2016
make -C /lib/modules/4.4.0-21-generic^M/build V=1 M=/var/lib/dkms/xpad/0.4/build modules
make[1]: *** /lib/modules/4.4.0-21-generic^M/build: Datei oder Verzeichnis nicht gefunden.  Schluss.
Makefile:5: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
```

Both packages, build-essential and linux-headers-generic are already installed.

dantob, 2016-04-26:
> /lib/modules/4.4.0-21-generic/build: File or directory not found.

This should be part of this package, not sure why its not installed, should've been pulled as a dependency. Try reinstall it if its already installed

> linux-headers-4.4.0-21-generic

Schroedingers-Cat, 2016-04-26:
The Folder `/lib/modules/4.4.0-21-generic/build` is there with a lot of links to different fodlers (like blcok, certs, crypto, Documentation ...). The package `linux-headers-4.4.0-21-generic` is also installed. I re-installed the packages linux-headers-4.4.0-21-generic, build-essential and linux-headers-generic. Problems remains the same.

Did you notice that the logfile states `/lib/modules/4.4.0-21-generic^M/build` - what does the `^M` mean? Because that directory is not there and I'd guess that is the problem.

dantob, 2016-04-26:
> what does the ^M mean?
not sure seems like a dkms bug

see if this works
> dkms install -m xpad -v 0.4 -k 4.4.0-21-generic

Schroedingers-Cat, 2016-04-26:
That gives me the same errors and the same log file.

dantob, 2016-04-26:
`KVERSION = $(shell uname -r)`

What does 'uname -r' give you? I don't know why it's adding the `^M` escape char, possibly related to i8n.

Heres a quick patch that just replaces KVERSION with your hardcoded kernel version. 
[tempfix.txt](https://github.com/paroj/xpad/files/237117/tempfix.txt)

Does it print the ^M char on this line after patch?
`Running kernel version is ...`

Schroedingers-Cat, 2016-04-26:
```bash
user@T430:/usr/src/xpad-0.4$ uname -r
4.4.0-21-generic
```
After applying the patch (I did the changes manually because I got this error message `Hunk #1 FAILED at 2 (different line endings).`) I get a slightly changed error messsage:
```bash
user@T430:/usr/src/xpad-0.4$ sudo dkms install -m xpad -v 0.4

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....(bad exit status: 2)
.... KERNELRELEASE=4.4.0-21-generic KVERSION=4.4.0-21-generic
ERROR (dkms apport): binary package for xpad: 0.4 not found
.ko failed for: 4.4.0-21-generic (x86_64)
Consult the make.log in the build directory
/var/lib/dkms/xpad/0.4/build/ for more information.
```

The log is different this time: http://pastebin.com/raw/eKRAq8w8

dantob, 2016-04-26:
`Hunk #1 FAILED at 2 (different line endings)`

Okay now I think I see where the ^M chars are coming from. what editor are you using, seems like its adding DOS line breaks. Try using nano.

`/var/lib/dkms/xpad/0.4` 
Why are you building here. Remove what you have and start again.

Do this first (from the README)
`sudo git clone https://github.com/paroj/xpad.git /usr/src/xpad-0.4`
`sudo dkms install -m xpad -v 0.4`

If that is successful
`sudo dkms remove -m xpad -v 0.4 --all`
`cd /usr/src/xpad-0.4`
`download and apply the first patch I posted`
`sudo dkms install -m xpad -v 0.4`

Schroedingers-Cat, 2016-04-26:
I'm using nano and didn't touch the file Makefile when I tried to patch it. I just retried it, I reset the makefile with git and try to apply the patch and it fails with the quoted line.
This is my .gitconfig:
```
[core]
autocrlf = True
safecrlf = false
```

I followed strictly the README before I visited this issue. Also, I didn't manually cd to /var/lib/dkms/xpad/0.4 and didn't start the build there. 
When I remove /var/lib/dkms/xpad and /usr/src/xpad and start over again, following the README, I run into the same problem:

dantob, 2016-04-26:
`autocrlf = True`
This is causing it, not sure why you have it. Unless you are working with windows repositories, just remove both options.

`ERROR (dkms apport): binary package for xpad: 0.4 not found
.ko failed for: 4.4.0-21-generic (x86_64)`
Not sure what this is, It appears to compile okay, something breaks at install.

try removing this line from dkms.conf
`DEST_MODULE_LOCATION[0]="/extra"`

Schroedingers-Cat, 2016-04-26:
That fixed the problems with dkms and the patching, thank you. I'm working with windows repositories, so I'll just disable these options when I'm pulling from this repo.
After applying the patch and installing the kernel modules, the controller shows up in Steam as XInput and the controllers LEDs stays active. However, no button or axis or anything has an effect in Steam or any game that is compatible with xbox controllers. I also checked it with some scripts I wrote for Unity3d, were I can check for any axis or button activity on up to eleven plugged in gamepads and there is nothing going on. How well has this been tested?

dantob, 2016-04-26:
I'll assume were still talking about the Xbox One Elite controller. The patch is based on what steamos is using, whether the device is actually working with steamos idk. Patch just adds the usb ID. The rest of the code just appears to allow the paddles to be mapped separately from ABXY and not necessary to get the device to work. Possibly more of the steamos initialization packets are needed.

You should hopefully see something in dmesg
`input: Microsoft X-Box One pad (Firmware 2015) as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/input/input48`

Try the jstest tool that comes from this repository https://sourceforge.net/projects/linuxconsole/
`jstest /dev/input/js0` (for first joystick device)

Should see output something like this, press the buttons and make sure they respond correctly.
`Driver version is 2.1.0.
Joystick (Microsoft X-Box One pad (Firmware 2015)) has 8 axes (X, Y, Z, Rx, Ry, Rz, Hat0X, Hat0Y)
and 11 buttons (BtnA, BtnB, BtnX, BtnY, BtnTL, BtnTR, BtnSelect, BtnStart, BtnMode, BtnThumbL, BtnThumbR).
Testing ... (interrupt to exit)
Axes:  0:  2009  1:  1031  2:-32767  3:   266  4:     0  5:-32767  6:     0  7:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 
`

Schroedingers-Cat, 2016-04-26:
Yes, I'm using an Xbox One Elite version of the controller.
This is from dmesg:
```bash
[ 4477.543744] usb 1-1.2: new full-speed USB device number 11 using ehci-pci
[ 4477.639219] usb 1-1.2: New USB device found, idVendor=045e, idProduct=02e3
[ 4477.639223] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4477.639226] usb 1-1.2: Product: Controller
[ 4477.639227] usb 1-1.2: Manufacturer: Microsoft
[ 4477.639229] usb 1-1.2: SerialNumber: 7EED85BA111B
[ 4477.639817] input: Microsoft X-Box One Elite pad as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/input/input21
[ 4477.640672] xpad 1-1.2:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[ 4477.652840] xpad 1-1.2:1.0: xpad_irq_in - nonzero urb status received: -75
[ 4477.665890] xpad 1-1.2:1.0: xpad_irq_in - urb shutting down with status: -2
[ 4477.699830] xpad 1-1.2:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[ 4477.733892] xpad 1-1.2:1.0: xpad_irq_in - urb shutting down with status: -2
[ 4478.667857] xpad 1-1.2:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[ 4478.680900] xpad 1-1.2:1.0: xpad_irq_in - nonzero urb status received: -75
[ 4478.689935] xpad 1-1.2:1.0: xpad_irq_in - urb shutting down with status: -2
```
With jstest, I get the following output:
```bash
$ jstest /dev/input/js0
Driver version is 2.1.0.
Joystick (Microsoft X-Box One Elite pad) has 8 axes (X, Y, Z, Rx, Ry, Rz, Hat0X, Hat0Y)
and 11 buttons (BtnA, BtnB, BtnX, BtnY, BtnTL, BtnTR, BtnSelect, BtnStart, BtnMode, BtnThumbL, BtnThumbR).
Testing ... (interrupt to exit)
Axes:  0:     0  1:     0  2:-32767  3:     0  4:     0  5:-32767  6:     0  7:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 
```
However, moving an axis or pressing a button doesn't change any of the values.

paroj, 2016-04-26:
just adding the id is not enough. The steamos patch also increases the packet size and adds a 3 packet init sequence. (I will remove the clrf discussion as it is not relevant here)

dantob, 2016-04-26:
> The steamos patch also increases the packet size

`nonzero urb status received: -75`
EOVERFLOW, looks like it is needed.

> and adds a 3 packet init sequence

The other two init packets look the same as those initially used (and deemed unnecessary) for the xbox one controller. It's possible the windows driver just tries all init packets for all device types and doesn't filter by ID.

paroj, 2016-04-26:
so in theory https://github.com/paroj/xpad/commit/744b785e6f43c7f69798d08092b296feefcde9c1 should do it? could somebody with the elite controller test current master?

dantob, 2016-04-27:
https://franticrain.github.io/sniffs/XboxOneSniff.html has some good info. It appears to show the packets are (odly) 33 bytes long?

Schroedingers-Cat, 2016-04-27:
I just tried the latest master and it works! jstes, steam and games react to input from the controller! Thank you for your work!

Some games however treat the controller as a "nonstandard" controller, i guess because they do not know that the Xbox One Elite controller basically works the same as the Xbox 360 controller. 
I just checked and realized they have one tiny difference. On the Xbox360, the D Pad is reported as both, as button (11-14) and axis (6 and 7), while the Xbox One Elite only sends the D Pad as Axis (6 and 7). Do you think it would be possible to allow a special mode where the Xbox One Elite controller is reported to be an Xbox 360 controller? And in that mode, the Elite controller would also report its D Pad axes as buttons, just to be safe?

paroj, 2016-04-27:
> On the Xbox360, the D Pad is reported as both, as button (11-14) and axis (6 and 7), while the Xbox One Elite only sends the D Pad as Axis (6 and 7).

this is only the case for the wireless controller see 8be1d4a4d14345e4420bd5f9d4f9d173e0d90791

closed 2016-04-27 by paroj

Schroedingers-Cat, 2016-04-27:
Thanks for the info!
But what about an option to trick the games into thinking that the Xbox One Elite Controller is a Xbox 360 controller? Again, several games just look if it is an xbox 360 controller and if it's not they treat it as an unknown gamepad where you have the map the controls manually (for instance Fez).

paroj, 2016-04-27:
The kernel driver is the wrong place to handle such quirks. See the discussion in #21. Maybe Fez can be "fixed" by adding the elite controller to the SDL2 database.

dantob, 2016-04-29:
@Schroedingers-Cat Do the elite paddles work at all, do they map to they same buttons as ABXY?

Schroedingers-Cat, 2016-05-02:
The elite paddles are mapped as buttons 0-3, from paddle 1-4.

However, I don't use the configuration app for the elite controller as I don't want to use a Microsoft account. So maybe the mapping configured with that app is saved into a memory on the controller.