paroj/xpad #24: Xbox One Elite Controller Crash

pathammer, 2016-02-12:
Just wanted to document that this module does not support the Xbox One Elite controller. Here's the syslog when it's connected:

> Feb 12 10:38:09 ph kernel: [ 1563.206414] usb 3-8.4.1: USB disconnect, device number 16
Feb 12 10:38:35 ph kernel: [ 1589.245090] usb 3-8.4.1: new full-speed USB device number 17 using xhci_hcd
Feb 12 10:38:36 ph kernel: [ 1589.334372] usb 3-8.4.1: New USB device found, idVendor=045e, idProduct=02dd
Feb 12 10:38:36 ph kernel: [ 1589.334375] usb 3-8.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 12 10:38:36 ph kernel: [ 1589.334377] usb 3-8.4.1: Product: Controller
Feb 12 10:38:36 ph kernel: [ 1589.334378] usb 3-8.4.1: Manufacturer: Microsoft
Feb 12 10:38:36 ph kernel: [ 1589.334379] usb 3-8.4.1: SerialNumber: 7EED89C19B94
Feb 12 10:38:36 ph mtp-probe: checking bus 3, device 17: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8.4/3-8.4.1"
Feb 12 10:38:36 ph mtp-probe: bus: 3, device: 17 was not an MTP device
Feb 12 10:38:36 ph kernel: [ 1589.340462] input: Microsoft X-Box One pad (Firmware 2015) as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8.4/3-8.4.1/3-8.4.1:1.0/input/input23
Feb 12 10:38:36 ph kernel: [ 1589.340512] xpad 3-8.4.1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
Feb 12 10:38:36 ph kernel: [ 1589.340573] usbcore: registered new interface driver xpad
Feb 12 10:38:36 ph kernel: [ 1589.354574] xpad 3-8.4.1:1.0: xpad_irq_in - urb shutting down with status: -2
Feb 12 10:38:36 ph kernel: [ 1589.384650] xpad 3-8.4.1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
Feb 12 10:38:36 ph kernel: [ 1589.394624] xpad 3-8.4.1:1.0: xpad_irq_in - urb shutting down with status: -2
Feb 12 10:38:59 ph /py3status: message repeated 58 times: [ user method getText failed (getText() takes 1 positional argument but 3 were given)]
Feb 12 10:39:00 ph /py3status: i3status set_time_modules time failed ('e' is a bad directive in format '%A %e.%B - %l:%M   %Y %m %d')
Feb 12 10:39:00 ph /py3status: message repeated 6 times: [ i3status set_time_modules time failed ('e' is a bad directive in format '%A %e.%B - %l:%M   %Y %m %d')]
Feb 12 10:39:00 ph /py3status: user method getText failed (getText() takes 1 positional argument but 3 were given)
Feb 12 10:39:00 ph /py3status: i3status set_time_modules time failed ('e' is a bad directive in format '%A %e.%B - %l:%M   %Y %m %d')
Feb 12 10:39:00 ph /py3status: i3status set_time_modules time failed ('e' is a bad directive in format '%A %e.%B - %l:%M   %Y %m %d')
Feb 12 10:39:01 ph /py3status: user method getText failed (getText() takes 1 positional argument but 3 were given)
Feb 12 10:39:05 ph kernel: [ 1618.511994] xpad 3-8.4.1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
Feb 12 10:39:05 ph kernel: [ 1618.528107] xpad 3-8.4.1:1.0: xpad_irq_in - urb shutting down with status: -2
Feb 12 10:39:08 ph kernel: [ 1621.322810] xpad 3-8.4.1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
Feb 12 10:39:48 ph kernel: [ 1661.783950] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.788028] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.792026] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.795981] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.799944] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.804024] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.808023] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.811976] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.815938] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.819952] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.823951] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.827950] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.831949] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.835941] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.839939] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.844007] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.848005] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.852006] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.856006] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.860003] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.864003] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.867997] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.871999] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.875938] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.879924] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.883935] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.887932] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.891930] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.895925] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.899923] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.903922] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.907917] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.911915] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.915915] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.919911] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.923909] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.927908] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.931903] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.935902] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.939900] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.943897] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.947896] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
Feb 12 10:39:48 ph kernel: [ 1661.951895] xpad 3-8.4.1:1.0: xpad_irq_in - nonzero urb status received: -71
[... 54 lines cut ...]
Feb 12 10:39:56 ph kernel: [ 1670.206062]  [<ffffffff814b943b>] __device_attach+0x3b/0x40
Feb 12 10:39:56 ph kernel: [ 1670.206064]  [<ffffffff814b71b3>] bus_for_each_drv+0x63/0xa0
Feb 12 10:39:56 ph kernel: [ 1670.206066]  [<ffffffff814b8fa8>] device_attach+0x88/0xa0
Feb 12 10:39:56 ph kernel: [ 1670.206068]  [<ffffffff814b8438>] bus_probe_device+0x98/0xc0
Feb 12 10:39:56 ph kernel: [ 1670.206070]  [<ffffffff814b62a4>] device_add+0x4f4/0x630
Feb 12 10:39:56 ph kernel: [ 1670.206072]  [<ffffffff81561288>] usb_new_device+0x3c8/0x790
Feb 12 10:39:56 ph kernel: [ 1670.206074]  [<ffffffff8156a432>] ? usb_get_descriptor+0x82/0xc0
Feb 12 10:39:56 ph kernel: [ 1670.206076]  [<ffffffff815624d8>] hub_port_connect+0x578/0xa00
Feb 12 10:39:56 ph kernel: [ 1670.206078]  [<ffffffff81563153>] hub_events+0x7f3/0xce0
[... cut ...]
Feb 12 10:39:56 ph kernel: [ 1670.206118] RIP  [<ffffffffc120dc63>] xpad_probe+0x293/0x5d0 [xpad]
Feb 12 10:39:56 ph kernel: [ 1670.206120]  RSP <ffff880416e838f8>
Feb 12 10:39:56 ph kernel: [ 1670.206121] CR2: 000000000000004e
Feb 12 10:39:56 ph kernel: [ 1670.206123] ---[ end trace 7ad4cee274d90638 ]---
Feb 12 10:39:56 ph kernel: [ 1670.217283] xpad 3-8.4.1:1.0: xpad_irq_in - urb shutting down with status: -2
Feb 12 10:39:56 ph kernel: [ 1670.233277] xpad 3-8.4.1:1.0: xpad_irq_in - urb shutting down with status: -2

Let me know if I can provide more info to help debug.

LiranV, 2016-02-12:
Take a look at this #23

pathammer, 2016-02-12:
@LiranV beautiful, this is a duplicate then.

closed 2016-02-12 by pathammer