paroj/xpad #25: xone: Wireless Adapter - wifi protocol

paroj, 2016-02-21:
@dodslaser
Bought an adapter yesterday and just hooked it up to my win10 laptop. The adapter loads three drivers; `FW_ACC_00U.bin` (firmware I assume), `xboxgip.sys` (no clue) and `mt7612US.sys` (driver for the mediatek mt7612 wireless chip). So the controller must be communicating over 802.11b/g/n/ac right? And the controllers connect to the adapter like a hotspot? I don't have a linux box handy right now, but maybe sniffing the authentication with normal wifi tools, like aircrack-ng, might tell us something.

**Edit:** Searching FCC database for the FCC-ID (C3K 1713) yeilds a lot of information (Internal photos etc.). It is listed as capable of using the 5.2, 5.8 and 2.4 GHz bands. [LINK](https://fccid.io/C3K1713)

**Edit 2:** The test reports seem to confirm that the adapter is in fact just a WiFi adapter. Maybe we could even use other adapters if we are able to come up with a Linux driver?

**Edit 3:** This document claims that the wireless platform is proprietary, dual band and uses 5GHz (So 5.2 and 5.8 I assume) for communication. [TDS_XboxOneControllerPlusWDAforWindows_1510A.pdf](https://github.com/paroj/xpad/files/139111/TDS_XboxOneControllerPlusWDAforWindows_1510A.pdf)

paroj, 2016-02-21:
@Almamu
Looks like xboxgip.sys is the driver of the xbox controllers:
http://batcmd.com/windows/10/services/xboxgip/ 

Remember that XBOX pads use wireless communication and not bluetooth so it makes sense to have that driver loaded for the wireless adapter.

I don't think we could use different adapters for this but it's worth trying, although i'm not knowledged in wifi adapters and linux drivers at all... I'm still trying to figure out how to interpret the usbcap captures with wireshark to understand the way windows and the adapter talk to each other and adding support for it on this driver...

paroj, 2016-02-21:
@xdarklight
mt76 is the open source driver for the MT76x2 family (which includes the MT7612).
It's firmware looks suspiciously similar to what FW_ACC_00U.bin contains: https://github.com/openwrt/mt76/tree/master/firmware (I've compared it with mt7662_firmware_e3_v1.9.bin)

rpcastagna, 2016-02-21:
I'm not sure about the PHY layer details for the adapter, but the packets themselves seem pretty standard and not particularly difficult to map (though I haven't really begun in earnest, so who knows). I can write the code to do that mapping but I'm not familiar with the existing driver code and I'm a little confused. I understand the packet/input system, but am I reading it right that the 360 wireless input didn't need to be translated into the wired version's input? It seems like they both just pick up the packets and send them along with minimal changes, which I wouldn't expect -- but I've also only done this kind of thing for networking protocols where you do a lot of parsing.

eligao, 2016-02-29:
+1 I just tried wiresharking the wireless packets between Xbox one controller and windows adapter, part of the protocols seem pretty straightforward though some other parts are not understood yet.

rpcastagna, 2016-03-26:
I've been working with this more and things are more complicated than I might have hoped.

1. Windows recognizes the adapter and the controller as two distinct devices. It would assign a USB port registry to the adapter itself, and then when the controller linked in Windows created another device number for it. Fortunately it doesn't seem that Windows interacts very much with the adapter itself and the communication between the adapter and the controller seems to be transparent to the OS; all the OS sees are packets coming directly from the new controller device.

2. The initial handshake between the controller and the OS seems to be very elaborate.

3. Almost all communication between the controller and the OS happens in URB_BULK packets.

4. From what I can tell, there is a LOT of polling taking place happening on the order of .1 second increments. The annoying thing is that there seems to be some kind of sequence number/timestamp included in the "nothing happening" packets (probably the "action" packets too) both to and from the controller, and the pattern is not immediately obvious to me. I'd assume that the exchange of these packets are needed to keep the link alive, however. 

5. There seem to be a few different types of "no action" packets, each with different lengths. I'm seeing 59 byte, 43 byte, and 51 byte packets most commonly, but the order/pattern is not obvious to me. Actions seem to be transmitted in packets around 110 bytes. There are occasional 1059 byte packets initiated by the OS that are mysterious to me.

devkid, 2016-11-05:
I'll chime in on this: The wireless adapter really just seems to be a 5 GHz Wifi Dongle. With the dongle connected, I see traffic on 5 GHz channels, it seems to be hopping: first I saw traffic at channel 44, then at channel 48. So, if it really contains the mt7612 chip, I could simply force-bind the driver to the USB device. (I need to get and compile the driver first, it seems like it's somewhere in OpenWrt's repo).

After I paired the dongle and the controller, I observed the following:
* Packets were sent as long as the controller was connected to the dongle (i.e. power on and the X LED being on).
* Beacons were always sent. (as long as the dongle was plugged in)

In fact, the "pairing" could possibly be in WPS style. I'll go through the airodump-ng recordings and dump there here in a while.

devkid, 2016-11-05:
Here are the Wifi dumps for any actions on the controller: http://dl.devkid.net/uNqiOXi32H2Ul-_pEZtUHw/xbone-wireless.tgz (including powering on the controller (connecting to the dongle), any button presses (I forgot analog stick presses), and powering off the controller (disconnecting from the dongle)).

During this second testing, the devices didn't hop channels, they stayed on channel 36. airodump-ng displayed WEP as encryption, but that might be a bug (it displayed also "-1" as channel number although I configured it to use channel 36).

devkid, 2016-11-05:
Hex representation of packets sent for button A:

* pressed: `8811-a000-6245b4f44151-7eed801f6d53-6245b4f44151-100000002000020e-1000-0000-0000-4f039c02-b4ff26ff`
* released: `8811-a000-6245b4f44151-7eed801f6d53-6245b4f44151-200000002000030e-0000-0000-0000-4f039c02-b4ff26ff`

`6245b4f44151` seems to be the MAC address of my dongle (that's probably the changing part people keep seeing in the USB dumps) and `7eed801f6d53` the MAC address of my controller. The dongle's MAC address appears twice in Wireshark ("Receiver" and "Destination"). `8811` is the "Frame Control Field". The last part with many zeroes seems to be the actual data.

Taking apart the data part: (bytes counted in order of arrival, bits counted from MSB to LSB, i.e. bit 0 is MSB)
* byte 0-1:
  * bits 0-3 together with bits 8-15 seem be a 12-bit counter that increases with every button press, analog stick move etc.
  * bits 4-7 seem to be constant zero
* bytes 2-5: seems to be constant `0x00002000`
* byte 6: seems to be a 8-bit counter, it seem to be equal to the following bits of byte 0: [bits 8-11, bits 0-3]+1 (plus 1)
* byte 7: seems to be constant `0x0e`
* bytes 8-9: seem to be the button map, where the bits equal: (0 = pressed, 1 = released)
  * 0: Y
  * 1: X
  * 2: B
  * 3: A
  * 4: left buton
  * 5: right button
  * 10: RB
  * 11: LB
  * 12: DPAD right
  * 13: DPAD left
  * 14: DPAD down
  * 15: DPAD up
* bytes 10-11: LT position
* bytes 12-13: RT position
* bytes 14-17: left analog stick position
* bytes 18-21: right analog stick position

The XBox button produces another pattern for the data part:
* pressed: `2002010007200302015b`
* pressed: `3002010007200402005b`
bits:
* the same rules seem to apply for the counter bits (bytes 0-1, 6)
* byte 4-7: seems to be constant `0x01000720`
* byte 7: seems to be constant `0x02`
* byte 8:
  * bits 0-6: seem to be constant zero
  * bit 7: seems to be the negated state of the button (0 = released, 1 = pressed)
* byte 9: seems to be constant `0x5b`

Guys, please stop me here, if someone has already deciphered this :)

devkid, 2016-11-07:
On Windows, the driver seems to write out raw 802.11 MAC frames over the USB connection. That means that it should indeed be possible to use any 5 GHz module that has SoftMAC. I'm on it.

devkid, 2016-11-12:
I managed to get [this](http://www.mediatek.com/AmazonS3/Downloads/linux/MT7612U_DPO_LinuxSTA_3.0.0.1_20140718.tar.bz2) driver to compile on a Ubuntu 12.10 with Linux 3.5. I changed the first entry in `rtusb_dev_id[]` at `common/rtusb_dev_id.c` to the USB vendor and product ID of the original dongle (045e:02e6) and got a `wlan1` interfaces when I plugged in the dongle, but I couldn't get it up with `ip link set wlan1 up` (Operation not permitted) and when I unplugged it, the kernel paniced. 

I think using this driver directly wouldn't be of much use as it doesn't seem to implement mac80211 anyways. But maybe some initialization steps could be used from the code.

xdarklight, 2016-11-12:
@devkid it's probably a lot of work but one could try adding USB support to [Felix' mt76 driver](/openwrt/mt76).
not sure how many differences there are between USB and PCIe devices, but Felix is a nice guy and will probably share his thoughts if you ask him about adding MT7612 USB support to mt76.

devkid, 2016-11-14:
@xdarklight Thanks for the link! Who is Felix? nbd186?

FYI: I have now finally managed to output some 802.11 frames on 5240 MHz on my ThinkPad laptop after fiddling around for hours (after I actually got it working on 2.4 GHz! – but no errors on 5 GHz, just nothing going over the air). This laptop has an Intel Wireless 7260 card (but this may as well affect other wireless devices, too). If you have a look at the output of `iw phy`, there might be a flag "no IR" (no Initiating Radiation) for all 5 GHz frequencies. That means that you cannot send beacon frames (amongst other frame types) on those frequency. Even changing the regulation domain with `iw reg set DE|US` didn't help. The solution was to recompile the `iwlwifi` driver and remove every line which set the `IEEE80211_CHAN_NO_IR` flag for the channels; this might even help for other drivers that implement mac80211.

I'm now going to implement the protocol for the pairing, connecting and data exchange in userland. This will probably make for an ok'ish driver until we get to figure out the MT7612U driver.

devkid, 2016-11-23:
I have pasted the analysis of everything I've figured out about the protocol [here](https://gist.github.com/devkid/4b3bd50760504d1b93ea684cfd3ed895). Maybe someone has enough muse to decipher those mysterious reserved frames or status messages :)

perillamint, 2018-01-10:
Hello everyone.

I want to work on this issue. Recently, I purchased Xbox One Elite controller and I'm planning to do my research using some ath9k USB dongles.

My question is: is there any available information(i.e. 802.11 packet capture) about pairing stage?

chris-ritsen, 2018-09-26:
If anyone just wants to use this controller on Linux in any way possible, I've found the ConsoleTuner Titan 2 device supports this.  The device acts as an input translator and has its own firmware to use the wireless adapter.

Vegz78, 2020-12-09:
With all you guys’ amazing research and work here and elsewhere on the «XBox wireless» protocol on the XBox adapter and controllers, did you ever find out how to monitor and inject a wifi device with the controller’s sequence to power on the XBox One from energy-saving mode? (PSU is really loud and draining 15W in instant-on mode...).

If so, would it be possible to get a peek? If not, do you think it would be possible to achieve with  
https://github.com/seemoo-lab/nexmon on the Raspberry Pi, with internal or a capable USB wifi adapter?

chromey85, 2021-02-27:
Is it possible for the xbox one controller to be forced to connect to win10 via the home WiFi as opposed to using the dongle?

M3hmetSa1t, 2022-05-05:
someone can try this dongle and use xbox wireless adapter drivers? maybe it can be works https://edup.en.alibaba.com/product/60623990718-803973115/MT7612_wireless_usb_wifi_adapter_1200mbps_usb_wifi_dongle.html