paroj/xpad #28: Force feedback not working

ggeo1992, 2016-04-09:
I'm using this module with a xbox one controller (2015 firmware) on a raspberry pi 2 with the latest version of raspbian installed, the kernel version that i have is  4.1.20-v7: 
Bus 001 Device 005: ID 045e:02dd Microsoft Corp.
 Everything work ok except the vibrations. I didn't managed to find anything on the web related to this issue. I also tried to test the vibrations using the following command with various signals:
fftest /dev/input/by-id/usb-Microsoft_Controller_7EED809D6B4F-event-joystick
This is what i found is /var/log/sys:

Apr  9 17:56:35 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Sat Apr  9 17:57:05 2016 [try http://www.rsyslog.com/e/2007 ]
Apr  9 17:56:36 raspberrypi kernel: [18156.736148] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:54 raspberrypi kernel: [18174.766241] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.216228] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.266227] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.316229] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.366240] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.416227] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.466225] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.516226] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.566225] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.616226] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.666225] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.716244] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:56:55 raspberrypi kernel: [18175.736234] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:57:32 raspberrypi kernel: [18212.629964] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1
Apr  9 17:57:32 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Sat Apr  9 17:58:02 2016 [try http://www.rsyslog.com/e/2007 ]
Apr  9 17:57:35 raspberrypi kernel: [18215.709914] xpad 1-1.5:1.0: xpad_prepare_next_out_packet - found pending output packet 1

Please let me know if more info is needed.

paroj, 2016-04-16:
well it does send the FF packets, so probably the packets themselves are wrong. Will take a look at it in a couple of weeks.

MattSturgeon, 2016-06-01:
Could be fixed by [this](https://patchwork.kernel.org/patch/9145737/) from Cameron Gutmon

amenophobis, 2016-06-03:
jep, it works...

changed it with gnome builder, the changes need to be made from line 1022

paroj, 2016-06-05:
applied e2418b29e3480e71c686f9dfad8dc592b1ac36ac to master

closed 2016-06-05 by paroj