paroj/xpad #31: xone elite: allow mapping back-paddles seperately

paroj, 2016-04-27:
take parts of this commit https://github.com/ValveSoftware/steamos_kernel/commit/76e4b04b93b40db186d0c4bbbd1824f5d98c76a9

dantob, 2016-04-29:
My xbox one controller sends 18 byte packets. Bytes 0-17 do not appear to be different on the xbox one elite controllers (backwards compatibility?). Bytes 18-32 appear to be elite only. 

Bytes 18-31 appear to be a mirror of bytes 4-17, but used for button remaps? Byte 32 appears to be special and relates to preset toggle switch

dantob, 2016-04-29:
Problem with this though is it's going to register paddles as two presses (paddle + whatever its mapped to). Might not be possible without special casing elite controllers like steamos.
[xpad-first-try-mapping-elite-paddles-separately.txt](https://github.com/paroj/xpad/files/241887/xpad-first-try-mapping-elite-paddles-separately.txt)

devsnek, 2020-09-18:
Any word on this? I've been trying to patch this locally but I am not at all familiar with the code here so I haven't been able to get anything to work.

```diff
diff --git a/xpad.c b/xpad.c
index ed90ca4..d0d931d 100644
--- a/xpad.c
+++ b/xpad.c
@@ -143,6 +143,7 @@ static const struct xpad_device {
        { 0x045e, 0x02dd, "Microsoft X-Box One pad (Firmware 2015)", 0, XTYPE_XBOXONE },
        { 0x045e, 0x02e3, "Microsoft X-Box One Elite pad", 0, XTYPE_XBOXONE },
        { 0x045e, 0x02ea, "Microsoft X-Box One S pad", 0, XTYPE_XBOXONE },
+       { 0x045e, 0x0b00, "Microsoft X-Box One Elite Series 2 pad", 0, XTYPE_XBOXONE },
        { 0x2e24, 0x0652, "Hyperkin Duke X-Box One pad", 0, XTYPE_XBOXONE },
        { 0x045e, 0x0719, "Xbox 360 Wireless Receiver", MAP_DPAD_TO_BUTTONS, XTYPE_XBOX360W },
        { 0x046d, 0xc21d, "Logitech Gamepad F310", 0, XTYPE_XBOX360 },
@@ -494,6 +495,19 @@ static const u8 xboxone_s_init[] = {
        0x05, 0x20, 0x00, 0x0f, 0x06
 };
 
+/*
+ * For Xbox One Elite Series 2 (0x045e:0x0b00)
+ */
+static const u8 xboxone_elite_init1[] = {
+       0x05, 0x20, 0x00, 0x09, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x55, 0x53
+};
+static const u8 xboxone_elite_init2[] = {
+       0x05, 0x20, 0x01, 0x01, 0x00
+};
+static const u8 xboxone_elite_init3[] = {
+       0x0A, 0x20, 0x02, 0x03, 0x00, 0x01, 0x14
+};
+
 /*
  * This packet is required for the Titanfall 2 Xbox One pads
  * (0x0e6f:0x0165) to finish initialization and for Hori pads
@@ -551,9 +565,12 @@ static const u8 xboxone_rumbleend_init[] = {
 static const struct xboxone_init_packet xboxone_init_packets[] = {
        XBOXONE_INIT_PKT(0x0e6f, 0x0165, xboxone_hori_init),
        XBOXONE_INIT_PKT(0x0f0d, 0x0067, xboxone_hori_init),
+       XBOXONE_INIT_PKT(0x045e, 0x0b00, xboxone_elite_init1),
+       XBOXONE_INIT_PKT(0x045e, 0x0b00, xboxone_elite_init2),
+       XBOXONE_INIT_PKT(0x045e, 0x0b00, xboxone_elite_init3),
        XBOXONE_INIT_PKT(0x0000, 0x0000, xboxone_fw2015_init),
        XBOXONE_INIT_PKT(0x045e, 0x02ea, xboxone_s_init),
-       XBOXONE_INIT_PKT(0x045e, 0x0b00, xboxone_s_init),
+       /* XBOXONE_INIT_PKT(0x045e, 0x0b00, xboxone_s_init), */
        XBOXONE_INIT_PKT(0x0e6f, 0x0000, xboxone_pdp_init1),
        XBOXONE_INIT_PKT(0x0e6f, 0x0000, xboxone_pdp_init2),
        XBOXONE_INIT_PKT(0x24c6, 0x541a, xboxone_rumblebegin_init),
@@ -877,6 +894,14 @@ static void xpadone_process_packet(struct usb_xpad *xpad, u16 cmd, unsigned char
        input_report_key(dev, BTN_X,    data[4] & 0x40);
        input_report_key(dev, BTN_Y,    data[4] & 0x80);
 
+       /* elite controller paddles; bit 5 depends on preset switch */
+       if (data[32] != 0x00) {
+               input_report_key(dev, BTN_BASE,   data[32] & 0x01); /* upper left paddle */
+               input_report_key(dev, BTN_BASE2,  data[32] & 0x02); /* upper right paddle */
+               input_report_key(dev, BTN_BASE3,  data[32] & 0x04); /* lower left paddle */
+               input_report_key(dev, BTN_BASE4,  data[32] & 0x08); /* lower right paddle */
+       }
+
        /* digital pad */
        if (xpad->mapping & MAP_DPAD_TO_BUTTONS) {
                /* dpad as buttons (left, right, up, down) */
```

zakk4223, 2021-10-20:
I did a bit of protocol peeking and figured out you can query the controller for the paddle mappings and then just expose any that are un-mapped. 

However, doing this leads me to believe it is a terrible user experience. The problem is if you expose them as BTN_BASE(2|3|4) like the valve patch does it screws up all the button mappings for almost every game I tried. I suspect a lot of the controller detection out there just assumes 'first button is A, second is B' etc. 

There are some BTN_ slots AFTER all the normal buttons, but I'm not sure if things will get confused with paddles sending stuff like BTN_STYLUS etc. I'll give it a shot soon

chaorace, 2022-03-26:
Anyone interested in this issue may be interested in testing PR #195, which implements paddle controls for the Elite controllers. The more testers, the better!

PR #195 (2022-03-28): Input: xpadone - add support for XBOX One Elite paddles (attempt #2)
Resolves #31

Second attempt at adding support for XBOX One Elite paddles. This time using BTN_TRIGGER_HAPPY for the paddles instead of BTN. See #194 regarding why the first attempt had to be reverted, see #193 for the original PR.

I've used jstest to verify the bits are working as expected. The button mapping works out as follows:
| Paddle Position | Mapping | Button Number |
| --- | --- | :--- |
| Left hand, upper | BTN_HAPPY3 | 13 |
| Left hand, lower | BTN_HAPPY4 | 14 |
| Right hand, upper | BTN_HAPPY1 | 11 |
| Right han [... cut]
closed 2022-03-28 by paroj