paroj/xpad #37: xpad one: (0)07 packets also send axis data

dantob, 2016-05-04:
Currently axis data is only read from (0)20 packets, (0)07 packets also appear to send axis data for triggers and sticks. Missing these packets could cause input lag / delays.

Packets look like this:

>( 0)  20 Button packet 
( 1)  ??Always 00
( 2)  Packet id, inc from 00, wraps at FF 
( 3)  ??Always 0E
( 4)  Menu = 04, View = 08, A = 10, B = 20, X = 40, Y = 80
( 5)  UP = 01, DOWN = 02, LEFT = 04, RIGHT = 08, LB = 10, RB = 20, LSTICK = 40, RSTICK = 80
( 6)  LT AXIS (2 bytes) le
( 7)  LT AXIS (2 bytes) le
( 8)  RT AXIS (2 bytes) le
( 9)  RT AXIS (2 bytes) le
(10)  LSTICK LEFT-RIGHT AXIS (2 Bytes) le
(11)  LSTICK LEFT-RIGHT AXIS (2 Bytes) le
(12)  LSTICK DOWN-UP AXIS (2 Bytes) le
(13)  LSTICK DOWN-UP AXIS (2 Bytes) le
(14)  RSTICK LEFT-RIGHT AXIS (2 Bytes) le
(15)  RSTICK LEFT-RIGHT AXIS (2 Bytes) le
(16)  RSTICK DOWN-UP AXIS (2 Bytes) le
(17)  RSTICK DOWN-UP AXIS (2 Bytes) le

>( 0)  07 Home button special packet (packet also sends axis but not digital buttons)
( 1)  ??Always 20
( 2)  Packet id, inc from 00, wraps at FF 
( 3)  ??Always 02
( 4)  HOME = 01
( 5)  ??Always 5B
( 6)  LT AXIS (2 bytes) le
( 7)  LT AXIS (2 bytes) le
( 8)  RT AXIS (2 bytes) le
( 9)  RT AXIS (2 bytes) le
(10)  LSTICK LEFT-RIGHT AXIS (2 Bytes) le
(11)  LSTICK LEFT-RIGHT AXIS (2 Bytes) le
(12)  LSTICK DOWN-UP AXIS (2 Bytes) le
(13)  LSTICK DOWN-UP AXIS (2 Bytes) le
(14)  RSTICK LEFT-RIGHT AXIS (2 Bytes) le
(15)  RSTICK LEFT-RIGHT AXIS (2 Bytes) le
(16)  RSTICK DOWN-UP AXIS (2 Bytes) le
(17)  RSTICK DOWN-UP AXIS (2 Bytes) le

dantob, 2016-05-04:
Probably less important, but (0)03 heartbeat packets also send axes data. But there appears to be a bug with LT (bytes 6-7) are reset to 00. RT & sticks are sent same as above

paroj, 2016-05-04:
is the 07 packet sent on any button press or only on home button press?

dantob, 2016-05-04:
07 appears to be once for press and once for release of home button. Possibly Windows asks for it more often, only logging on Linux.
03 appears to be every 20 seconds

I used this in xpadone_process_packet to print packets & dmesg --follow
```
/* print recieved packet to kernel log */
u8 i;
for (i = 0; i < XPAD_PKT_LEN; i++) {
    	printk("(%i)%02x ", i, data[i]);
} printk("\n");
```

paroj, 2016-05-05:
> Missing these packets could cause input lag / delays.

this is rather theoretical then..

dantob, 2016-05-05:
Yeah, (0)20 packets appear to be sent about every 8 milliseconds minimum. When a (0)03 packet is sent it doesn't appear to increase that. Not sure about (0)07 packets yet.

```
[52606.292251] (0)20 ...
[52606.300249] (0)20 ... 7998
[52606.308242] (0)20 ... 7993
[52606.316247] (0)20 ... 8005
[52606.324247] (0)20 ... 8000
[52606.332216] (0)20 ... 7969
[52606.340248] (0)20 ... 8032	
[52606.344239] (0)03 ...		3991  appears to just slot itself between the 8ms button packets
[52606.348246] (0)20 ... 7998   4007
[52606.356245] (0)20 ... 7999
[52606.364212] (0)20 ... 7967
[52606.372236] (0)20 ... 8024
```