paroj/xpad #42: Elite Wireless Ctrl (via USB) qemu: libusb_set_interface_alt_setting: -5 [NOT_FOUND]

alchristiansen, 2016-06-21:
He.

Dont know if this is really related to xpad.

When i try to passthrough the Elite Wireless Controller with qemu to a Windows 10 box i get:
(qemu) libusb_set_interface_alt_setting: -5 [NOT_FOUND]
The Elite Wireless Controller does show up in Windows device manager with code 10 (unusable).

After researching a bit, it looks like this has to do with sub device handling for the Elite Controller. Maybe cause by xpad?

Running Linux 4.7.0-rc4, qemu 2.1.2, Debian 8

By the way, i am quite happy that my box does not crash any more when using the Elite Wireless Controller. Great :)

Axel

paroj, 2016-09-04:
ce711a47fcf682c5cebae29275559fe3dabdc6aa could have helped here. please test with latest master.

paroj, 2016-12-20:
no reply from reporter

closed 2016-12-20 by paroj