paroj/xpad #43: xpad-dbg messages are flooding my dmesg and crippling my disk space

brujoand, 2016-07-31:
```
[79027.005134] xpad-dbg: 00000000: 02 20 01 1c fc ae 52 60 4f 05 00 00 c6 24 1a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[79027.005156] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[79027.605214] xpad-dbg: 00000000: 02 20 02 1c fc ae 52 60 4f 05 00 00 c6 24 1a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[79027.605231] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[79028.205303] xpad-dbg: 00000000: 02 20 03 1c fc ae 52 60 4f 05 00 00 c6 24 1a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[79028.205320] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[79028.805376] xpad-dbg: 00000000: 02 20 04 1c fc ae 52 60 4f 05 00 00 c6 24 1a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[79028.805394] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[79029.405456] xpad-dbg: 00000000: 02 20 05 1c fc ae 52 60 4f 05 00 00 c6 24 1a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[79029.405473] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

After a day or so my 16gb sd card on my RP3 is clogged. I've tried

```
sudo dmesg -n 1
```

```
sysctl -w kernel.printk="2 4 1 7"
```

But nothing seems to suppress these messages. Please advice on how to disable them.

dantob, 2016-07-31:
If you remove or comment out '#define DEBUG' on line 77 and recompile you shouldn't see any more messages https://github.com/paroj/xpad/commit/9b8b78cc26033c05b69c091ffe43a509663290f6

Maybe it shouldn't be set by default.

MattSturgeon, 2016-07-31:
Shouldn't it be possible to define `DEBUG`  through the Makefile? That would mean it could be set using an enviroment variable or CLI flag or something

brujoand, 2016-08-01:
Thanks, recompiling with line 77 gone did the trick. I would be nice to be able to alter this behaviour during runtime though like pointed out by Matt.

dantob, 2016-08-02:
Here's a go at using a module option instead, does this look okay @paroj? https://github.com/dantob/xpad/commit/a3ce3c24b94d0d610e31016568ee271bf98fa163
You can drop commit https://github.com/paroj/xpad/commit/c3e8784d73cc6f326a479e6c7025e0ea826375d2, this replaces it

paroj, 2016-08-03:
I dont think we need this debug output most of the time and recompiling seems fine to be to enable it. Therefore I changed the guarding define to DEBUG_VERBOSE, effectively commenting it out for now.

closed 2016-08-03 by paroj