paroj/xpad #44: Kernel module does not load

amenophobis, 2016-08-02:
Hi,

I needed to reinstall my system and now the xpad kernel module does not load... on the old system there was the same version as on the new, but it won't work on the new system...

Here's my install log:

`Creating symlink /var/lib/dkms/xpad/0.4/source ->
                 /usr/src/xpad-0.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.6.4-301.fc24.x86_64 KVERSION=4.6.4-301.fc24.x86_64....
cleaning build area...

DKMS: build completed.

xpad.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.6.4-301.fc24.x86_64/extra/
Adding any weak-modules

depmod....

DKMS: install completed.`

amenophobis, 2016-08-02:
There seems to be a problem with secure boot... Since kernel version 4.4.0-20, it was enforced that unsigned kernel modules will not be allowed to run with Secure Boot enabled...

Here's a workaround to sign kernel modules

http://gorka.eguileor.com/vbox-vmware-in-secureboot-linux/

paroj, 2016-08-03:
closing as this is not an xpad issue.

closed 2016-08-03 by paroj