paroj/xpad #50: Xbox One Firmware 2015

cap-42, 2016-10-15:
I'm really grateful for all the great work you guys put into this and since I came here last, you added support for the newer models as well.
I do however have a few issues when my controller is connected or more specifically after I've used it. All the buttons work fine, but after pressing certain buttons, there seem to be some ghostly remnants of the keys pressed. For example, after pressing right trigger, the mouse constantly moves to the top of the screen. Unplugging stops this instantly. There was more odd behavior, but I can't reproduce it at the moment.
Is there a file I may need to edit that uses controller input for keyboard/mouse?

paroj, 2016-10-21:
this driver does not provide mouse/ keyboard input. This issue seems not xpad related.

closed 2016-11-22 by paroj

dantob, 2016-12-31:
Just for reference, this is caused by xf86-input-joystick. Simply remove it unless you actually want mouse cursor control. The triggers value at rest is -32767 and this as you say locks the cursor at the top. I think it was the Z axis that causes trouble.