paroj/xpad #51: cannot turn off blinking light /sys/class/leds/xpad0/.... missing

fnpanic, 2016-10-20:
Hi,

i compiled 0.4 on my osmc (Linux osmc 4.4.16-6-osmc #1 SMP PREEMPT Sun Oct 2 13:04:26 UTC 2016 armv7l GNU/Linux) and the driver works like a charm with my xbox 360 wireless controller.

Now i want to turn off the blinking green circle but i am missing

/sys/class/leds/xpad0/....

Anything i need to do?

Maybe someone has a hint.

paroj, 2016-10-21:
the blinking should stop automatically once connected. Can you post the output of `dmesg` after you connect the controller?

fnpanic, 2016-10-21:
Hi,

thanks for the quick reply

When i plug in the wireless reciever in the raspberry pi 3 dmesg shows this

[82465.013734] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.013795] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.013851] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.013878] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.014717] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.014751] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.014833] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.014857] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.015706] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.015731] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.015828] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.015852] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.016705] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.016733] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.016832] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.016856] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.017709] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.017735] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.017832] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.017857] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.018708] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.018735] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.018825] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.018849] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.019724] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.019834] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.019855] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.019873] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.020702] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.020722] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.020825] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.020857] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.021709] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.021731] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.021827] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.021847] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.022713] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.022734] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.022752] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.022770] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.023712] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.023734] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.023752] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.023770] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.024712] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.024733] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.024751] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.024768] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.025715] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.025737] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.025757] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.025775] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.026711] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.026732] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.026750] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.026768] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.027713] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.027737] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.027756] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.027773] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.028722] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.028746] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.028765] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[82465.028784] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[... 432 lines cut ...]
[82465.137713] xpad 1-1.4:1.4: xpad_irq_in - nonzero urb status received: -32
[82465.137727] xpad 1-1.4:1.6: xpad_irq_in - nonzero urb status received: -32
[82465.137739] xpad 1-1.4:1.0: xpad_irq_in - nonzero urb status received: -32
[82465.137750] xpad 1-1.4:1.2: xpad_irq_in - nonzero urb status received: -32
[... cut ...]
[82490.957837] xpad 1-1.4:1.2: xpad_prepare_next_out_packet - found pending output packet 0
[82490.958817] xpad 1-1.4:1.4: xpad_prepare_next_out_packet - found pending output packet 0
[82490.959933] xpad 1-1.4:1.6: xpad_prepare_next_out_packet - found pending output packet 0
[82511.957509] input: Xbox 360 Wireless Receiver (XBOX) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/input/input5

I need to wake up the pad to connect it again and it works in emulation station very well so far (at least in the UI)

paroj, 2016-10-25:
seems like the underlying USB communications is very unstable on your system. You could try connecting the receiver using a powered USB hub.

Ntemis, 2016-12-15:
Hi
Am getting the same issue as reported on amlogic 3.10 kernel with this or vanilla driver, all controller leds blink( blinking green circle) vs only a stable one quarter and is annoying.
Can provide anything asked to get this fixed

paroj, 2016-12-17:
please provide the output of `dmesg` after you connect the controller

Ntemis, 2016-12-17:
[  498.454924@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -5
[  515.092824@3] input: Xbox 360 Wireless Receiver (XBOX) as /devices/lm1/usb1/1-1/1-1.3/1-1.3:1.0/input/input2

Ntemis, 2016-12-17:
Here is full dmesg after usb receiver unplug then replug and re-sync wireless controller
[  800.024160@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -32
[  800.024271@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -32
[  800.024309@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -32
[  800.024346@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -32
[  800.025143@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -32
[  800.025268@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -32
[  800.025303@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -32
[  800.025337@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -32
[  800.026143@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -32
[  800.461424@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -32
[  800.478174@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.478226@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.478244@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.479146@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.481149@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.481165@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.481179@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.482141@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.484159@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.484176@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.484189@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.485161@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.487161@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.487178@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.487192@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.488147@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.490156@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.490171@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.490184@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.491147@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.493157@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.493172@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.493185@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.494145@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.496153@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.496170@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.496186@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.497145@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.499153@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.499166@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.499178@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.500146@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.502157@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.502173@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.502186@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.503156@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.505165@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.505182@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.505196@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.506153@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.508161@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.508177@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.508190@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.509154@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.511170@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.511202@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.511218@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.512154@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.514165@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.514189@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.514205@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.515157@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.517163@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[  800.517178@3] xpad 1-1.3:1.2: xpad_irq_in - nonzero urb status received: -71
[  800.517191@3] xpad 1-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[  800.518155@3] xpad 1-1.3:1.6: xpad_irq_in - nonzero urb status received: -71
[  800.520162@3] xpad 1-1.3:1.4: xpad_irq_in - nonzero urb status received: -71
[... cut ...]
[  810.738489@3] xpad 1-1.3:1.2: xpad_prepare_next_out_packet - found pending output packet 0
[  810.739007@3] xpad 1-1.3:1.4: xpad_prepare_next_out_packet - found pending output packet 0
[  810.739522@3] xpad 1-1.3:1.6: xpad_prepare_next_out_packet - found pending output packet 0
[  820.685098@3] input: Xbox 360 Wireless Receiver (XBOX) as /devices/lm1/usb1/1-1/1-1.3/1-1.3:1.0/input/input3
[  881.997637@3] input: Xbox 360 Wireless Receiver (XBOX) as /devices/lm1/usb1/1-1/1-1.3/1-1.3:1.0/input/input4

paroj, 2016-12-18:
you are also using an ARM SoC right? can you try a powered USB hub? I think the issue is related with the receiver not getting enough power over the USB port.

Ntemis, 2017-02-15:
Ok i will be back on this asap, powered usb hub is here so we can continue to debug this one

Ntemis, 2017-02-25:
I want to report that after your last change this is not an issue for me anymore, controller connects and upper left light stays green, For me you can close this down.Issue fixed! And no usb hubs needed.

closed 2017-02-27 by paroj