paroj/xpad #52: send Power off to xbox controller

martinalberto, 2016-11-20:
Hello:
Thanks for this works, it is perfect. 
I want to power off my controller by terminal.
There is some command that i can send by terminal  to Power off.?

I find this section in you code: 
```
static bool auto_poweroff = true;
module_param(auto_poweroff, bool, S_IWUSR | S_IRUGO);
MODULE_PARM_DESC(auto_poweroff, "Power off wireless controllers on suspend");
```

but i can't  suspend my controller.
I am use the Retrpo pie, Hardware Raspberry pi 3, SO Raspbian.

Thanks. 
Thanks very much.

paroj, 2016-12-20:
powereoff currently only works for x360w controllers. The option you see sends the power off command automatically on suspend. There is no option to trigger this otherwise - you can only disable the LEDs via terminal as described in the readme.

martinalberto, 2016-12-22:
Ok. thanks very much.
I hope, i could suspend my Xbox 360 controler.

sand3ro, 2022-04-09:
could it be possible to power off Xbox360 controller automatically also on  system shutdown/reboot?

carlosefr, 2024-03-07:
Powering off the controllers on shutdown is the most obvious use case for being able send the power off comand on demand. Currently, the controllers stay on after shutdown and the only way to power them off is by pulling the battery pack.

the-isz, 2026-10-12:
I know that this is marked as low priority, but I'd still like to note that this is the **only** thing I'm missing in the otherwise phenomenal xpad driver.

Having to remember to manually turn off the controller when shutting down the computer or pulling the battery pack / USB power cable afterwards is pretty inconvenient.