paroj/xpad #53: XBox one Bluetooth Controller

DigitalKrony, 2016-11-21:
This is a question more than an issue,

Is there any information on pairing and connecting the new XBox One Wireless controller released with the `S`?

While using `bluetoothctl`
- The controller is seen
- I can `pair` the controller
- I add it to the `trust` list
- When I attempt a `connect` to the controller, I get a `bluez` error.
    - there's little information other than Failed to Connect

paroj, 2016-11-22:
see here: https://github.com/paroj/xpad/issues/14#issuecomment-249609916

DigitalKrony, 2016-11-22:
Brilliant. Now to get that surfaced better on a search or "How To" so it's more easily found. Thanks so much for pointing me in the right direction.

Zsuark, 2016-11-27:
I followed these instructions but my XBox One Wireless controller still has it's light blinking.

`bluetoothctl` reports that it is paired and trusted, but the connection seems to drop in and out.

I'm using Linux Mint Sarah and kernel 4.7. (Should I be using kernel 4.8?)

Zsuark, 2016-11-28:
Hi everyone,

I'm using the xpad driver under linux (Linux Mint 18) - still no joy having the XBox One S controller supported. It connects temporarily, but the controller isn't recognised in steam or jstest. (Yes I've tried `echo 1 > /sys/module/bluetooth/parameters/disable_ertm` - which helps but no button presses or stick movements register.)

If there is anything I can do to help - please let me know.

Thanks

R

Spiffyk, 2016-12-01:
Hi, I'm having this problem on Linux Mint 18, too. I did try `disable_ertm` but no luck. When hooked up with a cable, it works as expected.

terramare, 2016-12-15:
After much trial and error, my Xbox wireless controller is paired and connects to Retropie. Two things made the difference for me: 1) first connecting the controller to a Windows 10 VM vis USB and updating the firmware via the XBox Accessories app, and 2) running the `echo 1 > /sys/module/bluetooth/parameters/disable_ertm` command mentioned above. Now if I could just get all buttons to register in the Configure Input process in emulationstation, I'd be in business.

terramare, 2016-12-17:
@DigitalKrony Is your Xbox One S controller fully functional over Bluetooth? I can't get my Select button to be recognized in Input Configuration. Every other button is fine. What's more, if I press the Select button to initiate the configuration, it shows "Keyboard" rather than "Xbox Wireless Controller" on-screen. 

I know the physical button is okay because I have two controllers and both are fine wired and both exhibit the same Select button problem over Bluetooth. I have also done a fresh Retropie install.

I am new to the platform and have run out of ideas of things to try. Any leads or ideas most welcome.

Should I submit this as an issue here?

Spiffyk, 2016-12-18:
After doing pretty much what @terramare did (upgrading the firmware via a VM and disabling ertm) I also managed to connect my Xbox One S controller via bluetooth. However, the mappings are all messed up with many games. I have found one oddity that might be related to the problem: When using jstest-gtk, the controller is displayed as `Microsoft X-Box One S pad` when hooked up using a cable. However, when I connect it via bluetooth, it is displayed as `Xbox Wireless Controller`. Also, all the mappings of buttons and axes are in a different order and have different names for each of these cases. There are also more buttons registered when it's connected via Bluetooth.

Here's what it looks like when connected via USB:
[image]

Here's what it looks like when connected via Bluetooth:
[image]

cgutman, 2016-12-18:
Yes, that's because the controller exposes an Android-friendly mapping over Bluetooth since the November firmware update. 

This is also the reason that some special buttons like back don't work correctly in some applications. They are now HID Consumer usages rather than HID Button usages. This is required to get the desired back and home behavior on Android's default keymap. 

I believe an SDL2 patch has been checked in to fix the inability to use consumer buttons in controller configurations.

DigitalKrony, 2017-01-02:
@terramare - I wasn't able to try things out til now. I've been denied permission by the system to run the bash task successfully. Trying to track down the case of that now. So as of right now, I've had not progress.

terramare, 2017-01-02:
@DigitalKrony Do you mean when you are trying to run the echo command? Depending on your distro, first type `sudo su`, followed by the echo command, and then issue `exit` to get out of the root account.

DigitalKrony, 2017-01-02:
@terramare - Thanks for that. I got it connect. Solid light on the Controller but no access in RetroPie. When I run jstest-gtk the Controller throws numbers so I know it's connected and is sharing data. However, on restart of the PI, I loosed the capability to connect to the controller. Also, because it's not showing up in RetroPie, it's not building a *.cfg file so I can't set the appropriate button mappings.

DigitalKrony, 2017-01-03:
Ok, general update.

I did fresh RetroPie install.
switched into `root` and ran the bash command from putty.
paired/trusted/connect the XBox One controller.
restarted just emulation station (not the entire OS).
emulation station was able to read and setup the XBox One Controller. 
After reboot, however, the XBox controller no longer connects. 

Anyone have any ideas?

P.S. when I run the echo bash command from `root` it doesn't echo anything back. Is that normal? or should I expect a response from a lasting and successful command request?

terramare, 2017-01-03:
@DigitalKrony Your echo command is working as it should; this change does not persist a reboot on Retropie. You can add this command to your rc.local file and it will run each time your Pi reboots.

Open the file in a text editor:

`sudo nano /etc/rc.local`

and then add:

`echo 1 > /sys/module/bluetooth/parameters/disable_ertm`

near the bottom of the file, but before the `exit 0` line at the end.

This should get you to the same boat I'm in, which is a non-functional Select button, but otherwise working bluetooth controller. Please confirm if this is the case.

Would love to know of a workaround.

DigitalKrony, 2017-01-03:
I seem to be moving backwards. 

The controller won't connect correctly. 

Getting a Bluez on connect attempt and it never fully connects (but the light on the controller itself becomes solid). 

`connect C8:3F:26:32:0F:61`
`Attempting to connect to C8:3F:26:32:0F:61`
`Device C8:3F:26:32:0F:61 Connected: yes`
`Failed to connect: org.bluez.Error.Failed`

DigitalKrony, 2017-01-04:
@terramare - Got it. I was able to get things up and running. As for the `select` button issue you're running into. It's a quick and easy solution.

Utilizing `$ jstest /dev/input/js0` I was able to record all of the Axes and Buttons associated with the new controller. This is what I got. Oddly enough, `start` does not work in EmulationStation, but it does work in game.

`
input_device = "Xbox Wireless Controller"
input_driver = "udev"

input_up_btn = "-7"
input_right_btn = "+6"
input_down_btn = "+7"
input_left_btn = "-6"

input_a_btn = "0"
input_b_btn = "1"
input_x_btn = "3"
input_y_btn = "4"

input_l_btn = "8"
input_l2_btn = "4"
input_l3_btn = "-5"

input_r_btn = "9"
input_r2_btn = "5"
input_r3_btn = "+4"

input_l_y_plus_axis = "-1"
input_l_x_plus_axis = "+0"
input_l_y_minus_axis = "+1"
input_l_x_minus_axis = "-0"

input_r_y_plus_axis = "-3"
input_r_x_plus_axis = "+2"
input_r_y_minus_axis = "+3"
input_r_x_minus_axis = "-2"

input_start_btn = "7"
input_select_btn = "6"

input_enable_hotkey_btn = "6"

input_reset_btn = "3"
input_menu_toggle_btn = "0"
input_exit_emulator_btn = "7"

input_save_state_btn = "4"
input_load_state_btn = "5"
input_state_slot_increase_btn = "+6"
input_state_slot_decrease_btn = "-6"
`

Jacob-Newton, 2017-01-23:
I've managed to connect mine up to the point where everything works except the select button, and binding the triggers never seems to work correctly in emulation station. Has anyone managed to get these working?
Using the model 1708, and a raspberry pi 3 with the built in Bluetooth chip

Also the only way I can get the controller to pair is by using the echo command mentioned above, followed by going into bluetoothctl and running the agent on commands, as well as removing and repairing the controller. Is there any way to run these commands on boot?

codeart1st, 2017-02-04:
Hey guys which version of bluez do you use. I have no success with my 5.37 version.

```
bluetoothctl
scan on
scan off
```

connect, pair, trust all combinations of it didn't seem to work. Also using register-profile with the given HID uuid didn't work for me.

I also tried with updating the firmware of the gamepad to the newest version, but still not working.

I using an ODROID C2 instead of pi2 or pi3.

mauriliodc, 2017-02-20:
@DigitalKrony how did you manage to get the select working via Bluetooth? 
All my buttons are working but the select (using the built-in Bluetooth of my laptop, 16.04 4.4.0-31)

DigitalKrony, 2017-02-20:
@mauriliodc I gotta be honest, I didn't do anything special. I was using a Raspberry Pi 2 and a Bluetooth Dongle. But when I moved the SD card with active OS loaded on it to a Pi 3 and a built-in BT, I got totally different button mappings and the select button no longer worked.

howl, 2017-06-03:
The issue with the keys is more than only the back, homepage, gas, brake codenames issue.

When using with usb cable the module used is xpad, and it works as expected, I mean, the buttons and axes the pad have are detected. When using with bluetooth connection xpad is not used, I suppouse that there is a generic bluetooth hid, and the problem is that despite select becames back and mode becames homepage, there are buttons detected that doesn't exists in the pad.

This makes A button0, B button 1, X button 3 and Y button 4, and the button 2 doesn't exists in the pad.

With evtest and usb you can see:
```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x45e product 0x2ea version 0x301
Input device name: "Microsoft X-Box One S pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   1538
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 1 (ABS_Y)
      Value    582
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max     1023
    Event code 3 (ABS_RX)
      Value   -439
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 4 (ABS_RY)
      Value   -340
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max     1023
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)
Properties:
```

And with bluetooth:
```
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x45e product 0x2fd version 0x903
Input device name: "Xbox Wireless Controller"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 158 (KEY_BACK)
    Event code 172 (KEY_HOMEPAGE)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 309 (BTN_Z)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value  34120
      Min        0
      Max    65535
      Fuzz     255
      Flat    4095
    Event code 1 (ABS_Y)
      Value  33514
      Min        0
      Max    65535
      Fuzz     255
      Flat    4095
    Event code 2 (ABS_Z)
      Value  32261
      Min        0
      Max    65535
      Fuzz     255
      Flat    4095
    Event code 5 (ABS_RZ)
      Value  32244
      Min        0
      Max    65535
      Fuzz     255
      Flat    4095
    Event code 9 (ABS_GAS)
      Value      0
      Min        0
      Max     1023
      Fuzz       3
      Flat      63
    Event code 10 (ABS_BRAKE)
      Value      0
      Min        0
      Max     1023
      Fuzz       3
      Flat      63
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
```

So over bluetooth there are 17 buttons where the pad only have 11, BTN_C, BTN_Z, BTN_TL2, BTN_TR2, BTN_SELECT and BTN_MODE shouldn't appear.

I don't know where to file bug for the bluetooth mapping get corrected.

Also using MAME under windows it shows button names, for example, Joy 1 button 0 is shown in linux but in Windows shows something like this Joy 1 Button A.

daviderickson, 2017-06-03:
All-
I've got a couple of fixes that solve the select problem for Retropie emulators using libsdl or libretro:

https://github.com/libretro/RetroArch/pull/4692 - MERGED

https://github.com/RetroPie/SDL-mirror/pull/4
https://github.com/RetroPie/SDL-mirror/pull/5
https://github.com/RetroPie/SDL-mirror/pull/6

The last three have not been merged, despite the fact that the code in the pull requests is actually just backported code that has already been accepted into libsdl upstream.  I encourage you to comment on these and encourage @joolswills to test/merge them.

howl, 2017-06-03:
The problem about appearing more buttons than the pad have over bluetooth is not related to that patches.

daviderickson, 2017-06-04:
@howl the problem at least in Retropie is that the emulators and their libraries are not expecting to scan for key codes outside of a very specific range, and the Xbox One pad via Bluetooth is sending these keys.  So these patches fix that.  The main problem folks were reporting there was the lack of a select button, which is one of these other key codes.

howl, 2017-06-04:
I know that issue, moreover there should be also lack of of MODE button as it's HOMEPAGE button over bluetooth, also I don't know in retropie, but it's very possible to be also a problem with the button order, as there are buttons detected over bluetooth than doesn't exists in the pad.

Also if we are 100% purist, all this issue is not related to xpad, and that's what i'm asking for, where is the code for the mapping of the pad over bluetooth? In BlueZ layer? or evinput layer? Because there are more issues over bluetooth to address, I repeat not related to xpad, like your patches that also are not related to xpad.

howl, 2017-06-04:
I have found that in https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/input/gamepad.rst says the mapping scheme for gamepads over the kernel.

There are discrepancies with the xbox one buttons over usb and over bluetooth as north and west buttons are swapped, Y should be north button but it's west, and X should be west but it's north. Also with xbox 360 pads north and west are also swapped.

Also If I understand it right DPAD should be buttons instead axis, as hat axis are for pressure buttons and at least the xbox pads I could test are just 0 1 buttons.

BlackVodka, 2017-06-17:
Hi there folks,
please excuse my short question here in between your discussion:
@daviderickson, I cloned and compiled your forked and patched SDL repo from https://github.com/RetroPie/SDL-mirror.git and built and installed that on my retropie / RPi 3 system, as well.
Also, I built the current master of Retroarch, as your patches are already merged into upstream/master branch there AFAIK.
Shouldn't the select button on my Xbox One Bluetooth pad now be working?
It still outputs justs an '@' in jstest .
Am I missing something here, perhaps something quite stupid as a 'sudo ldconfig' or so?
I'd really love to get this pad working here.
Thanks in advance and let me know if I could anyhow help you or if you need more information.

Spiffyk, 2017-06-17:
@BlackVodka I've been facing this problem yesterday. I've found that the Xbox One S controller probably has two modes for Bluetooth. One intended for Android (which causes these problems) and one intended for Windows (which works correctly). My assumption is that these moves are set via some signal from the host (that xpad does not send). I managed to switch the controller into "Windows mode" by connecting it to a Windows machine via Bluetooth.

BlackVodka, 2017-06-17:
@Spiffyk Thanks for your quick response.
This is an interesting idea, will try that.
Have you complete unpaired the controller from the Pi before connecting to the windows client?
Also, has anybody tried the original 'Microsoft Wireless receiver' bluetooth-dongle on the pi?
Perhaps this thingy does some sort of remapping?

Spiffyk, 2017-06-17:
@BlackVodka Well I haven't tried it with a Pi, I use Ubuntu on my laptop but I don't think this behaviour would be affected by this. I also assume the vendor of your Bluetooth receiver has nothing to do with this, I'd say this is a software issue (may or may not have to do with xpad directly, I don't know what xpad does and does not deal with exactly).

Regarding your question: I think I did unpair the controller from my laptop before trying to connect it to a Windows PC. Basically what I did was unpair the controller from my laptop, paired it with the Windows machine, waited for Windows to identify the device as an Xbox Controller, then unpaired it from the Windows machine and paired it back to the laptop and since then it worked as intended.

daviderickson, 2017-06-17:
@BlackVodka FYI the receiver is actually 802.11 not bluetooth.

howl, 2017-06-18:
@Spiffyk I  will try too, I suppouse that the "Windows mode" that Windows sets, last till the controller is blinking without auto powering off. I will try to check also to pair with an android phone when in that mode.

Anyways at least me the controller in bluetooth mode doesn't use xpad at all, it gets connected as a standard bluetooth hid device.

PD: Which Windows version are you using?

Spiffyk, 2017-06-18:
@howl I have no idea when the mode is set on the controller as I had it running fine in the "Windows mode" and then for some reason it switched to the "Android mode"... but now that I'm thinking about it, that might be the case. Let me try, too.

I used Windows 10 for this, as it has the Xbox Accessories app that can be used for updating the controller's firmware.

BlackVodka, 2017-06-18:
@Spiffyk I fiddled here a bit around between my raspberry and my windows 10 pc and, unfortunately, wasn't successful in getting any change in the pad's behaviour.
I tried just going from one room into the other and connect it to the windows pc, tried to first unpair it from the pi and just afterwards pairing it with the PC, followed by a reconnect/repair with the pi and all such scenarios...no luck :/
You were absolutely right btw. that using the original/bundled bt-receiver by microsoft (and disabling the pi's internal bluetooth by blacklisting the according module) didn't change anything at all.

@daviderickson I thought these new One Pads are using bluetooth (I'm connecting it using the RPi3's internal bluetooth, not the wireless chipset AFAIK) ?
What I was talking about are the receivers coming along with these new pads (I bought a bundle about a month ago) and these are clearly stated as bluetooth device in windows.

@howl Okay, interesting.
Just for learning how things work, how did you notice that?
Does anyone know how e.g. a ps3 controller (which works here like a charm btw.) does work (precisely, the handling of the bt-stack and the kernel driver module itself)?
This has to be kind of the same way, like kernel -> bt-stack -> gamepad kernel module/driver , so I'm curious what's getting handled differently with an ps3 pad in comparison to the one pad?

Spiffyk, 2017-06-18:
I'm confused. Once I got it into the "Windows mode", I can't seem to get it back to the "Android mode". So I can't test this. Of course no modes are mentioned anywhere officially but I'm pretty sure they are a thing because of the inconsistent behaviour. I'll try investigating further.

@BlackVodka The new XBOne pads use 802.11 when connected via the dongle or directly to the console. They only use Bluetooth with a PC or another Bluetooth-capable device.

howl, 2017-06-18:
@Spiffyk I can't get the windows mode, perhaps I don't do like you do.

I pair it in Windows 10 with it's bluetooth screen, then I switch back to linux and I press again the connect button in the pad to pair it with linux (it doesn't pair if I don't do it). When testing I get the usual homepage, back, brake, gas button trigger names.

howl, 2017-06-18:
@BlackVodka I don't have xpad module loaded if I only connect it with bluetooth, that how I notice it and also by dmesg, as it shown hid-generic. When using the pad connected to with the usb cable then xpad gets loaded and works with it, and it's ok with no issues, the issues are only when using over bluetooth (buttons and triggers that doesn't exits detected and buttons working as gas, brake, homepage, back instead of z, rz mode and select).

Spiffyk, 2017-06-19:
I've been fiddling around with it and I found out I don't have the xpad module loaded either when connected via bluetooth, even though my controller is working properly... so it may not even be an xpad issue after all and the problem might be lying elsewhere. But I'm really running out of ideas...

howl, 2017-06-19:
@Spiffyk it's strange as if you connect the usb the pad stops bluetooth comunication.

The cable you are using is a data cable or it only have ground and 5 V pins used.

I'm going to try blacklisting the xpad module to test if it retains bluetooth connection even with the usb cable.

howl, 2017-06-19:
@Spiffyk I have done the test and only if I don't let xpad module to be loaded it stays connected over bluetooth even with the usb cable. But if xpad it's present when you connect it via usb cable it gets loaded, and the the bluetooth connection it's closed.

Test the pad with xpad loaded and connecting it via usb cable, you will see that then the pad behaves like the previous pads, z and rz triggers, correct number of buttons detected... and xpad manages it.

BlackVodka, 2017-06-20:
Hi, just here from work in my lunch break:
I compared (just a quick view) how the ps3 controller works (https://github.com/RetroPie/sixad/blob/master/sixad-bin.cpp) compared to https://github.com/paroj/xpad/blob/master/xpad.c .
Is it possible that's xpad just doesn't know anything of bluetooth and due to that the linux kernel manages this HID device, because I cannot find anything regarding to accessing the device besides directly via bluetooth in xpad's source? (in comparison sixad has a bluetooth HID daemon hid_server() in bluetooth.cpp).

howl, 2017-06-20:
Could be, but if the bugs can be addressed in the generic hid of the kernel that could be interesting, as the pad could work out of the box in more systems. For example in fedora to get xpad you must install kernel-modules-extra or from another external repository.

I don't know where can be found the code that actually manages the pad over bluetooth.

cgutman, 2017-06-20:
Ok there's a ton of confusion here.

Xpad has *nothing* to do with dealing with the Xbox One S Controller when connected via Bluetooth. The device exposes a standard HID interface over Bluetooth that requires no special driver like xpad to handle. 

I do not know of any bugs affecting the HID stack that affects this gamepad when connected via BT. However, there is an issue with the standard Linux Bluetooth stack which prevents the gamepad from connecting by default. 

Valve has written a hack to workaround the issue (which also changes mappings, but this is not necessary with a patched SDL): https://github.com/ValveSoftware/steamos_kernel/commit/549c3dc10fa3749b3999549a2672b14bf0e9786d

Android's Bluetooth stack is not affected by this behavior.

howl, 2017-06-20:
@cgutman with generic hid over bluetooth more than the mappings, there is the problem that there are more buttons and axes that the pad have. For example in mame you have to rebind all the controls as the button2 in the pad doesn't exists.

cgutman, 2017-06-20:
While not ideal, HID devices exposing extra non-functional buttons is not uncommon and should be properly handled by userspace. What exactly is problematic about having the extra buttons?

howl, 2017-06-21:
For example mame doesn't work out of the box (I don't mean the android mappings about back and homepage).

When in bluetooth the first buttons the pad reports are:
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)

So mame takes button 0 the south one that it's A button, button 1 east that is B button, and then button 2 is btn_c that doesn't exists in the pad, so the mame automapping fails and you have to reset the buttons. Doesn't happen only with that button because there is also the btn_z and the triggers z and rz despite they don't work as gas and brake take into account. The problem here is that the extra buttons detected makes automapping systems to fail by assigning the inexistant buttons.

One of the most typical system for example neogeo. If you start mame and play metal slug that uses just the first 3 buttons, you can't throw grenades as the button doesn't exists.

For my use case I have made my own mame ctrlr.ini file to workarround it without having to remap manually all the machines, but it could be interesting that it works out of the box, also because the same mapping would work for all the pads.

cgutman, 2017-06-21:
Yes, this is simply a bug in MAME. It should not assume the first 4 buttons are ABXY - there's no requirement for that to be true. They should add special handling in their automapper for this device.

Spiffyk, 2017-06-21:
So basically, as far as xpad is concerned, this issue is invalid, right? Shouldn't it be marked as such to prevent further confusion?

howl, 2017-06-21:
@Spiffyk indeed.

@cgutman thanks for the explanation, I will try to file a bug to mame, at least in Windows mame shows even the button names in the pad instead of button# so I think the linux event device or it's detection should be improved, don't know if in mame, sdl or in linux generic hid.

radical-equanimity, 2017-06-25:
Alright - with Ubuntu 16.04 I've downloaded kernel sources and rebuilt hid and xpad modules after manually adding the changes from commit [d6afa93787217033a0f0f5db5f557b252c36260e](https://github.com/ValveSoftware/steamlink-sdk/commit/d6afa93787217033a0f0f5db5f557b252c36260e) of ValveSoftware/ steamlink-sdk (recommended by user mooman [here](https://github.com/paroj/xpad/issues/14). Instructions for building kernel modules can be found [here](https://wiki.archlinux.org/index.php/Compile_kernel_module). Has there been any update to make mapping over Bluetooth non-Android-friendly (i.e., just like it is over USB)?

cgutman, 2017-06-25:
The HID mapping will never be the same as the xpad mapping.

xpad was written years before this gamepad was released and many userspace programs depend on its mappings the way they are today. Since we don't have a time machine, we can't go back and make xpad match Microsoft's HID mappings. Similarly, there are userpsace programs that depend on the HID mappings present over Bluetooth as they exist today (Chrome, for example).

Changing xpad's mappings to match HID or HID's mappings to match xpad would break userspace, so it would not be accepted upstream. The onus is on userspace (generally libraries like SDL) to abstract gamepad device differences and quirks.

radical-equanimity, 2017-06-25:
Sounds like middleware that can present the Bluetooth HID as an xpad device is out of the question. So in practical terms, unless Steam games running on Linux are coded to support the Bluetooth HID mapping directly or via libraries like SDL we should stick to USB rather than bluetooth.

cgutman, 2017-06-25:
@radical-equanimity Actually, that's exactly what you can use for games that don't support this gamepad. Xboxdrv provides an "Xbox 360 wired gamepad" emulation mode where it creates a fake controller that forwards input events from your real gamepad.

Steam can also help here, since it can feed games with SDL mappings for gamepads that the game itself didn't support when it was built via the SDL_GAMECONTROLLERCONFIG environment variable (which you can do by hand with other games too). You can try mapping the gamepad in Steam and seeing whether that fixes your Steam games.

radical-equanimity, 2017-06-29:
Had difficulty mapping bluetooth mode in Steam to match USB mode. I ordered a Steam Link since it's only $15 now on sale. On the Steamlink, bluetooth mode works as I want it to (as if via USB). I'm able to SSH in to the Steam Link and this is the output of dmesg when I connect my controller:

> [   70.841252] input: Xbox Wireless Controller as /devices/virtual/bluetooth/hci0/hci0:1/input3
> [   70.853519] microsoft 0005:045E:02FD.000B: input,hidraw8: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on e0:31:9e:46:bb:90

I don't think the hci0 bit appears when connecting controller via Bluetooth on my Ubuntu machine, and dug further to find this on the filesystem, residing in the file /etc/udev/rules.d/97-bluetooth-hid2hci.rules :

> \# Variety of Dell Bluetooth devices
> \#
> \# it looks like a bit of an odd rule, because it is matching
> \# on a mouse device that is self powered, but that is where
> \# a HID report needs to be sent to switch modes.
> \#
> \# Known supported devices:
> \#   413c:8154
> \#   413c:8158
> \#   413c:8162
> ACTION=="add", ENV{ID_VENDOR}=="413c", ENV{ID_CLASS}=="mouse", ATTRS{bmAttributes}=="e0", KERNEL=="mouse*", RUN+="/usr/sbin/hid2hci --method dell -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> 
> \# Logitech devices
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c703" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c704" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c705" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c70a" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c70b" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c70c" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c70e" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c713" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c714" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c71b" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="046d", ENV{ID_MODEL}=="c71c" RUN+="/usr/sbin/hid2hci --method logitech -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> 
> \# CSR devices (in HID mode)
> ACTION=="add", ENV{ID_VENDOR}=="0a12", ENV{ID_MODEL}=="1000" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="0458", ENV{ID_MODEL}=="1000" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> ACTION=="add", ENV{ID_VENDOR}=="05ac", ENV{ID_MODEL}=="1000" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hci"
> 
> \# CSR devices (in HCI mode)
> \#ACTION=="add", ENV{ID_VENDOR}=="0a12", ENV{ID_MODEL}=="0001" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hid"
> \#ACTION=="add", ENV{ID_VENDOR}=="0458", ENV{ID_MODEL}=="003f" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hid"
> \#ACTION=="add", ENV{ID_VENDOR}=="05ac", ENV{ID_MODEL}=="8203" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hid"
> \#ACTION=="add", ENV{ID_VENDOR}=="05ac", ENV{ID_MODEL}=="8204" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hid"
> \#ACTION=="add", ENV{ID_VENDOR}=="05ac", ENV{ID_MODEL}=="8207" RUN+="/usr/sbin/hid2hci --method csr -v $env{ID_VENDOR} -p $env{ID_MODEL} --mode hid"
Does anybody know more about what's going on here? What is hid2hci doing when I connect the controller?

jahrain, 2017-09-09:
I think I have found a working solution to get my Xbox One S Controller working *properly* with Bluetooth with the latest firmware in Ubuntu 16.04 with the hwe kernel 4.10 installed.

I think I have a basic grasp of what is going on and what is wrong through my fixes are mashup of various kernel patches and work-arounds due to my lack of knowledge in the linux kernel. So far I can tell this that this has nothing to do with the xpad driver and the work-arounds are done with HID and bluetooth drivers patches in the Linux kernel.

So far here is a recollection of issues I have run into and each solution/workaround.
# Controller does not pair properly or randomly disconnects.
  * Fixed by applying a Bluetooth driver kernel patch to l2cap_core from https://patches.linaro.org/patch/85881/. and https://github.com/ValveSoftware/steamlink-sdk/commit/d6afa93787217033a0f0f5db5f557b252c36260e#diff-45ba5a41f43606276c9bf9018eed43c2
 * Another workaround seems to involve the disable_ertm flag.

# Left analog stick moves the mouse. This is probably the most important issue.
 * I haven't fully figured this one out yet. But I found an easy work around from here: https://wiki.archlinux.org/index.php/Gamepad#Disable_joystick_from_controlling_mouse
 * The location to apply the fix in Ubuntu is by creating the file here instead /usr/share/X11/xorg.conf.d/51-joystick.conf

# Controller is detected as a generic HID device.
  I With the latest firmware installed on the controller, seems like the device IDs have been changed. Before they used to be 045E:02E0, now it is reported as 045E:02FD after a firmware update that fixed compatibility with Windows 10 Anniversary update. This breaks many existing fixes and work-arounds, including hid kernel driver patches from Valve that have been available on the internet from like a year ago. 
 I This was fixed by patching the drivers/hid/hid-ids.h changing the USB_DEVICE_ID_MS_XBOX_ONE_S definition from 02e0 to 02fd. 

# Incorrect button mapping and Back and Home buttons don't work. 
  I Seems like when connected by bluetooth on linux, the controller sends some weird HID when debugging the HID drivers and looking at wireshark captures of the bluetooth traffic.
  I Reports "AC Home" or the homepage consumer usage page HID (0x000c0223) when pressing the home key. Most stuff expect the BTN_MODE input event for the home button, not KEY_HOMEPAGE which is how the HID drivers report that HID.
 * Reports "AC Back" or the browser back button consumer usage page HID (0x000c0224) when pressing the home key. Most stuff expect the BTN_SELECT input event for the back button, not KEY_BACK which is what the HID drivers report for HID.
* I fixed this with a patch adding some additional input key remapping to the hid-microsoft kernel module.

# I noticed a few other strange quirks in the key mappings though they don't seem to hamper usage of the controller in games. 
* The controller reports Simulation HIDs for the L and R triggers. It reports Brake HID (0x000200c5) and Gas HID (0x000200c4), while at the same time it uses the standard L Trigger (Z Axis) HID (0x00010032)  to represent the Right Analog X Axis. and the standard R Trigger (RZ Axis) HID (0x00010035) to represent the Right Analogy Y Axis. I can't imagine the horrors of game developers when seeing this axis mapping mixup. I think Steam has already implemented work-arounds for these since if I try and change the mappings, it makes Steam confused, but other tools like jstest-gtk seem to be no longer confused.

Anyways, I have a gist here of all the driver patches I applied, including ones from Valve and my own here: https://gist.github.com/jahrain/2006b6cd3d5bbf9d39fd45f20abbe9c9

It would be great if someone who actually knew what they were doing could get an actual non-hack solution implemented and patched upstream.

Enigma0, 2017-09-10:
I think this actually used to just work in ~16.04 but having same issue in 17.04. Tried `disable_ertm` with no success. Somehow 4 instances of an xbox controller device were created and were breaking controller support for Steam games in general.

rootyElf, 2017-11-04:
I’d just like to share my observations.

I also had this issue with messed up button mapping. The back key was not registered as gamepad button but rather as back key like you would have on a mouse. So for example it causes the web browser to go one page back. The Home button is not recognized either. It still seems to work though → when I press the home button in steam it opens Big Picture Mode. I don’t know if it’s a bug or intended.

On my computer I have both Arch Linux and Windows 10 installed. I used this method to dual pair it with both OSes: https://askubuntu.com/a/741821 What I found out is that the button mapping is different depending on which OS the controller was paired with last.

When I pair the controller with Linux I get the following button layout:
A -> B0
B -> B1
X -> B3
Y -> B4
LB -> B6
RB -> B7
START -> B11
L3 -> B13
R3 -> B14

LS -> A0 & A1
RS -> A2 & A3
RT -> A4
LT -> A5
DPAD -> A6 & A7

that’s probably the result most of you get. Now, when I pair the controller with Windows 10 and then copy the link key over to Linux I get a different button layout (in Linux)

A -> B0
B -> B1
X -> B2
Y -> B3
LB -> B4
RB -> B5
BACK -> B6
START -> B7
L3 -> B8
R3 -> B9

LS -> A0 & A1
RS -> A2 & A3
RT -> A4
LT -> A5
DPAD -> A6 & A7

There’s also button 10 – 14 which don’t seem to do anything. 
This should mostly be the same as the on in Windows and it also looks a bit more sane (imho – except for button 10 - 14). The thing with the home button is still the same but on Windows it’s also not shown as button and as I said – it somehow still works at least for steam. 

So the button issue might depend on the state the controller is in (so if it either had normal pairing or special Windows 10 pairing) rather than on how the computer handles the device. So I think it is the pairing process which might need some tweaking. 

I don't know if this was addressed before. If not maybe someone might try to confirm/reproduce this.

Btw. If you do the reverse thing (pair the controller on Linux and then use it on Windows) you get quite a funny result. I don’t know how to explain it, but it’s definitely not working properly.

jahrain, 2017-11-05:
The controller seems to have some internal state that switches to a special "Android Controller Mode" when paired with Linux, and switches back to normal "Xbox One S Controller" mode when paired with Windows.

In the "Android Controller Mode", the back button gets mapped to the "back keyboard button" and the xbox button gets mapped to the "home keyboard button". These keys are used to navigate the "home" and "back" Android UI function.

kakra, 2018-01-06:
I now managed to pair the controller via bluetooth by applying and crafting my own kernel patch collected from different sources (SteamOS kernel and @jahrain patch). The select and home buttons now correctly report as button events in jstest, additionally to sending a key XF86_Back und XF86_Home in xev. I think that's fine so far.

But still, in Steam games the select button does not work, it is simply ignored. Any ideas?

atar-axis, 2018-02-20:
take a look at https://github.com/atar-axis/xpadneo, which supports

* rumble over BT, it
* patches the ERTM and
* solves the hid-generic problem

I am also working on a correct mapping for all models and firmware versions.
I would appreciate any support (especially on the mapping and integration into other distributions).

[33 more replies]

closed 2022-09-08 by paroj