paroj/xpad #54: The wrong product ID is exposed for the Wireless 360 controllers

Omar007, 2016-11-23:
The current implementation exposes the wireless receiver itself as 4 controllers, however the receiver and controllers are actually different devices with their own product ID.

The receiver has a product ID of 0x0719 and is a hub-style device that has 4 controllers attached to it.
The controllers have a different product ID; 0x02A1.

Right now you get 4 controllers with a product ID of 0x0719.
Instead, the receiver should be an unique device with 4 children; receiver (0x0719) -> 4 controllers (0x02A1).

Because the wrong PID is advertised by the drivers, some games do not automatically pick up the controller as an Xbox controller.
This was discovered through a game running in WINE [1].

NOTE: This potentially also applies to the Xbox receiver (PID 0x0291) but I can't personally verify this.
NOTE2: Bug was originally reported on the kernel.org bugzilla [2] but received to response for 3 months.

[1]: https://bugs.winehq.org/show_bug.cgi?id=39891
[2]: https://bugzilla.kernel.org/show_bug.cgi?id=153131

paroj, 2016-11-24:
does the above commit help? (x360w_id branch)

cgutman, 2016-11-25:
FWIW, Windows also exposes wireless Xbox 360 controllers with product ID 0x02A1.

paroj, 2016-11-25:
yeah, from the Wine Bugreport:

> Windows:
TL/DR; Both the receiver and controllers are unique devices.
> - Receiver
-- Separate device, attached to a USB port, hub-style device ('Xbox 360 peripheral')
-- Product ID 0x0719
> - Controllers (x4)
-- Separate device, attached to the receiver, game controller
-- Product ID 0x02A1
> 
> Linux:
TL/DR; The receiver and controllers are the same device (receiver = 4 controllers).
> - Receiver
-- Attached to a USB port, seen as 4 game controllers.
-- Product ID 0x0719.
> - Controller (4x)
-- Not handled as a separate device.
-- Should have Product ID = 0x02A1.
-- In the system tree the controllers are located underneath the receiver:
   /sys/bus/usb/devices/[receiver usb]/[receiver-usb]:[controller]

Omar007, 2016-11-26:
It does indeed help :)
If there are wireless controllers with an ID other than 0x02A1 this might have to be revisited to account for that but for now this seems to do it just fine.

cgutman, 2016-12-30:
This patch is queued for upstream so this issue can probably be closed now.

closed 2016-12-30 by paroj