paroj/xpad #55: Error when installing on Antergos

Hungry-Hobo, 2016-11-30:
I tried to install the driver doing what the directions said, and got this error message when I typed "sudo dkms install -m xpad -v 0.4":
`Creating symlink /var/lib/dkms/xpad/0.4/source ->`
                           `/usr/src/xpad-0.4`
`DMKS: add completed`
`Error! echo`
`Your kernel headers for kernel 4.8.10-1-ARCH cannot be found at`
`/usr/lib/modules/4.8.10-1-ARCH/build or /usr/lib/modules/4.8.10-1-ARCH/source.`

cgutman, 2016-11-30:
4.8 is pretty new. I don't think anything substantial is present in here that isn't upstream yet. What are you trying to fix by using this driver?

Hungry-Hobo, 2016-11-30:
When I plug in an Xbox One controller it turns on but won't register any button presses in any of the games I tried. I thought using this driver would fix the problem.

cgutman, 2016-11-30:
Can you post the output of `lsusb -v` for your controller? And also `dmesg` when plugging in the controller?

paroj, 2016-11-30:
regarding your installation error: you are missing the kernel headers.

paroj, 2017-03-29:
no response by submitter

closed 2017-03-29 by paroj