paroj/xpad #56: Xbox one controler

futex, 2016-12-19:
I have xbox one wireless adapter controler and pc wireless adapter, but the adapter seem not running (no led blink), in dmesg i have:

`[13825.993241] usb 1-1.5.4: new high-speed USB device number 9 using ehci-pci
[13826.088305] usb 1-1.5.4: New USB device found, idVendor=045e, idProduct=02e6
[13826.088310] usb 1-1.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[13826.088312] usb 1-1.5.4: Product: XBOX ACC
[13826.088314] usb 1-1.5.4: Manufacturer: Microsoft Inc.
[13826.088316] usb 1-1.5.4: SerialNumber: 112437
`
The xpad module is running, what i have to do to use the xbox in wireless mode?

paroj, 2016-12-20:
the wireless adapter is currently not supported. the progress is tracked in #14. Do not expect it to be working anytime soon though.

closed 2016-12-20 by paroj

futex, 2016-12-20:
I saw some guys used a bluetooth controller with the xbox one S controller is it possible with the xbox one controller? I saw nothing with mine

devkid, 2016-12-20:
@futex No, the "normal" XBox One Wireless Controller is using 802.11 Wifi, not Bluetooth.