paroj/xpad #57: Titanfall 2 X-box One controller unsupported

powersurge360, 2016-12-20:
Hey! I hope I've captured enough info to be useful, but please let me know if you need more.

Here's what `lsusb -v` has to say about the device.

```
Bus 002 Device 006: ID 0e6f:0165 Logic3 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x0e6f Logic3
  idProduct          0x0165 
  bcdDevice            1.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           64
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e0  1x 224 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0080  1x 128 bytes
        bInterval               1
```

I was optimistic that I could simply add the product to the `xpad_device` array and specify that it is an xbox one controller but no dice. The controller is recognized but the inputs are not. Let me know anyway I can help, I'd like to get this guy set up as soon as possible!

powersurge360, 2016-12-20:
Additional info from `dmesg`

```
[21816.892128] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input37
[21816.892247] xpad 2-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[22472.601581] usb 2-1: USB disconnect, device number 19
[22472.601645] xpad 2-1:1.0: xpad_irq_in - urb shutting down with status: -108
```

powersurge360, 2016-12-21:
Been debugging this a little on my own, although I'm way over my head. I turned on verbose debugging and it looks like it's sending the following two lines over and over again:

```
[23378.120220] xpad-dbg: 00000000: 02 20 21 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[23378.120223] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Is this an attempt at a handshake?

dantob, 2016-12-27:
`[21816.892128] input: Generic X-Box pad`
If you've added the id before this, it doesnt seem to have worked correctly. Packets starting with `02` supposedly indicate 'waiting for connection' but xpad doesn't do anything with them.

Try this, or compare it to what you have. It's just adding the id for your controller. https://github.com/dantob/xpad/commit/0098e8af630614843451cc12d20cdc75f62cd3b0

powersurge360, 2016-12-27:
Yea I had added it correctly but in the info I had posted, I had already begun the debugging process and was trying to follow the detection code for generic pads. Neither path bore fruit, however, and both recognize the controller but does not accept inputs.

To be more add some detail, I tried adding my line both with Xbox one and Xbox 360 types manually specified (and verified the string was present in jstest) and additionally reverted those changes and found no discernible difference between the three.

dantob, 2016-12-27:
Try this commit https://github.com/dantob/xpad/commit/fd9d3d906e74ef8a480ce5aa3ecfda2247b20993 and make sure the device is being sent the initialize packet;  `05 20 00 01 00`
```
[208961.704936] usb 1-6: new full-speed USB device number 26 using xhci_hcd
[208961.837074] input: Microsoft X-Box One pad (Firmware 2015) as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/input/input79
[208961.837321] xpad 1-6:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[208961.839044] xpad-send: 00000000: 05 20 00 01 00 0f 00 00 00 00 ff 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[208961.851003] xpad-recv: 00000000: 03 20 07 04 8b 01 00 00 00 00 9e 05 07 01 f0 fc 1f fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[208961.851004] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[208961.855035] xpad-recv: 00000000: 20 00 43 0e 00 00 00 00 00 00 8c 05 5b 00 f0 fc 57 fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[208961.855036] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

powersurge360, 2016-12-30:
Sorry for the delay on this, I spent the last few days half dead with a sinus infection! I just pulled down your fork and recompiled and this is the relevant dmesg output:

```
[18728.003522] xpad-send: 00000000: 05 20 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18729.211640] xpad-recv: 00000000: 02 20 01 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18729.211644] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18729.803643] xpad-recv: 00000000: 02 20 02 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18729.803647] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18730.391633] xpad-recv: 00000000: 02 20 03 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18730.391637] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18730.979653] xpad-recv: 00000000: 02 20 04 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18730.979662] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18731.567622] xpad-recv: 00000000: 02 20 05 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18731.567625] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18732.155619] xpad-recv: 00000000: 02 20 06 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18732.155623] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18732.743513] xpad-recv: 00000000: 02 20 07 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18732.743516] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18733.331608] xpad-recv: 00000000: 02 20 08 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18733.331612] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18733.919609] xpad-recv: 00000000: 02 20 09 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18733.919613] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[18734.511603] xpad-recv: 00000000: 02 20 0a 1c 08 fc 25 3e 92 23 00 00 6f 0e 65 01 01 00 01 00 09 00 47 00 01 00 01 00 01 00 01 00
[18734.511607] xpad-recv: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Still no change in the behavior though. The controller is recognized but inputs are ignored: 
[image]

powersurge360, 2016-12-30:
After the initial send line, it just repeats the other two lines over and over ad infinitum.

cgutman, 2016-12-30:
@powersurge360 Can you give this xpad.c a shot? http://pastebin.com/raw/xUPqtvBa

Apologies in advance for not pushing this up to GitHub properly yet. I've added a second init packet that the macOS Xbox Driver project 360Controller uses.

powersurge360, 2016-12-30:
This corrected the issue for me! Woot woot. Is there anything I ought to turn off for this particular file? Is it in debug mode or in any other way unstable that I should be concerned about?

cgutman, 2016-12-30:
Are you using the v2 link or did you get the original (before I edited it)? Do you see anything nasty in dmesg?

powersurge360, 2016-12-30:
I used first one and then the other. I don't notice anything awry in dmesg for either file.

cgutman, 2016-12-31:
Then you should be fine. It's the current xpad.c queued for 4.10-rc2 plus that init change. I'll prepare a patch for upstream.

powersurge360, 2016-12-31:
Thanks for debugging my hardware you two! I was so heartbroken that my nice new controller wasn't working A+ open sourcing :)

paroj, 2016-12-31:
hmm.. we already had that packet once in #20 - go figure..

cgutman, 2017-01-01:
@powersurge360 Can you confirm that your pads still work with this xpad.c? http://pastebin.com/raw/Re8xby26

paroj, 2017-01-03:
@cgutman could you please in future provide test code via a fork of this repo where one can see the diff?
Right now we are teaching people to compile kernel modules from pastebin, which is not a good thing..

cgutman, 2017-01-03:
@paroj I agree I was being a bit lazy. I've setup a fork with the code - https://github.com/cgutman/xpad/tree/3rdparty

cgutman, 2017-01-23:
@powersurge360 If you're still willing to test, can you confirm that the issue remains fixed on my "for_upstream" branch? https://github.com/cgutman/xpad

If that works, may I add a "Tested-By: Firstname Lastname Email address" from you for my submission upstream?