paroj/xpad #59: Hori Horipad - Non-repsonsive analog sticks

orbea, 2017-01-01:
OS: `Slackware64-current`
`Linux-4.9.0`

I got a new Hori Horipad today and found that neither the left or right analog stick do anything, moving them in any direction is not recognized at all. However the L3, R3, and all of the other buttons work as expected. This has been confirmed in sdl1-jstest, sdl2-jstest and RetroArch. Is this a problem with the hardware, any firmware or in the xpad driver? I can recompile my kernel or the xpad module to test any patches, thanks!

Here are some information that hopefully helps, if there is anything more I can provide please let me know.

```
$ lsusb | grep -i hori
Bus 008 Device 029: ID 0f0d:0067 Hori Co., Ltd
```
```
$ usb-devices
T:  Bus=08 Lev=02 Prnt=02 Port=02 Cnt=03 Dev#= 16 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=ff Prot=ff MxPS=64 #Cfgs=  1
P:  Vendor=0f0d ProdID=0067 Rev=01.01
S:  Manufacturer=HORI CO.,LTD.
S:  Product=HORIPAD ONE
S:  SerialNumber=00005DC471C67F48
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=47 Prot=d0 Driver=xpad
```
/sys/kernel/debug/usb/devices
```
T:  Bus=08 Lev=02 Prnt=02 Port=02 Cnt=03 Dev#= 38 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=ff Prot=ff MxPS=64 #Cfgs=  1
P:  Vendor=0f0d ProdID=0067 Rev= 1.01
S:  Manufacturer=HORI CO.,LTD.
S:  Product=HORIPAD ONE
S:  SerialNumber=00005DC471C67F48
C:* #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=47 Prot=d0 Driver=xpad
E:  Ad=01(O) Atr=03(Int.) MxPS=  64 Ivl=4ms
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=4ms
```

orbea, 2017-01-01:
`xpad-2016.12.31_bd4f517_4.9.0_master-x86_64-1_git`

I compiled the current git master and pleasantly found that this has already been fixed, thanks! I will close this now.

If you don't mind me asking, when is the current master supposed to be merged into the kernel?

closed 2017-01-01 by orbea

cgutman, 2017-01-01:
Interesting. I guess it must be bd4f51797c2a518be90b470d499059dc088ddbd3. I was hoping to see if we could also get #48 fixed (likely with additional init packets) before going upstream with the patch. If we have to send more than 2 init packets, we'll need to do sequencing that would require rewriting the majority of the patch.

orbea, 2017-01-01:
Thanks for the info, I'm more than happy to wait until you think its ready for it to be merged upstream. Also that is pretty good luck on my part that it was fixed the day before it came in the mail!

cgutman, 2017-01-01:
@orbea Can you try with this xpad.c? http://pastebin.com/raw/Re8xby26

orbea, 2017-01-01:
It at least does not happen nearly as fast with the master, I will continue testing.

cgutman, 2017-01-23:
Thanks, can you test my "for_upstream" branch? https://github.com/cgutman/xpad

If that works, may I add a "Tested-By: Firstname Lastname <email>" from you for my submission upstream?

orbea, 2017-01-24:
@cgutman I tried your branch, it had the same result as before. The analog sticks work at first, but then quickly stop responding.

cgutman, 2017-01-24:
@orbea Ah, I think I understand the issue. Let me fixup my branch.

cgutman, 2017-01-24:
@orbea Please try again with my "for_upstream" branch. I've amended the patches with a fix for your issue.

Each time the device was opened, it was starting over from 0 in the sequence numbers in each packet. Your gamepad was rejecting the initialization packets after the first open because it wasn't expecting sequence numbers to start from 0 again.

orbea, 2017-01-25:
@cgutman I have tried your branch again, this time the initial problem is gone.

However for reference this controller still has these two issues. https://github.com/paroj/xpad/issues/61 https://github.com/paroj/xpad/issues/62