paroj/xpad #60: Xbox one controller not sending data

AdamAbonyi, 2017-01-01:
Hi,
I am sorry to bother, I am not sure where else to write. In the past i have successfully got my xbox one controller running.  Just for info I am running it with moonlight-embedded on osmc. 
I tried to recompile the module for the new kernel and run into some issues that i don't know how to tackle. I have the ordinary xbox one controller that came with the xbox one bundle. 

My controller lights up as usual but I get no button presses in the system. jstest /dev/input/js0 returns  [http://pastebin.com/fJtZMx6C](url). Pressing any buttons doesn't do anything on this screen, it stays the same no mater what I do with the controller.

using fftest /dev/input/event2 allows me to make the controller react and rumble.

Debug information as described on the mainpage also yields no result. I have no xpad-dbg messages or anything similar. I don't know how to enable verbose logging like it was described in [https://github.com/paroj/xpad/issues/57](url).

Could someone please help an advise me what im doing wrong or how i can enable verbose debugging to add more information here? 

Thank you.

exentio, 2017-01-09:
I have the same problem, I got it to work, but when I rebooted it didn't work anymore. I was using this when it worked: https://github.com/cgutman/xpad/tree/3rdparty

cgutman, 2017-01-11:
Please try again with the "powera" branch of my repo. https://github.com/cgutman/xpad/tree/powera

AdamAbonyi, 2017-01-11:
I will try once i get home and have some time today. Thank you in advance, I will let you know how it went.

AdamAbonyi, 2017-01-11:
Once again thanks for your time. I have tried your branch and the result is basically the same. The controller powers up and i can get it to vibrate but i dont get any input from it. 
Does anyone know, how i can enable verbose debugging to see the communication between the controller and the xpad driver? 
I have successfully got my xbox one gamepad working on a previous version of my kernel :-(
Any ideas?