paroj/xpad #61: Hori Horipad - Non-repsonsive after savestates in n64 emulator.

orbea, 2017-01-10:
OS: `Slackware64-current`
`RetroArch-2017.01.09_338980ef2_master-x86_64-1_git`
`GLupeN64-2017.01.09_799d362_master-x86_64-1_git`
`xpad-2016.12.31_bd4f517_4.9.0_master-x86_64-1_git`

Most of the time when creating or loading a savestate in the n64 emulator GLupeN64 using the RetroArch frontend the gamepad will be non-responsive for several seconds where I will have to press buttons until it wakes up again. This used to work in this emulator until this commit https://github.com/GLupeN64/GLupeN64/commit/2f21040d7658ef70ec95d46c33ec97217451a6eb and might again if the save state code was made more efficient and didn't create temp files. However it already seems to work fine with any other gamepad such my my ps3 gamepad connected to the same usb port or my keyboard. So this might also be possible to improve in the xpad driver? Especially given that this gamepad was only recently fixed as discussed in issue https://github.com/paroj/xpad/issues/59.

Here is an related issue for the GLupeN64 repo.
https://github.com/GLupeN64/GLupeN64/issues/157

And for reference the RetroArch repo.
https://github.com/libretro/RetroArch

paroj, 2017-01-11:
it might be that the other gamepads just have different power requirements. What kind of machine is the pad attached to?

orbea, 2017-01-11:
Its a desktop computer, the motherboard is a gigabyte 990FXA-UD5 R5.

orbea, 2017-01-12:
Is there anything more I could try to help debug this further? I'm bit at a loss at what would be helpful. :)

paroj, 2017-01-12:
the log you posted in GLupeN64/GLupeN64#157 looks unsuspicious as does the linked commit. So I do not really know what might cause this..

orbea, 2017-01-17:
That is understandable and thanks for your time.

After making this issue I noticed this will only happen if I use the `Save State` and `Load State`  options in the RetroArch quick menu while it works fine with the corresponding gamepads, `F2` (Save State) and `F4` (Load State). This makes me wonder if there is also something in RetroArch itself that may be leading to this, but they are currently working feverishly on the new `1.4.0` release so something that seems to affect only one gamepad may need to wait. :)

orbea, 2017-01-30:
This was fixed in this GLupeN64 commit https://github.com/GLupeN64/GLupeN64/commit/4fc82bfc9f72e213eff988f1d9107081e9647910 so I will close this now. If anyone is interested in fixing this further in xpad I would suggest trying older commits of GLupeN64 where this was a problem, but since I do not know any other emulators with a similar problem it might not be worth much effort.

closed 2017-01-30 by orbea