paroj/xpad #62: xone: Hori Horipad - Rumble gets stuck

orbea, 2017-01-17:
OS: `Slackware64-current`
`Linux-4.9.3`
`xpad-2016.12.31_bd4f517_4.9.0_master-x86_64-1_git`

Once in a while when using the horipad one gamepad rumble will get stuck (Input is frozen) and continue vibrating for several seconds, once it stops analog and button input will be partially or entirely non-responsive until I unplug and re-plug the controller. I've experienced this with mupen64plus-libretro + RetroArch, mupen64plus (Upstream) and pcsx2.

One potentially easy way to reproduce it:
1. Install RetroArch + mupen64plus-libretro and find the game Zelda: Ocarina of Time
2. Start the game and enable the rumble pak in the core options.
3. Repetitively attack the wall for a few minutes until the rumble gets stuck.

Here is a save that makes that easier to test.
http://ks392457.kimsufi.com/orbea/stuff/games/n64/saves/Zelda%20OoT%2064.srm.xz

After this happens I noticed this in dmesg.
```
[50047.400038] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 1
[50052.729005] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 1
[50052.745006] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 1
[50052.829132] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 1
[50052.844507] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 1
[50052.847023] xpad 8-1.3:1.0: xpad_irq_in - nonzero urb status received: -71
[50067.722980] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.732940] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.742992] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.753083] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.762893] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.775992] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.785993] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.795993] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.806058] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.815995] xpad 8-1.3:1.0: xpad_irq_out - nonzero urb status received: -71
[50067.817850] usb 8-1.3: USB disconnect, device number 22
[50067.818368] xpad 8-1.3:1.0: xpad_irq_out - urb shutting down with status: -108
[50067.818437] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -108
[50068.245792] usb 8-1.3: new full-speed USB device number 23 using xhci_hcd
[50068.329628] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[50068.329634] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[50068.329637] usb 8-1.3: Product: HORIPAD ONE
[50068.329640] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[50068.329642] usb 8-1.3: SerialNumber: 00005DC471C67F48
[50068.332377] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input38
[50068.332582] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[50068.337736] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
```

loganmc10, 2017-01-18:
@orbea in regards to the other issue you mentioned to me on IRC (rumble stopping/dying after a few minutes in GLupeN64).

I tested Ocarina of Time with an official Xbox One controller. I pounded on the wall with the sword for probably 3-4 minutes and the rumble never stopped.

orbea, 2017-01-18:
It might be connected to this issue then, but with a different symptom? In short instead of the rumble getting stuck on it just stops and the gamepad continues to work without it. Though this has only happened in GLupeN64 for me.

paroj, 2017-01-18:
for reference you are getting [EPROTO (-71)](http://www.virtsync.com/c-error-codes-include-errno) which [might mean this](http://www.makelinux.net/ldd3/chp-13-sect-3).

orbea, 2017-01-25:
I've noticed that if I wait longer, the controller will eventually disconnect and then reconnect, but the analog sticks will be non-functional.

orbea, 2017-01-25:
Additionally I can no longer reproduce the different behavior with GLupeN64, it acts much like the other emulators now. Instead of loosing the rumble it looses the analog sticks. I'm not sure if this was a change between 4.9 and 4.10 kernels, RetroArch or in GLupeN64 itself. I will have to test more later.

orbea, 2017-02-17:
I can confirm this also happens with dolphin-emu. This led me to notice that if I reproduce the bug I will lose analog sticks until I unplug the controller, but this bug also vanishes until the controller is unplugged as both the rumble and dpad still work.

orbea, 2018-07-25:
I now use dualshock3 controllers without any issue, but I can reproduce this issue with the ds3 if they have a near 0 battery charge. It works fine as long as they are charged properly.

My hypothesis is that xpad is not giving the controllers enough power for the rumble to work correctly.

closed 2020-06-24 by orbea