paroj/xpad #66: Xbox button works only once on Microsoft X-Box One S pad

oprypin, 2017-03-20:
Hi, I run `Linux 4.10.3-1-ARCH x86_64` (I never installed xpad myself) and I have a recent X-Box One gamepad, which shows up like this:

```console
$ cat /proc/bus/input/devices
I: Bus=0003 Vendor=045e Product=02ea Version=0301
N: Name="Microsoft X-Box One S pad"
P: Phys=usb-0000:03:00.0-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:1c.0/0000:03:00.0/usb3/3-1/3-1:1.0/input/input28
U: Uniq=
H: Handlers=event15 js0 
B: PROP=0
B: EV=20000b
B: KEY=7cdb000000000000 0 0 0 0
B: ABS=3003f
B: FF=107030000 0
```

When I press the Xbox button (the one lit up in the middle), I get only one "down" event, and then the button is stuck in that state, never producing any events until I reconnect the gamepad.

```console
$ jstest /dev/input/js0
Axes:  0:     0  1:     0  2:-32767  3:     0  4:     0  5:-32767  6:     0  7:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:on   9:off 10:off 
```

My naive attempts at debugging haven't shown any results at a level which is lower than what jstest shows. I do not see xpad events in the kernel log, probably because it's the one provided with the kernel.

cgutman, 2017-03-20:
I submitted a fix for this upstream for 4.11.

https://github.com/torvalds/linux/commit/57b8443d3e5bd046a519ff714ca31c64c7f04309

oprypin, 2017-03-20:
Thank you very much. I confirm that the upstream fix works.

closed 2017-03-20 by oprypin