paroj/xpad #67: Rumble Support

Noah-Steinberg, 2017-03-31:
Is there any rumble support for any controllers using xpad?
I am currently working on my own driver that supports rumble for the microsoft xbox one controllers, and if it is wanted, I could look at doing it in xpad as well. If I had the time, I could look into other controllers as well.

cgutman, 2017-03-31:
Yes, rumble is supported for all gamepads that xpad supports: Xbox, Xbox 360 (Wired and Wireless), and Xbox One (Wired).

See: https://github.com/paroj/xpad/blob/master/xpad.c#L1013

Noah-Steinberg, 2017-03-31:
Ah, I see, I had not looked closely enough at xpad. Thanks!

Noah-Steinberg, 2017-03-31:
Is there any wish for more rumble support than a single packet type with strong/weak?

cgutman, 2017-03-31:
You mean support rumble on the triggers? or something else?

Supporting trigger rumble is trivial from an xpad perspective, but I couldn't find a sensible way to map to the force-feedback API in the kernel.

Noah-Steinberg, 2017-03-31:
I am currently looking into any way to do it. The only issue is having to construct the packet byte by byte. So far I have support for the different kinds of rumble that the kernel specifies, but in very primitive and mostly hard coded ways. Also, Im getting all my information from the Xbox One Controller Protocall