paroj/xpad #68: Some new 360w controllers not able to pair

rigoleto, 2017-04-14:
I have 4 new 360 wireless controllers, but interestingly only one of them is able to pair with this xpad module loaded. All of genuine Microsoft controllers bought from the same Amazon seller.

Here's the debug output from the good one:
```
[ 7491.454661] usb 2-2.4: new full-speed USB device number 27 using xhci_hcd
[ 7491.574126] usb 2-2.4: New USB device found, idVendor=045e, idProduct=0719
[ 7491.574128] usb 2-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7491.574129] usb 2-2.4: Product: Xbox 360 Wireless Receiver for Windows
[ 7491.574130] usb 2-2.4: Manufacturer: ©Microsoft
[ 7491.574131] usb 2-2.4: SerialNumber: E108F6F0
[ 7491.575323] xpad 2-2.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[ 7491.575429] xpad 2-2.4:1.2: xpad_prepare_next_out_packet - found pending output packet 0
[ 7491.575523] xpad 2-2.4:1.4: xpad_prepare_next_out_packet - found pending output packet 0
[ 7491.575630] xpad 2-2.4:1.6: xpad_prepare_next_out_packet - found pending output packet 0
[ 7491.580524] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580526] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580532] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580533] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580536] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580537] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580539] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7491.580541] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.976621] xpad-dbg: 00000000: 08 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.976623] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.976737] xpad 2-2.4:1.0: xpad_prepare_next_out_packet - found pending output packet 2
[ 7492.976812] input: Xbox 360 Wireless Receiver as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.4/2-2.4:1.0/input/input26
[ 7492.979557] xpad-dbg: 00000000: 00 0f 00 f0 00 cc e1 08 f6 f0 b6 10 8a b0 00 20 13 e3 20 1d 30 03 40 01 50 01 ff ff ff 00 00 00
[ 7492.979559] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.980555] xpad-dbg: 00000000: 00 01 00 13 e2 13 00 00 00 00 00 f6 f0 f8 7e f0 ac f4 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.980556] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.987569] xpad-dbg: 00000000: 00 01 00 00 20 13 00 00 00 00 00 f6 f0 f8 f7 f0 29 f4 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.987571] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.995603] xpad-dbg: 00000000: 00 00 00 13 e2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7492.995610] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7493.003584] xpad-dbg: 00000000: 00 00 00 20 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7493.003585] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7493.011620] xpad-dbg: 00000000: 00 00 00 30 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7493.011622] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7493.019563] xpad-dbg: 00000000: 00 00 00 40 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7493.019564] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

And the bad one (missing the 5th packet starting with `08 80`):
```
[ 7369.682191] usb 2-2.4: new full-speed USB device number 26 using xhci_hcd
[ 7369.802206] usb 2-2.4: New USB device found, idVendor=045e, idProduct=0719
[ 7369.802209] usb 2-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7369.802212] usb 2-2.4: Product: Xbox 360 Wireless Receiver for Windows
[ 7369.802214] usb 2-2.4: Manufacturer: ©Microsoft
[ 7369.802216] usb 2-2.4: SerialNumber: E108F6F0
[ 7369.803376] xpad 2-2.4:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[ 7369.803821] xpad 2-2.4:1.2: xpad_prepare_next_out_packet - found pending output packet 0
[ 7369.804203] xpad 2-2.4:1.4: xpad_prepare_next_out_packet - found pending output packet 0
[ 7369.804577] xpad 2-2.4:1.6: xpad_prepare_next_out_packet - found pending output packet 0
[ 7369.807165] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807166] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807170] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807171] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807173] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807173] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807175] xpad-dbg: 00000000: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 7369.807176] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Do you have an idea of what's going on here?

paroj, 2017-04-15:
do they work in windows? the missing packet contains the information that there is actually a pad connected, see:
https://github.com/paroj/xpad/blob/master/xpad.c#L574

rigoleto, 2017-04-15:
They do work on Windows, and I was able to get them going on Linux too. My error was in not holding down the pairing buttons long enough. For whatever, one of the controllers works without that, but the other ones required holding both the receiver and controller pairing buttons for longer than you might expect. I hope someone else can learn from my noobishness. Closing this.

closed 2017-04-15 by rigoleto