paroj/xpad #69: xone: Hori Horipad - Non-repsonsive analog sticks after pm-suspend

orbea, 2017-04-25:
OS: `Slackware64-current`
`xpad-2017.03.10_d254f94_4.10.12_master-x86_64-1_git`
`linux-4.10.12`

When resuming my computer after `pm-suspend`, the analog sticks on my xpad gamepad fail to respond. (See issue https://github.com/paroj/xpad/issues/59)

This was previously working with linux-4.10.0 so I at first I thought this was the new commit (d254f94), but when testing the previous commit (01f149b) that was working for me the issue persisted so my guess this regressed elsewhere in the kernel? Additionally, the xpad driver provided with 4.10.* seems to still suffer from the original issue.

Dmesg will produce this output with either commit after booting or resuming the computer.
```
[40198.509324] ------------[ cut here ]------------
[40198.509329] WARNING: CPU: 2 PID: 1187 at /tmp/SBo/xpad/xpad.c:956 xpadone_send_init_pkt+0xd5/0xe0 [xpad]
[40198.509330] Modules linked in: xpad(O) lzo zram zsmalloc cpufreq_ondemand fuse hid_microsoft joydev evdev mxm_wmi snd_hda_codec_hdmi kvm_amd kvm snd_hda_codec_realtek snd_hda_codec_generic irqbypass snd_hda_intel snd_hda_codec crct10dif_pclmul crc32_pclmul nouveau snd_hda_core video ttm drm_kms_helper drm agpgart fb_sys_fops syscopyarea sysfillrect sysimgblt snd_hwdep i2c_algo_bit snd_pcm fam15h_power k10temp hwmon crc32c_intel snd_timer r8168(O) ghash_clmulni_intel tpm_infineon button wmi snd soundcore acpi_cpufreq tpm_tis tpm_tis_core tpm loop [last unloaded: xpad]
[40198.509357] CPU: 2 PID: 1187 Comm: acpid Tainted: G        W  O    4.10.12 #1
[40198.509358] Hardware name: Gigabyte Technology Co., Ltd. To be filled by O.E.M./990FXA-UD5 R5, BIOS F3 04/01/2015
[40198.509359] Call Trace:
[40198.509364]  dump_stack+0x4d/0x64
[40198.509366]  __warn+0xdd/0x100
[40198.509367]  warn_slowpath_null+0x1d/0x20
[40198.509369]  xpadone_send_init_pkt+0xd5/0xe0 [xpad]
[40198.509370]  xpad_start_xbox_one+0x1e/0x40 [xpad]
[40198.509372]  xpad_start_input+0x4b/0x70 [xpad]
[40198.509373]  xpad_open+0x15/0x20 [xpad]
[40198.509376]  input_open_device+0x8e/0xc0
[40198.509378]  joydev_open+0x17c/0x193 [joydev]
[40198.509381]  chrdev_open+0xa8/0x1c0
[40198.509383]  do_dentry_open+0x21a/0x300
[40198.509384]  ? cdev_put+0x30/0x30
[40198.509386]  vfs_open+0x51/0x80
[40198.509388]  ? may_open+0x9b/0x100
[40198.509389]  path_openat+0x297/0x1330
[40198.509391]  ? check_preempt_curr+0x7b/0x90
[40198.509393]  ? ttwu_do_wakeup+0x19/0x100
[40198.509395]  do_filp_open+0x81/0xe0
[40198.509397]  ? __mutex_unlock_slowpath.isra.6+0xa8/0xf0
[40198.509399]  ? __alloc_fd+0xae/0x170
[40198.509401]  do_sys_open+0x126/0x200
[40198.509403]  SyS_open+0x1e/0x20
[40198.509404]  entry_SYSCALL_64_fastpath+0x17/0x98
[40198.509406] RIP: 0033:0x7fec8beba5e0
[40198.509407] RSP: 002b:00007ffe429938f8 EFLAGS: 00000246 ORIG_RAX: 0000000000000002
[40198.509408] RAX: ffffffffffffffda RBX: 000000000000001a RCX: 00007fec8beba5e0
[40198.509409] RDX: 00007ffe42993a60 RSI: 0000000000080800 RDI: 00007ffe42993a60
[40198.509410] RBP: 00007ffe42993a60 R08: 00000000000000f4 R09: 0000000000000000
[40198.509410] R10: 0000000000000008 R11: 0000000000000246 R12: 00007ffe42993a60
[40198.509411] R13: 00007ffe42993b60 R14: 0000000000000080 R15: 00007ffe42993a50
[40198.509412] ---[ end trace 1e9097f1794c9129 ]---
[40198.517488] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40198.520743] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[40208.413855] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40208.427206] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40258.552508] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[40280.672560] usb 8-1.3: USB disconnect, device number 127
[40281.099583] usb 8-1.3: new full-speed USB device number 5 using xhci_hcd
[40281.183434] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[40281.183439] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[40281.183442] usb 8-1.3: Product: HORIPAD ONE
[40281.183444] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[40281.183446] usb 8-1.3: SerialNumber: 00005DC471C67F48
[40281.186144] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input145
[40281.186263] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40281.191617] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40281.192964] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[40281.197607] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40281.203557] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40281.213994] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[40295.826587] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40295.828515] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[40345.959654] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
```

cgutman, 2017-04-25:
4.10 doesn't have the xpad init quirks support yet. Those should be coming in 4.12, then the built-in driver should work for your hardware.

So the WARN_ON is specific to the xpad changes here, the upstream version will not puke like that. I needed to deliver possibly more than 2 packets, so we don't rely on never having a pending packet (which is what triggered the warning here). Can you try the upstream version to see if you still have broken analog sticks on resume? Just replace your xpad.c with this one from Dmitry's tree: https://git.kernel.org/pub/scm/linux/kernel/git/dtor/input.git/plain/drivers/input/joystick/xpad.c

That said, it is quite peculiar that you do have a pending packet during the open path. Are those particular logs from boot or wake from suspend?

orbea, 2017-04-25:
Those particular logs were from after suspend when the analog sticks stopped working.

I rebuilt the master with the new xpad.c from your link and it successfully silenced error output in dmesg. However the original issue (https://github.com/paroj/xpad/issues/59) was reintroduced... In short after 5 - 10 minutes the analog sticks will stop responding where I will have to unplug and plug back in the gamepad for it to work again for 5 - 10 more minutes...

Here is a diff of the xpad.c from master and from your link.
```
--- old/xpad.c	2017-04-25 10:21:12.287562785 -0700
+++ xpad.c	2017-04-25 10:08:48.000000000 -0700
@@ -74,7 +74,7 @@
  *
  * Later changes can be tracked in SCM.
  */
-#define DEBUG
+
 #include <linux/kernel.h>
 #include <linux/input.h>
 #include <linux/rcupdate.h>
@@ -229,7 +229,6 @@
 	{ 0x24c6, 0x5506, "Hori SOULCALIBUR V Stick", 0, XTYPE_XBOX360 },
 	{ 0x24c6, 0x5b02, "Thrustmaster, Inc. GPX Controller", 0, XTYPE_XBOX360 },
 	{ 0x24c6, 0x5b03, "Thrustmaster Ferrari 458 Racing Wheel", 0, XTYPE_XBOX360 },
-	{ 0x1532, 0x0a03, "Razer Wildcat for Xbox One", 0, XTYPE_XBOXONE },
 	{ 0xffff, 0xffff, "Chinese-made Xbox Controller", 0, XTYPE_XBOX },
 	{ 0x0000, 0x0000, "Generic X-Box pad", 0, XTYPE_UNKNOWN }
 };
@@ -327,7 +326,6 @@
 	XPAD_XBOX360_VENDOR(0x1430),		/* RedOctane X-Box 360 controllers */
 	XPAD_XBOX360_VENDOR(0x146b),		/* BigBen Interactive Controllers */
 	XPAD_XBOX360_VENDOR(0x1532),		/* Razer Sabertooth */
-	XPAD_XBOXONE_VENDOR(0x1532),		/* Razer Wildcat */
 	XPAD_XBOX360_VENDOR(0x15e4),		/* Numark X-Box 360 controllers */
 	XPAD_XBOX360_VENDOR(0x162e),		/* Joytech X-Box 360 controllers */
 	XPAD_XBOX360_VENDOR(0x1689),		/* Razer Onza */
@@ -339,6 +337,64 @@
 
 MODULE_DEVICE_TABLE(usb, xpad_table);
 
+struct xboxone_init_packet {
+	u16 idVendor;
+	u16 idProduct;
+	const u8 *data;
+	u8 len;
+};
+
+#define XBOXONE_INIT_PKT(_vid, _pid, _data)		\
+	{						\
+		.idVendor	= (_vid),		\
+		.idProduct	= (_pid),		\
+		.data		= (_data),		\
+		.len		= ARRAY_SIZE(_data),	\
+	}
+
+
+/*
+ * This packet is required for all Xbox One pads with 2015
+ * or later firmware installed (or present from the factory).
+ */
+static const u8 xboxone_fw2015_init[] = {
+	0x05, 0x20, 0x00, 0x01, 0x00
+};
+
+/*
+ * This packet is required for the Titanfall 2 Xbox One pads
+ * (0x0e6f:0x0165) to finish initialization and for Hori pads
+ * (0x0f0d:0x0067) to make the analog sticks work.
+ */
+static const u8 xboxone_hori_init[] = {
+	0x01, 0x20, 0x00, 0x09, 0x00, 0x04, 0x20, 0x3a,
+	0x00, 0x00, 0x00, 0x80, 0x00
+};
[... 135 lines cut ...]
-static int xpad_start_xbox_one(struct usb_xpad *xpad)
-{
-	static const u8 xbone_init_pkt0[] = {0x01, 0x20, 0x00, 0x09, 0x00,
-			0x04, 0x20, 0x3a, 0x00, 0x00, 0x00, 0x80, 0x00};
-	static const u8 xbone_init_pkt1[] = {0x05, 0x20, 0x00, 0x01, 0x00};
-	int retval;
-
-	retval = xpadone_send_init_pkt(xpad, xbone_init_pkt0, sizeof(xbone_init_pkt0));
-	if (retval)
-		return retval;
-
-	return xpadone_send_init_pkt(xpad, xbone_init_pkt1, sizeof(xbone_init_pkt1));
-}
-
 #ifdef CONFIG_JOYSTICK_XPAD_FF
 static int xpad_play_effect(struct input_dev *dev, void *data, struct ff_effect *effect)
 {
@@ -1732,6 +1798,7 @@
 	.disconnect	= xpad_disconnect,
 	.suspend	= xpad_suspend,
 	.resume		= xpad_resume,
+	.reset_resume	= xpad_resume,
 	.id_table	= xpad_table,
 };
 ```

cgutman, 2017-04-25:
Anything from xpad or USB in dmesg during your testing and when the analog sticks die?

orbea, 2017-04-25:
Just this over and over.
```
[45698.684022] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input262
[45740.925909] usb 8-1.3: USB disconnect, device number 122
[45741.179734] usb 8-1.3: new full-speed USB device number 123 using xhci_hcd
[45741.263450] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[45741.263455] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[45741.263458] usb 8-1.3: Product: HORIPAD ONE
[45741.263460] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[45741.263462] usb 8-1.3: SerialNumber: 00005DC471C67F48
[45741.266135] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input263
```

orbea, 2017-04-25:
I recompiled it again keeping the debug bits and dmesg now shows this.
```
[50317.537436] usbcore: registered new interface driver xpad
[50327.870573] usb 8-1.3: new full-speed USB device number 9 using xhci_hcd
[50327.954418] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[50327.954423] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[50327.954426] usb 8-1.3: Product: HORIPAD ONE
[50327.954428] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[50327.954430] usb 8-1.3: SerialNumber: 00005DC471C67F48
[50327.957093] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input272
[50327.984949] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[50384.112712] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[50390.166838] usb 8-1.3: USB disconnect, device number 9
[50390.437224] usb 8-1.3: new full-speed USB device number 10 using xhci_hcd
[50390.521129] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[50390.521134] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[50390.521137] usb 8-1.3: Product: HORIPAD ONE
[50390.521139] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[50390.521141] usb 8-1.3: SerialNumber: 00005DC471C67F48
[50390.523749] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input273
[50390.531605] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[50390.555667] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
```

orbea, 2017-04-25:
It seems I misunderstood the issue with the master. It will print warning and Call trace after rebooting, but the analog sticks will still stop working after some time as seen in the dmesg output below. (I guess that just took a while the other day?)
```
[52771.130128] usbcore: registered new interface driver xpad
[52791.903892] usb 8-1.3: new full-speed USB device number 18 using xhci_hcd
[52791.986505] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[52791.986508] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52791.986510] usb 8-1.3: Product: HORIPAD ONE
[52791.986512] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[52791.986513] usb 8-1.3: SerialNumber: 00005DC471C67F48
[52791.989156] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input282
[52791.989324] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52791.994678] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52791.997273] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[52792.004124] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52792.009622] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52792.019333] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[52794.921575] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52794.926807] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52845.099133] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[52848.024702] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52848.086012] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52898.156031] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[52901.102624] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52901.120203] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52951.237905] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[52954.046085] usb 8-1.3: USB disconnect, device number 18
[52954.474329] usb 8-1.3: new full-speed USB device number 19 using xhci_hcd
[52954.557945] usb 8-1.3: New USB device found, idVendor=0f0d, idProduct=0067
[52954.557947] usb 8-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52954.557949] usb 8-1.3: Product: HORIPAD ONE
[52954.557950] usb 8-1.3: Manufacturer: HORI CO.,LTD.
[52954.557952] usb 8-1.3: SerialNumber: 00005DC471C67F48
[52954.560636] input: HORIPAD ONE as /devices/pci0000:00/0000:00:09.0/0000:02:00.0/usb8/8-1/8-1.3/8-1.3:1.0/input/input283
[52954.560764] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52954.566434] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52954.586842] xpad 8-1.3:1.0: xpad_irq_in - urb shutting down with status: -2
[52959.963666] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[52959.968755] xpad 8-1.3:1.0: xpad_prepare_next_out_packet - found pending output packet 0
```
To be percise, what worked was `linux-4.10.0` and `xpad-2016.12.31_bd4f517_4.10.0_master-x86_64-1_git`. The repo was force pushed so bd4f517 is now 01f149bc9198d08a4a4bcaa3bf1322d7fcafeb58. Assuming this is right, then I'm still a bit lost on what changed?

cgutman, 2017-04-28:
Try the xpad.c from bd4f517 to make sure: https://raw.githubusercontent.com/paroj/xpad/bd4f51797c2a518be90b470d499059dc088ddbd3/xpad.c

cgutman, 2017-04-30:
Can you also test master plus adding `#define DEBUG_VERBOSE` near the existing `#define DEBUG` line?

orbea, 2017-05-20:
Sorry for the delay in response, other responsibilities and distractions took over. I am now on `linux-4.11.1` and this is still occurring so I built the xpad master with this patch.
```
diff --git a/xpad.c b/xpad.c
index 4179ca2..e0accdb 100644
--- a/xpad.c
+++ b/xpad.c
@@ -75,6 +75,7 @@
  * Later changes can be tracked in SCM.
  */
 #define DEBUG
+#define DEBUG_VERBOSE
 #include <linux/kernel.h>
 #include <linux/input.h>
 #include <linux/rcupdate.h>
```
dmesg then printed this before the analog sticks stopped responding.
https://pastebin.com/06ebAjjL

I then tested the `xpad.c` from https://github.com/paroj/xpad/commit/bd4f51797c2a518be90b470d499059dc088ddbd3 and the problem seems to be gone. Here is a diff between 01f149b and bd4f517.
```
--- xpad.c.new	2017-05-20 13:20:42.765734413 -0700
+++ xpad.c.old	2017-05-20 12:16:27.841624888 -0700
@@ -320,18 +320,18 @@
 	XPAD_XBOXONE_VENDOR(0x0738),		/* Mad Catz FightStick TE 2 */
 	XPAD_XBOX360_VENDOR(0x0e6f),		/* 0x0e6f X-Box 360 controllers */
 	XPAD_XBOXONE_VENDOR(0x0e6f),		/* 0x0e6f X-Box One controllers */
-	XPAD_XBOX360_VENDOR(0x0f0d),		/* Hori Controllers */
-	XPAD_XBOXONE_VENDOR(0x0f0d),		/* Hori Controllers */
+	XPAD_XBOX360_VENDOR(0x0f0d),		/* Hori X-Box 360 Controllers */
+	XPAD_XBOXONE_VENDOR(0x0f0d),		/* Hori X-Box One Controllers */
 	XPAD_XBOX360_VENDOR(0x12ab),		/* X-Box 360 dance pads */
 	XPAD_XBOX360_VENDOR(0x1430),		/* RedOctane X-Box 360 controllers */
-	XPAD_XBOX360_VENDOR(0x146b),		/* BigBen Interactive Controllers */
+	XPAD_XBOX360_VENDOR(0x146b),		/* BigBen Interactive X-Box 360 Controllers */
 	XPAD_XBOX360_VENDOR(0x1532),		/* Razer Sabertooth */
 	XPAD_XBOX360_VENDOR(0x15e4),		/* Numark X-Box 360 controllers */
 	XPAD_XBOX360_VENDOR(0x162e),		/* Joytech X-Box 360 controllers */
 	XPAD_XBOX360_VENDOR(0x1689),		/* Razer Onza */
 	XPAD_XBOX360_VENDOR(0x1bad),		/* Harminix Rock Band Guitar and Drums */
-	XPAD_XBOX360_VENDOR(0x24c6),		/* PowerA Controllers */
-	XPAD_XBOXONE_VENDOR(0x24c6),		/* PowerA Controllers */
+	XPAD_XBOX360_VENDOR(0x24c6),		/* PowerA X-Box 360 Controllers */
+	XPAD_XBOXONE_VENDOR(0x24c6),		/* PowerA X-Box One Controllers */
 	{ }
 };
 
@@ -609,36 +609,14 @@
 }
 
 /*
- *	xpadone_process_packet
+ *	xpadone_process_buttons
  *
- *	Completes a request by converting the data into events for the
- *	input subsystem. This version is for the Xbox One controller.
- *
- *	The report format was gleaned from
- *	https://github.com/kylelemons/xbox/blob/master/xbox.go
+ *	Process a button update packet from an Xbox one controller.
  */
-static void xpadone_process_packet(struct usb_xpad *xpad, u16 cmd, unsigned char *data)
+static void xpadone_process_buttons(struct usb_xpad *xpad,
+				struct input_dev *dev,
+				unsigned char *data)
 {
-	struct input_dev *dev = xpad->dev;
-
-	/* the xbox button has its own special report */
-	if (data[0] == 0X07) {
-		/*
[... 215 lines cut ...]
 		mutex_lock(&input->mutex);
-		if (input->users)
+		if (input->users || xpad->xtype == XTYPE_XBOXONE)
 			xpad_stop_input(xpad);
 		mutex_unlock(&input->mutex);
 	}
@@ -1708,16 +1724,8 @@
 		retval = xpad360w_start_input(xpad);
 	} else {
 		mutex_lock(&input->mutex);
-		if (input->users) {
+		if (input->users || xpad->xtype == XTYPE_XBOXONE)
 			retval = xpad_start_input(xpad);
-		} else if (xpad->xtype == XTYPE_XBOXONE) {
-			/*
-			 * Even if there are no users, we'll send Xbox One pads
-			 * the startup sequence so they don't sit there and
-			 * blink until somebody opens the input device again.
-			 */
-			retval = xpad_start_xbox_one(xpad);
-		}
 		mutex_unlock(&input->mutex);
 	}
 
```

orbea, 2017-07-03:
This issue now is present in `4.12` kernels.

orbea, 2017-11-29:
Hi, can anything be done about this?

closed 2020-06-24 by orbea