paroj/xpad #70: Yet another "new" Logic 3 Rock Candy controller

nentibusarchitectura, 2017-04-30:
Hi,

Recently I acquired a brand new XBOX One controller, it is made by "rock candy" apparently.

It is not supported by xpad, when I connect it to the computer I get the following ID when I run lsusb:

Bus 004 Device 004: ID 0e6f:0346 Logic3

I tried to add it to the driver by adding the following line to the list of devices:

{ 0x0e6f, 0x0346, "Logic 3 - Rock Candy Wired Controller for Xbox One", 0, XTYPE_XBOXONE },

Once I compile and install the driver via DKMS, once I connect the controller I can see that a js device is added under /dev/input/js0.

I can run jstest /dev/input/js0 and I get a list of all axes and buttons, however no input is registered, the controller does nothing.

If I connect and re-connect the controller one out of 20 times it will produce input.

I'm not a developer, but I can follow instructions, please let me know how can I help debug this issue.

Many thanks in advance.

cgutman, 2017-04-30:
Hmm, I wonder if we need to wait for some time before sending the second init packet on these gamepads. Do you see any errors reported in your `dmesg` output? Does it work if you boot your computer with the gamepad plugged in?

nentibusarchitectura, 2017-04-30:
Hi,

No, working on boot is as random as getting it to work by plugging it in after boot.

This is what I get on dmesg upon boot:

```
[   15.780346] xpad: loading out-of-tree module taints kernel.
[   15.780391] xpad: module verification failed: signature and/or required key missing - tainting kernel
[   15.785174] usbcore: registered new interface driver xpad
[   16.786054] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   16.789491] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   16.812491] xpad 4-1:1.0: xpad_irq_in - urb shutting down with status: -2
[   17.600069] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   17.601490] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   17.616487] xpad 4-1:1.0: xpad_irq_in - urb shutting down with status: -2
```

This is what I get if I unplug and re-plug:

```
[   15.780346] xpad: loading out-of-tree module taints kernel.
[   15.780391] xpad: module verification failed: signature and/or required key missing - tainting kernel
[   15.785174] usbcore: registered new interface driver xpad
[   16.786054] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   16.789491] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   16.812491] xpad 4-1:1.0: xpad_irq_in - urb shutting down with status: -2
[   17.600069] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   17.601490] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[   17.616487] xpad 4-1:1.0: xpad_irq_in - urb shutting down with status: -2
```

This is the output of jstest:

```
# jstest /dev/input/js0 
Driver version is 2.1.0.
Joystick (Logic3 - Rock Candy Wired Controller for Xbox One) has 8 axes (X, Y, Z, Rx, Ry, Rz, Hat0X, Hat0Y)
and 11 buttons (BtnX, BtnY, BtnTL, BtnTR, BtnTR2, BtnSelect, BtnThumbL, BtnThumbR, ?, ?, ?).
Testing ... (interrupt to exit)
Axes:  0:     0  1:     0  2:-32767  3:     0  4:     0  5:-32767  6:     0  7:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off ^C
```

It is very random, I'm sure used to work more often a few months ago, but now doesn't work very often.

Regards.

cgutman, 2017-04-30:
Can you try adding adding `#define DEBUG_VERBOSE` near the existing `#define DEBUG` line in xpad.c and reinstalling the module? Then paste the `dmesg` output from xpad which should have some hex dumps of incoming packets from the gamepad.

nentibusarchitectura, 2017-04-30:
Hi I removed the driver, added the line right below the other debug line as:

```
#define DEBUG
#define DEBUG_VERBOSE
#include <linux/kernel.h>
...
```

Then recompiled re-installed the driver, I do not see anything different on the dmesg, it is still the same output as before.

Have I understood well what needed to be done?

Regards.

cgutman, 2017-04-30:
Hmm, that should be all that is needed. I wonder if `dkms remove` doesn't remove the module from memory. Can you unplug your gamepad, run `modprobe -r xpad` then plug it in again?

I'll try it here too just to make sure I'm not missing something.

nentibusarchitectura, 2017-04-30:
I removed the driver entirely, not just modprobe  -r xpad, but removed the module removed the git tree, re-downloaded, applied the changes to the xpad.c file, made sure it re-compiles and tried again, nothing different comes on the dmesg.

cgutman, 2017-04-30:
Maybe your gamepad really sends nothing? Can you try `jstest` and see if you get anything in `dmesg` from that?

nentibusarchitectura, 2017-05-01:
This is what I get:

[  +0.375954] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0
[  +0.004040] xpad 4-1:1.0: xpad_prepare_next_out_packet - found pending output packet 0

I tried to connect/reconnect the controller until it does something, in the not so distant past it used to work about 1 in 20 times, but now it doesn't seem to want to, strange.

I recently began using kernel 4.10 I wonder if that has made the problem worse or if it may have made it stop completely.

I will try to test tomorrow using a 4.4.x I have in the same box.

nentibusarchitectura, 2017-05-01:
I have verified that xpad produces hex output by plugging in an xbox360 controller, so the #define DEBUG_VERBOSE line is working.

Tested the rock candy controller with kernel 4.4.x and I do net get any hex output either.

Tried re-connecting the controller many times and couldn't get it to work even once.

So you're correct it looks like the gamepad doesn't send anything.

nentibusarchitectura, 2017-05-06:
Hi again,

So... is there anything else I can do from my end? any info I can provide? would I be better just getting another controller or is it worth it to try to do something?

cgutman, 2017-05-07:
Can you see what the output is on one of the random times when it does work?

nentibusarchitectura, 2017-07-22:
Hi,

I just tested the controller again today and suddenly it works and it works reliably.

During this time I have changed the kernel I run, I'm running 4.12.2 with the xpad that comes with the kernel and it runs flawlessly now.

The pad is detected as generic xbox controller.

Thanks for all your help with this.

closed 2017-12-17 by paroj