paroj/xpad #80: Checking Connection of XBox360 Wireless Controler

FreddyDoll, 2017-07-11:
Hello,

I want to control a robot using the XBox360 Wireless Controler.
The Problem i found (both with standard Linux driver and paroj/xpad) was:

-The /dev/input/js0 device keeps its state on controler disconnect(removing batteries). So all Buttons pressed when disconnecting keep beeing reported as pressed.

For my case it would be ideal if the Device would change its state to "neutral" (no buttons pressed, all axis 0) on disconnect.
I found this function in xpad.c which seems to know something about the controller state but i'm not sure how to change this.

```
static void xpad360w_process_packet(struct usb_xpad *xpad, u16 cmd, unsigned char *data)
{
	struct input_dev *dev;
	bool present;

	/* Presence change */
	if (data[0] & 0x08) {
		present = (data[1] & 0x80) != 0;

		if (xpad->pad_present != present) {
			xpad->pad_present = present;
			schedule_work(&xpad->work);
		}
	}

	/* Valid pad data */
	if (data[1] != 0x1)
		return;

	rcu_read_lock();
	dev = rcu_dereference(xpad->x360w_dev);
	if (dev)
		xpad360_process_packet(xpad, dev, cmd, &data[4]);
	rcu_read_unlock();
}
```
Greetings,

Frederik

cgutman, 2017-07-11:
This scenario appears to work fine for me on a stock 4.11.7 kernel. Are you not seeing the device disappear upon removal?
```
$ jstest /dev/input/js0
Driver version is 2.1.0.
Joystick (Xbox 360 Wireless Receiver) has 8 axes (X, Y, Z, Rx, Ry, Rz, Hat0X, Hat0Y)
and 15 buttons (BtnX, BtnY, BtnTL, BtnTR, BtnTR2, BtnSelect, BtnThumbL, BtnThumbR, ?, ?, ?, (null), (null), (null), (null)).
Testing ... (interrupt to exit)
Axes:  0:   774  1:  -871  2:-32767  3: -2031  4:   428  5:-32767  6:     0  7:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off 13:off 14:off 
jstest: error reading: No such device
```

cgutman, 2017-07-11:
It also looks like the input stack generates synthetic button release events (as you expect). I did not release the button in this scenario while I disconnected the batteries.

```
evtest /dev/input/event8
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x45e product 0x2a1 version 0x100
Input device name: "Xbox 360 Wireless Receiver"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 704 (BTN_TRIGGER_HAPPY1)
    Event code 705 (BTN_TRIGGER_HAPPY2)
    Event code 706 (BTN_TRIGGER_HAPPY3)
    Event code 707 (BTN_TRIGGER_HAPPY4)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   1280
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 1 (ABS_Y)
      Value  -1234
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max      255
    Event code 3 (ABS_RX)
      Value  -1768
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 4 (ABS_RY)
      Value    907
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max      255
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)
Properties:
Testing ... (interrupt to exit)
Event: time 1499775570.172573, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1499775570.172573, -------------- SYN_REPORT ------------
Event: time 1499775574.043408, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1499775574.043409, -------------- SYN_REPORT ------------
expected 24 bytes, got -1

evtest: error reading: No such device
```

FreddyDoll, 2017-07-11:
Hi,

Thank you so much for the quick reply. So it seems like i'm running the wrong Version of the driver/kernel. This seems obvious now since i'm running ROS Kinetic and installed Ubuntu 16.04.1 LTS for compatibility
here is what i could find out:
cat /proc/version: 
`Linux version 4.4.0-83-generic (buildd@lgw01-29) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017`

modinfo xpad:
```
filename:       /lib/modules/4.4.0-83-generic/updates/dkms/xpad.ko
license:        GPL
description:    X-Box pad driver
author:         Marko Friedemann <mfr@bmx-chemnitz.de>
srcversion:     96C3BF93D54528FF19E7047
alias:          usb:v24C6p*d*dc*dsc*dp*icFFisc47ipD0in*
alias:          usb:v24C6p*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v24C6p*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v1BADp*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v1BADp*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v1689p*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v1689p*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v162Ep*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v162Ep*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v15E4p*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v15E4p*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v1532p*d*dc*dsc*dp*icFFisc47ipD0in*
alias:          usb:v1532p*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v1532p*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v146Bp*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v146Bp*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v1430p*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v1430p*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v12ABp*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v12ABp*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v0F0Dp*d*dc*dsc*dp*icFFisc47ipD0in*
alias:          usb:v0F0Dp*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v0F0Dp*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v0E6Fp*d*dc*dsc*dp*icFFisc47ipD0in*
alias:          usb:v0E6Fp*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v0E6Fp*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v0738p*d*dc*dsc*dp*icFFisc47ipD0in*
alias:          usb:v0738p4540d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0738p*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v0738p*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v046Dp*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v046Dp*d*dc*dsc*dp*icFFisc5Dip01in*
alias:          usb:v045Ep*d*dc*dsc*dp*icFFisc47ipD0in*
alias:          usb:v045Ep*d*dc*dsc*dp*icFFisc5Dip81in*
alias:          usb:v045Ep*d*dc*dsc*dp*icFFisc5Dip01in*
depends:        ff-memless
vermagic:       4.4.0-83-generic SMP mod_unload modversions 
parm:           dpad_to_buttons:Map D-PAD to buttons rather than axes for unknown pads (bool)
parm:           triggers_to_buttons:Map triggers to buttons rather than axes for unknown pads (bool)
parm:           sticks_to_null:Do not map sticks at all for unknown pads (bool)
parm:           auto_poweroff:Power off wireless controllers on suspend (bool)
```
I tried installing xpad from github using:
```
sudo git clone https://github.com/paroj/xpad.git /usr/src/xpad-0.4
sudo dkms install -m xpad -v 0.4
```

i have the feeling this is where i failed....
Any tip on how to load the current driver?

Thanks again

Frederik

cgutman, 2017-07-11:
Ok, so it seems like you're definitely using 4.4's xpad driver. I remember that disconnect bug from back then.

Have you tried rebooting after the dkms install? I wonder if DKMS doesn't rmmod the already loaded driver when it does the initial install.

FreddyDoll, 2017-07-11:
OK.... Turn it off and on again is what you're saying....
DAMN it worked :D 

Thank you so much!!

closed 2017-12-17 by paroj