paroj/xpad #83: Ifyoo V-one ( SHANWAN ? ) Controller doesn't send any buttons when in "xinput" mode ( mimicing xbox 360 controller )

tacchinotacchi, 2017-08-22:


Jigoku, 2018-08-13:
Just got this controller, Same problem here...
`Bus 011 Device 011: ID 045e:028e Microsoft Corp. Xbox360 Controller`

Jigoku, 2018-08-13:
@tacchinotacchi 
If it means anything, i just ran  `sdl2-jstest  -r 0` which causes this controller to rumble whilst in Xinput mode. So the gamepad is able to receive packets. 

Testing input shows nothing; `sdl2-jstest --test 0`

There's also this:
```
$ sdl2-jstest --list
Found 1 joystick(s)

Joystick Name:     'Microsoft X-Box 360 pad'
Joystick GUID:     030000005e0400008e02000010010000
Joystick Number:    0
Number of Axes:     6
Number of Buttons: 11
Number of Hats:     1
Number of Balls:    0
GameController:
  Name:    'XInput Controller'
  Mapping: '(null)'
```

Could it be possible that the last line there may have something to do with it? `   Mapping: '(null)'`

tacchinotacchi, 2018-12-27:
I too ended up returning it a week after, never figured out what was special about the packets it received from windows

djurbloopers, 2019-02-10:
The fixcontroller.py script by dnmodder mentioned in #119 worked fine with a SHANWAN in X input mode. 
That is, sending `dev.ctrl_transfer(0xc1, 0x01, 0x0100, 0x00, 0x14)`  makes the gamepad work. 

This particular model is a Trust GXT 540 Yula USB Wired Gamepad.

Jigoku, 2019-02-10:
[djurbloopers](/djurbloopers) That's the identical vendor/product id i had previously with my SHANWAN device. Glad someone sussed that out. I am tempted to get another one of those controllers and see if that works for me. As it was a rather nice gamepad, despite it not working at the time.

Cheers for pointing that out!