paroj/xpad #90: Support for a new Xbox one controller wireless receiver

cszipper, 2018-01-06:
Hey, I have got a Xbox one controller with a wireless receiver but it can't work with this driver. I noticed that this product id is not included in xpad_devices.
> kernel: [ 2359.371456] usb 3-6: New USB device found, idVendor=045e, idProduct=02fe
> kernel: [ 2359.371457] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> kernel: [ 2359.371457] usb 3-6: Product: XBOX ACC
> kernel: [ 2359.371458] usb 3-6: Manufacturer: Microsoft Inc.

perillamint, 2018-01-10:
It seems duplicate of #14.

paroj, 2018-01-10:
yes, the WiFi dongle is currently not supported. And we did not have much progress in the mentioned issues - except for users switching to the bluetooth version.

closed 2018-01-10 by paroj