paroj/xpad #96: please provide some instructions on adding new gamepads

ObiWahn, 2018-05-03:
Hi,

I posses a xbox one game-pad and would really like to use it. My device can be found at `Bus 005 Device 007: ID 045e:02e6 Microsoft Corp.` I decided to add it below https://github.com/paroj/xpad/blob/master/xpad.c#L144 installed and loaded the module. verified that the correct module is loaded with `modinfo` and was hoping to see something in the `dmesg` output. But I do not get logs with that command and `usb-devices` tells me that the device is not associated with any driver. Please provide me with some feedback that hints me how to debug this issue.

zimmertr, 2018-05-04:
Based on your ID I assume you're using a wireless controller with the Wireless Adapter. 

The Xbox Wireless controller is not detected by Windows as a gamepad. It is instead reported as a networking device to windows. The adapter and controller both have wireless network devices within them. This greatly complicates the process involved in writing a driver.

Also, I am not a developer on this project nor am I familiar with thr code base. I just simply read two results with the flowing Google Query:

`Xbox one "045e:02e6"`

Arranmc182, 2018-08-18:
For the Hyperkin Duke I just did "sudo nano /sys/bus/usb/drivers/xpad/new_id" and add device ID "2e24 0652" then rebooted the system, you can find your device ID via "lsusb" command