paroj/xpad #98: xone: Logic3

cynodesmus, 2018-06-03:
Please help configure and add support for the gamepad ID 0e6f:0160 Logic3 (for Xbox One).

Gamepad was bought recently in China. Now I added the line myself: { 0x0e6f, 0x0160, "Logic3 Xbox One wired controller", 0, XTYPE_XBOXONE } (by analogy with the same purchased devices from the same seller - 0x24c6, 0x543a, "PowerA Xbox One wired controller"). After reinstalling the xpad module, the gamepad is working, except for backlight and vibration. Instead of vibration from the gamepad you can hear a slight buzz. However, when you boot into Windows, the vibrating output works.

P.S. Who is the real producer of these controllers can not be determined (in China, they collect devices from what comes to hand). Both wired controllers on the adapter have the same model: WTYX-618

Arranmc182, 2018-08-18:
You might need ti add it's device its USB Device ID to xpad use command "lsusb" to find the ID then copy it but change the : for a space (E.g. 0e6f 02b5) and add it to  "sudo nano /sys/bus/usb/drivers/xpad/new_id" save and reboot and try again, this is a common issue with third party Xbox One Controllers as they show up with diffrent device ID that xpad is just not aware about yet, if you get a fix do post back.

cynodesmus, 2018-08-27:
Unfortunately the way with the change of "new_id" did not help.
Earlier, I wrote line "{ 0x0e6f, 0x0160, "Logic3 Xbox One Controller", 0, XTYPE_XBOXONE }", at xpad.c, followed by an update, but this option did not change the situation - instead of vibration output from the gamepad a quiet squeak is heard, the light indication also does not work.

P.S. In the debug log, no lines associated with the work of the gamepad are displayed ...