paroj/xpad #102: please add PDP 0x02a6

jd42x, 2018-06-29:
Hi,
my recent purchase "PDP Wired Controller for Xbox One - Stealth Series | Revenant Blue" seems pretty identical your most recent commit "Input: xpad - add support for PDP 0x02a8" - and indeed the following patch made it work:

--- xpad.c.orig 2018-06-28 21:42:40.905787251 +0200
+++ xpad.c      2018-06-29 00:06:32.314733856 +0200
@@ -233,6 +233,7 @@
        { 0x0e6f, 0x0246, "Rock Candy Gamepad for Xbox One 2015", 0, XTYPE_XBOXONE },
        { 0x0e6f, 0x02ab, "PDP Controller for Xbox One", 0, XTYPE_XBOXONE },
        { 0x0e6f, 0x02a4, "PDP Wired Controller for Xbox One - Stealth Series", 0, XTYPE_XBOXONE },
+       { 0x0e6f, 0x02a6, "PDP Wired Controller for Xbox One - Stealth Series", 0, XTYPE_XBOXONE },
        { 0x0e6f, 0x0301, "Logic3 Controller", 0, XTYPE_XBOX360 },
        { 0x0e6f, 0x0346, "Rock Candy Gamepad for Xbox One 2016", 0, XTYPE_XBOXONE },
        { 0x0e6f, 0x0401, "Logic3 Controller", 0, XTYPE_XBOX360 },
@@ -533,6 +534,8 @@
        XBOXONE_INIT_PKT(0x0e6f, 0x02ab, xboxone_pdp_init2),
        XBOXONE_INIT_PKT(0x0e6f, 0x02a4, xboxone_pdp_init1),
        XBOXONE_INIT_PKT(0x0e6f, 0x02a4, xboxone_pdp_init2),
+       XBOXONE_INIT_PKT(0x0e6f, 0x02a6, xboxone_pdp_init1),
+       XBOXONE_INIT_PKT(0x0e6f, 0x02a6, xboxone_pdp_init2),
        XBOXONE_INIT_PKT(0x0e6f, 0x02a8, xboxone_pdp_init1),
        XBOXONE_INIT_PKT(0x0e6f, 0x02a8, xboxone_pdp_init2),
        XBOXONE_INIT_PKT(0x24c6, 0x541a, xboxone_rumblebegin_init),
Could you please include it into your next release?
Thanks!

paroj, 2018-06-29:
can you create a pull-request?

jd42x, 2018-06-29:
...to be honest, I don't know how to do that...

jd42x, 2018-06-29:
ok... tried it... hope I did it right...

paroj, 2018-06-29:
:+1: thanks

PR #103 (2018-06-29): add PDP 0x02a6
fixes  #102
closed 2018-06-29 by paroj