paroj/xpad #106: Hyperkin Duke Xbox One/Windows 10 Controller, works but not out the box

Arranmc182, 2018-08-17:
The Hyperkin Duke for Xbox One & Windows 10 does in fact work fully with the xpad driver just seems like the device ID is not recognized by the Driver out the box it is a simple fix, "sudo nano /sys/bus/usb/drivers/xpad/new_id" and add device ID "2e24 0652", is there any chance going forward this could be setup out the box at all as the Duke Controller is officially licensed by MS and even on Windows shows up as an official Xbox controller and uses MS Xbox drivers.

Arranmc182, 2018-08-18:
To get this automated I just made a file called "new_id" and added the USB ID to it then made a script to just copy this over every time I log in. 

This is my script.

#!/bin/bash

echo YOUROWNPASSWORD | sudo -S rmmod xpad
echo YOUROWNPASSWORD | sudo -S modprobe xpad
echo YOUROWNPASSWORD | sudo -S cp /home/arran/Documents/xpad/new_id /sys/bus/usb/drivers/xpad/new_id

Now the controller will work without having to manually load the device ID every time.

Arranmc182, 2018-08-18:
This is the system I tested this on

OS: Ubuntu 18.04 bionic
Kernel: x86_64 Linux 4.18.3-041803-generic
Shell: bash 4.4.19
Resolution: 1920x1080
DE: MATE 1.20.0
WM: Compiz
CPU: AMD FX-8350 Eight-Core @ 8x 4GHz [18.7°C]
 GPU: GeForce GTX 970

Deerskin, 2018-10-07:
What's the output of `lsusb -v` when the controller is plugged in?

Arranmc182, 2018-10-20:
I would check out this video its not mine but should give more info [https://www.youtube.com/watch?v=zYTMTaeB0P4](url)

parkerlreed, 2019-01-07:
````
Bus 001 Device 004: ID 2e24:0652  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x2e24 
  idProduct          0x0652 
  bcdDevice            1.01
  iManufacturer           1 DDR
  iProduct                2 Hyperkin Duke
  iSerial                 3 0000976F10C8B159
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0060
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
````

parkerlreed, 2019-04-03:
Any info on when this might get into mainline?

parkerlreed, 2019-04-03:
Also Steam only detects it as generic controller when using this method. Is there a way to get it to see it as an Xbox controller?

parkerlreed, 2019-04-03:
Found the upstream mentioning the reboot issue https://patchwork.kernel.org/patch/10421897/

Arranmc182, 2019-05-02:
> Also Steam only detects it as generic controller when using this method. Is there a way to get it to see it as an Xbox controller?

This is common issue with 3rd party Xbox One controllers, I no longer own then the duske as sold it on to buy other controllers so far have found the 2 other one a PDP & other PowerA both got detected as generic xinput controllers so steams will only see it as generic controller, but some games will detect it as an xbox controller just fine with out help from steam but other will not.

MARKMENTAL, 2021-02-05:
Just used the OP's instructions to get my hyperkin duke working with openSUSE Leap 15.2. Thanks a bunch!!

ThisNekoGuy, 2021-12-26:
Is there a reason why this hasn't been added to the driver yet?
I'd have thought support for this would have been added 3-4 years ago... :woozy_face:

paroj, 2021-12-26:
actually, it has been added:
https://github.com/paroj/xpad/blob/0585a19cee92756bb916c8f08c808bbd6c70eeff/xpad.c#L313

so, we probably can close this.

closed 2021-12-26 by paroj

ThisNekoGuy, 2021-12-27:
> actually, it has been added

@paroj You say this but I still have this problem with kernel 5.14.21 with the driver in it :/
It always reboots constantly and only behaves when Steam is on, per the suggesting in the first post here

Malsententia, 2022-08-13:
5.18.16 here, it's still an issue.