paroj/xpad #107: Support Hori Xbox Fighting Commander ONE

palmerj, 2018-09-02:
Currently Hori Xbox Fighting Commander ONE this gamepad is not supported. 
On Ubuntu 18.04 the lsusb is not identifying the device by name, only `ID 24c6:5510`

This pad is working on MacOSX (using https://github.com/360Controller/360Controller), PC and XBox360 and Xbox One. It is a USB wired device. See http://stores.horiusa.com/fighting-commander-for-xbox-one/. It has Xbox One and Xbox 360 support modes. 

The device details of the pad are (taken from my Mac):

```
Fighting Commander ONE:

Product ID: 0x5510
Vendor ID: 0x24c6
Version: 1.00
Serial Number: 63AD1394
Speed: Up to 12 Mb/sec
Manufacturer: HORI CO.,LTD.
Location ID: 0x14100000 / 9
Current Available (mA): 500
Current Required (mA): 500
Extra Operating Current (mA): 0
```

Would fixing this be as easy as adding the following to xpad.c:

```c++
{ 0x24c6, 0x5510, "Hori Fighting Commander ONE", MAP_TRIGGERS_TO_BUTTONS, XTYPE_XBOX360 },
```

Wasn't sure about the MAP_TRIGGERS_TO_BUTTONS flag. As you can see from the controller picture it has RB and RT button on the face, but also the 4 triggers buttons on the shoulder.

Thanks heaps

tbsf, 2018-09-25:
Look at this PR for example of adding 

https://github.com/paroj/xpad/pull/109/files

There are 3 lines to add. After you add, remove, reinstall as in instructions for upgrading then reboot.

palmerj, 2018-09-25:
Thanks do you know if I should set the MAP_TRIGGERS_TO_BUTTONS flag. Also the controller has a Xbox 360 and Xbox One mode. Which type should I add? Both or does that cause a conflict?

tbsf, 2018-09-26:
I'm not sure. Try copying the lines for this HORIPAD:

https://github.com/paroj/xpad/blob/master/xpad.c#L251
https://github.com/paroj/xpad/blob/master/xpad.c#L531

I'm wondering if the packet for other HORI pads will initialize yours correctly. Curious to know if it works for you.

closed 2022-01-10 by paroj

palmerj, 2022-01-10:
Thank you very much. Do you know when you might be pushing these changes upstream?

paroj, 2022-01-12:
typically around december. but sometimes upstream pulls changes by themselves