paroj/xpad #117: PDP Stealth with ID 0e6f:02cb Logic3 incompatible.

averyre, 2018-11-29:
An incompatible controller in the Stealth line, getting this working would be amazing.

averyre, 2018-11-29:
I was able to do a temporary fix by replaceing the device IDs in xpad.c for the standard Stealth to 0e6f:02cb, reinstalling and rebooting. But I'm sure it will break if I ever update. An entry added would be nice.

cgutman, 2018-12-05:
@averyre I submitted a patch which was accepted to quirk all PDP gamepads. https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=a6754fae1e66e9a40fed406290d7ca3f2b4d227c

Does your issue go away if you use the latest upstream xpad.c? https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/plain/drivers/input/joystick/xpad.c?id=a6754fae1e66e9a40fed406290d7ca3f2b4d227c

averyre, 2018-12-05:
Unfortunately it does not, I checked the source file afterwards and noticed that there is no entry for 0e6f:02cb in the upstream xpad.c as well.

It's this specific controller: https://www.amazon.ca/Wired-Controller-Xbox-Purple-048-082-NA-CM05/dp/B07GDXXRK6 purchased here in Canada, if maybe it has a unique device id for that reason? I'm assuming it's just an issue of a missing entry, as when I replace any other PDP Stealth gamepad id in xpad.c with my own, it will operate the device just fine.

cgutman, 2018-12-05:
The entry with the gamepad name in the list is purely cosmetic and has no effect on the functionality of the gamepad. The only part that matters is the entries in the quirk list.

How are you testing?

cgutman, 2018-12-05:
The quirks list is that xboxone_init_packets array where I made that change in my patch.

If you run `jstest-gtk` does your gamepad show up at all when running the upstream xpad.c unmodified?

So that find+replace fixes your gamepad even starting from the upstream file I linked above? The only way think that would happen is if our heuristic for detecting the type of gamepad is wrong, but it's not something I've seen before.

Can you post the output of `lsusb -v` for that gamepad too?

averyre, 2018-12-06:
I'm an idiot. I was unzipping the GitHub package and not using the upstream version when making the second modification. It works fine with the upstream version. My apologies!

cgutman, 2018-12-06:
did you need to modify the upstream xpad.c or did it work with no changes?

averyre, 2018-12-06:
No changes required to the upstream xpad.c . Did you still want my lsusb -v?

cgutman, 2018-12-06:
Nope, all good. Thanks for testing!

wraithalchemist, 2018-12-11:
I'm am new and just bought this same controller for my raspberry pi 3b+ how do I replace xpad.c? I tried and it said access is denied?

averyre, 2018-12-12:
Follow the standard instructions but before installing the driver, replace xpad.c with the fix I uploaded.