paroj/xpad #118: Support for Gamesir G3w

Oblomov, 2018-11-30:
Hello,

I recently acquired a Gamesir G3w controller. This has two “operating modes”: a “native” one, which is more or less handled by the generic hid driver (aside from some issues about buttons, trigger and joysticks mapping), and an “XBOX 360” mode.

The device is initially recognized in “native” mode, but does a modeswitch very quickly into XBOX360 mode. it can be brought back to “native” by a long press of its home button. When switching to XBOX360, xpra attempts to manage it,but fails in xpad_irq_in, so that the controller appears dead (nothing seems to work).

Here's the relevant dmesg output:

```
[Dec 1 00:23] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[  +0.150227] usb 1-1: New USB device found, idVendor=05ac, idProduct=055b, bcdDevice= 2.00
[  +0.000010] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  +0.000006] usb 1-1: Product: Gamesir-G3w
[  +0.000004] usb 1-1: Manufacturer: Xiaoji
[  +0.004765] input: Xiaoji Gamesir-G3w as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:05AC:055B.0002/input/input24
[  +0.000341] hid-generic 0003:05AC:055B.0002: input,hidraw1: USB HID v1.10 Gamepad [Xiaoji Gamesir-G3w] on usb-0000:00:14.0-1/input0
[  +0.013228] usb 1-1: USB disconnect, device number 5
[  +0.655733] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[  +0.150845] usb 1-1: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.10
[  +0.000009] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  +0.000004] usb 1-1: Product: XBOX 360 For Windows
[  +0.000003] usb 1-1: Manufacturer: Xiaoji
[  +0.004556] xpad 1-1:1.0: xpad_prepare_next_out_packet - found pending output packet 2
[  +0.000066] input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/input/input25
[  +0.032555] xpad 1-1:1.0: xpad_irq_in - urb shutting down with status: -2
[  +0.048206] xpad 1-1:1.0: xpad_irq_in - urb shutting down with status: -2
[  +0.055941] xpad 1-1:1.0: xpad_irq_in - urb shutting down with status: -2
```
And nothing thereafter.

If there is anything I can do to help identify the issue and improve the compatibility with the device, please let me know.

Oblomov, 2018-12-01:
Some additional information

* if I prevent `usbhid` from attaching to the device (`usbhid.quirks:0x05ac:0x055b:0x4`), the switch from HID to XBOX 360 mode does not happen, which seems to suggest that at least on one direction the modeswitch can be triggered with an appropriate control packet (which apparently is part of standard probing?)

* this is the `lsusb` output of the device in HID mode:

```

Bus 001 Device 009: ID 05ac:055b Apple, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x05ac Apple, Inc.
  idProduct          0x055b 
  bcdDevice            2.00
  iManufacturer           1 Xiaoji
  iProduct                2 Gamesir-G3w
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     137
          Report Descriptor: (length is 137)
            Item(Global): Usage Page, data= [ 0x01 ] 1
                            Generic Desktop Controls
            Item(Local ): Usage, data= [ 0x05 ] 5
                            Gamepad
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x01 ] 1
            Item(Global): Physical Minimum, data= [ 0x00 ] 0
            Item(Global): Physical Maximum, data= [ 0x01 ] 1
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Global): Report Count, data= [ 0x0d ] 13
            Item(Global): Usage Page, data= [ 0x09 ] 9
                            Buttons
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            Button 1 (Primary)
            Item(Local ): Usage Maximum, data= [ 0x0d ] 13
[... 164 lines cut ...]
      ** UNRECOGNIZED:  10 21 10 01 01 24 81 14 03 00 03 13 02 00 03 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
Device Status:     0x0012
  (Bus Powered)
  Remote Wakeup Enabled
```

Oblomov, 2018-12-01:
Further information: if I passthrough the gamepad in xbox mode to a VM with Windows, Windows manages to properly initialize the device, which then works under Windows, and after I close the VM, I can use the gamepad under Linux with xpad, so the only thing missing is something in the initialization. I am attaching a capture of the initialization process (Wireshark + usbmon).

[win10-gamesir-x360-mode-init.zip](https://github.com/paroj/xpad/files/2636246/win10-gamesir-x360-mode-init.zip)

Oblomov, 2018-12-03:
Oh, this looks promising, thank you very much, I'll give it a test as soon as I get home.

(FWIW, the gist with the script doesn't seem to open correctly for me, I had to download the link separately and then open it in the browser).

dnmodder, 2018-12-03:
I did not place the links correctly :laughing:

Oblomov, 2018-12-03:
@dnmodder the control message you identified seems to work. I have not tested your script though: I have patched xpad to emit that control message as well as the two subsequent ones that I found in the wireshark output of the initialization of my contoller. It also takes three attempts to finally start responding, failing still with the same -2 error, but after that it's fully functional. I'll prepare an RFC merge.

BelT327l, 2019-03-26:
Is it possible, to make theese thing more precise?
Compatibility Mode (xbx360) is not very accurate.
PS4 Gamepad is MUCH MUCH MUCH more precise.

paroj, 2021-08-22:
duplicate of #119

closed 2021-08-22 by paroj

mrfixit2001, 2021-09-02:
The changes I've made might work for this.
https://github.com/mrfixit2001/rockchip-kernel/blob/release-4.4/drivers/input/joystick/xpad.c

Might also need to add another call to xpad_is_mfg for manufacturer Xiaoji, but I don't have this to test with so can't confirm.