paroj/xpad #122: Pull upstream xpad PDP quirk change

cgutman, 2018-12-15:
Please rebase to get my PDP patch upstream - https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/input/joystick/xpad.c?id=a6754fae1e66e9a40fed406290d7ca3f2b4d227c

This should address these PDP PRs and issues:
https://github.com/paroj/xpad/pull/104
https://github.com/paroj/xpad/pull/105
https://github.com/paroj/xpad/pull/108
https://github.com/paroj/xpad/pull/109
https://github.com/paroj/xpad/pull/112
https://github.com/paroj/xpad/pull/115
https://github.com/paroj/xpad/pull/116
https://github.com/paroj/xpad/issues/117

paroj, 2018-12-15:
done

closed 2018-12-15 by paroj

paroj, 2018-12-15:
but still need to upstream this, no?
https://github.com/paroj/xpad/commit/a66c53c008572db516a60a8d38ef4f823b3d3980

ewtoombs, 2019-04-11:
@cgutman: I tried the driver without my patch and it didn't work. That was a while ago, though, and I don't really have a way of figuring out what version it was I tested that failed. What version should I be trying?

ewtoombs, 2019-12-07:
@cgutman I just confirmed, it definitely will not work without the PR I just put through. So yeah, accept it please.