paroj/xpad #133: Controller "Spirit of Gamer PGX-WIRED Pro Gaming XBOXONE controller SDG-WXB1" does not work

LagodViads, 2019-06-02:
Hi and thanks for your work.

This controller, marketed as "xbox one" controller does not work with latest xpad git version.

While it seems correctly identified, it is currently unusable as none of the buttons and joysticks do anything at all and the leds keep blinking (while it promptly stops blinking on windows).

lsusb output :

`ID 045e:02d1 Microsoft Corp. Xbox One Controller`

dmesg output
```
[109057.297555] usb 3-5: new full-speed USB device number 15 using xhci_hcd
[109057.450946] usb 3-5: New USB device found, idVendor=045e, idProduct=02d1, bcdDevice= 1.01
[109057.450950] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[109057.450952] usb 3-5: Product: Controller
[109057.450954] usb 3-5: Manufacturer: ShanWan    
[109057.450956] usb 3-5: SerialNumber: 00008EED983A㐷㔵㘸㐴
```

pad plugin trace (by adding debug messages)
```
[109057.452401] **xpad_probe
[109057.452406] **xpad_init_output
[109057.452407] **xpad_init_input
[109057.452411] **xpad_set_up_abs
[109057.452412] **xpad_set_up_abs
[109057.452413] **xpad_set_up_abs
[109057.452413] **xpad_set_up_abs
[109057.452414] **xpad_set_up_abs
[109057.452415] **xpad_set_up_abs
[109057.452416] **xpad_set_up_abs
[109057.452417] **xpad_set_up_abs
[109057.452418] **xpad_init_ff
[109057.452420] **xpad_led_probe
[109057.452495] input: Microsoft X-Box One pad  as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/input/input28
[109057.467724] **xpad_open
[109057.467725] **xpad_start_intput
[109057.467730] **xpad_start_xbox_one
[109057.467730] **xpad_try_sending_next_out_packet
[109057.467731] **xpad_prepare_next_out_packet
[109057.467731] **xpad_prepare_next_init_packet
[109057.472238] **xpad_irq_in
[109057.472239] **xpad_irq_in2
[109057.472240] xpad-dbg: 00000000: 02 20 01 1c 7e ed 34 35 36 34 00 00 c6 24 2a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[109057.472241] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[109057.472241] **xpadone_presence_work
[109057.472242] **xpad_irq_in3
[109057.472242] **xpad_irq_in4
[109057.472246] **xpad_irq_out
[109057.472247] **xpad_prepare_next_out_packet
[109057.472247] **xpad_prepare_next_init_packet
[109057.480238] **xpad_irq_in
[109057.480239] **xpad_irq_in2
[109057.480241] xpad-dbg: 00000000: 02 20 01 1c 7e ed 34 35 36 34 00 00 c6 24 2a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[109057.480242] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[109057.480243] **xpadone_presence_work
[109057.480243] **xpad_irq_in3
[109057.480243] **xpad_irq_in4
[109057.486173] **xpad_close
[109057.486174] **xpad_stop_intput
[109057.486192] **xpad_irq_in
[109057.506933] **xpad_open
[109057.506934] **xpad_start_intput
[109057.506938] **xpad_start_xbox_one
[109057.506939] **xpad_try_sending_next_out_packet
[109057.506939] **xpad_prepare_next_out_packet
[109057.506940] **xpad_prepare_next_init_packet
[109057.512243] **xpad_irq_in
[109057.512243] **xpad_irq_in2
[109057.512245] xpad-dbg: 00000000: 03 20 01 04 80 00 00 00 36 34 00 00 c6 24 2a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[109057.512246] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[109057.512247] **xpadone_presence_work
[109057.512247] **xpad_irq_in3
[109057.512247] **xpad_irq_in4
[109057.512250] **xpad_irq_out
[109057.512251] **xpad_prepare_next_out_packet
[109057.512251] **xpad_prepare_next_init_packet
[109057.520245] **xpad_irq_in
[109057.520246] **xpad_irq_in2
[109057.520249] xpad-dbg: 00000000: 03 20 01 04 80 00 00 00 36 34 00 00 c6 24 2a 54 01 00 01 00 17 01 02 00 01 00 01 00 01 00 01 00
[109057.520250] xpad-dbg: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[109057.520251] **xpadone_presence_work
[109057.520251] **xpad_irq_in3
[109057.520251] **xpad_irq_in4
[109057.521738] **xpad_close
[109057.521739] **xpad_stop_intput
```

I can send a windows connection/disconnection wireshark trace if needed.

Thanks.

meteos77, 2019-10-11:
Where are you on the recognition of this controller?
I am not a specialist and I start with controller management
I'm testing with tuxracer

I added a few lines to xpad.c for limited functionality 
but some buttons are recognized correctly and others not...

on line 142
{ 0x045e, 0x02d1, "Microsoft X-Box One pad", 0, XTYPE_XBOXONE },

on line 530
/*
    531 * A specific rumble packet is required for Spirit Of Gamer
    532 * sending input reports. One of those pads is (0x045e:0x02d1).
    533 */
    534 static const u8 xboxone_sogpgxone_init[] = {
    535 0x09, 0x00, 0x00, 0x00, 0x09, 0x00, 0x0F, 0x00, 0x00,
    536 0x1D, 0x1D, 0x1D, 0xFF, 0x00, 0x00
    537 };

line 560
XBOXONE_INIT_PKT(0x045e, 0x02d1, xboxone_sogpgxone_init),

the init sequence is just taken from another controller so maybe find the meaning and modify it accordingly for better support.

my little contribution
translated with deepl (French to English)
Sincerely

Translated with www.DeepL.com/Translator

gnumoksha, 2020-07-11:
I have a gamepad identified with the same id 045e:02d1 and I'm convinced that the issue is related to #119

meteos77, 2020-07-11:
I was not able to modify the driver by fixcontroller.py
(no python knowledge) if someone knows how to send the init sequence through the python file?

I tried my old config but I had to change the xpad..c file to line 545 by changing the init sequence 

static const u8 xboxone_sogpgxone_init[] = {
        0x09, 0x02, 0x20, 0x00, 0x01, 0x00, 0xa0,
        0x1D, 0x1D, 0xFF, 0x00, 0x00
};
I've got almost all the buttons except the turbo and the led next to the start button which stays on. (tested with jtest-gtk)

my config => linuxmint 20 - kernel 5.4.0-40-generic

Translated with www.DeepL.com/Translator (free version)

gnumoksha, 2020-07-11:
@meteos77 I applied your two modifications in both xpad.c in this repository and the last [xpad.c from the kernel](https://github.com/torvalds/linux/blob/master/drivers/input/joystick/xpad.c) but the gamepad still don't work (leds blinking and no input events). How did you get those initialization values? I've intercepted the USB communication on Windows but I'm having trouble identifying the init sequence.

meteos77, 2020-07-12:
Sorry, I had tried like you to intercept the USB link on Windows without more success.
I don't know how to translate the windows init sequence to the xpad.c file.

if a pro could explain to us

so I just picked up an init sequence from another controller.  for my controller it works
are we talking about the same controller ?
Manufacturer : ShanWan  - Spirit Of Game WXB1

there may be different models in the same brand with the dmesg there's a serial number.

I'll give you my exit with the dmesg command:

 1531.044596] usb 2-1.7: USB disconnect, device number 4
[ 1533.315634] usb 2-1.7: new full-speed USB device number 5 using ehci-pci
[ 1533.429957] usb 2-1.7: New USB device found, idVendor=045e, idProduct=02d1, bcdDevice= 1.01
[ 1533.429960] usb 2-1.7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1533.429962] usb 2-1.7: Product: Controller
[ 1533.429964] usb 2-1.7: Manufacturer: ShanWan    
[ 1533.429965] usb 2-1.7: SerialNumber: 00008EED983A㐷㔵㘸㐴
[ 1533.430587] input: Microsoft X-Box One pad as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.7/2-1.7:1.0/input/input19

meteos77, 2020-08-03:
at each kernel change, I have to redo the installation of xpad.ko for the right kernel version.

how to transform the lines of xpad.c with the python script: fixcontroler.py

I can't transform this sequence of xpad.c
static const u8 xboxone_sogpgxone_init[] = {
        0x09, 0x02, 0x20, 0x00, 0x01, 0x00, 0xa0,
        0x1D, 0x1D, 0xFF, 0x00, 0x00

in python command dev.ctrl_transfer
if d.idVendor == 0x045e and d.idProduct == 0x02d1 :
       dev.ctrl_transfer(0x21, 0x09, 0x02, 0x20 (0x00, 0x01, 0x00, 0xa0, 0x1D, 0x1D, 0xFF, 0x00, 0x00))

Can you help me? Thank you.

Translated with www.DeepL.com/Translator (free version)

Carusu, 2020-08-17:
I think I have a similar controller (just a re-skin of yours) and am having the same issue. FWIW `sdl2-jstest -r 0` makes the controller vibrate and gives the following output: `error: failed to read gamecontrollerdb.txt: Invalid RWops`

Carusu, 2020-08-17:
> I was not able to modify the driver by fixcontroller.py
> (no python knowledge) if someone knows how to send the init sequence through the python file?
>
> I tried my old config but I had to change the [...]

Could you please share your xpad.c?
Many thanks

meteos77, 2020-08-17:
how to share the file?

Carusu, 2020-08-17:
you can attach them here, just drag them in the text box

meteos77, 2020-08-17:
remove extension .ok.txt
[xpad.c.ok.txt](https://github.com/paroj/xpad/files/5082670/xpad.c.ok.txt)

Carusu, 2020-08-17:
Oh man, this works wonder, you should submit your patch and hope that the developers have a look at it

meteos77, 2020-08-17:
with my version I think the modification causes problems on the original xbox controller versions because it's the same 0x045e, 0x02d1 identifier. Shanwan took 0x045e which is microsoft.
I run a special init sequence for the Shanwan which may not work for the original xbox controller. (not testing)

small details :
1 - if the version of the kernel changes, force to rebuild.
 hence my question about the python script.

2 - on line 142: { 0x045e, 0x02d1, "Microsoft X-Box One pad", 0, XTYPE_XBOXONE },
if you change the 0 to 1, or 2 or 3, some buttons will not be seen the same way.

Translated with www.DeepL.com/Translator (free version)

Carusu, 2020-08-17:
> with my version I think the modification causes problems on the original xbox controller versions because it's the same 0x045e, 0x02d1 identifier. Shanwan took 0x045e which is microsoft.
> I run a [...]

I see, it is indeed a tricky problem as the identifier is the same. I also do not have an original controller to test.
For 1) you could look into dkms, which would automatically compile the module at each at each kernel update

albertxiao, 2021-03-06:
I have same problem, controller detected with same vendor and device id, but able to get all keys works except the led between start and select keep blinking. the controller itself is fake xbox one controller, the text in sticker is Wired Controller for XB-ONE Model_No: HSY-008. this is modification done in xpad.c

Add 1 line in line in xpad_device[]
`{ 0x045e, 0x02d1, "Microsoft Corp. Xbox One Controller", 0, XTYPE_XBOXONE },`

Add 2 line in xboxone_init_packet
`
	XBOXONE_INIT_PKT(0x045e, 0x02d1, xboxone_rumblebegin_init),
	XBOXONE_INIT_PKT(0x045e, 0x02d1, xboxone_rumbleend_init),`

meteos77, 2021-09-26:
Hi, thanks for the developments

I have a controller that is recognized as a Microsoft Xbox One Controller 
Vendor=045e ProdID=02d1 Rev=01.01

but the init sequence is not the right one

Manufacturer= ShanWan
SerialNumber=00008EED983A㐷㔵㘸㐴

I can make it work by fiddling (modifying the init code) but I would like it to be recognized without these modifications

can you add a test on the "Manufacturer" field and connect it to the right init sequence

vendor=045e + ProdID=02d1 + Manufacturer=Microsoft to init sequence Microsoft_init

vendor=045e + ProdID=02d1 + Manufacturer=ShanWan to sequence init ShanWan_init

I have the impression that with the addition of Oblomov and paroj
commit 1eab9e91a898f6346b9c8c360113a11db119998e
it should work with xbox_360 controllers but not for xbox_one

I am not competent to program these modifications

if a developper feels the courage in advance thank you

Translated with www.DeepL.com/Translator (free version)