paroj/xpad #135: Move to mainline kernel ?

tomleb, 2019-08-12:
Hi,

just bought a controller and I was surprised that it did not work out of the box with the kernel's xpad module. It works fine with your module though.

I was wondering, are there any reason to not push this upstream to add it to the mainline kernel ?

I would be interested in helping doing that. From ./scripts/get_maintainer.pl, it seems like the person to contact is Dmitry Torokhov <dmitry.torokhov@gmail.com>.

paroj, 2021-01-29:
see https://github.com/paroj/xpad/issues/159#issuecomment-756734217

actually this is synced about once a year with mainline. Its just a PITA to push patches through the mailing list workflow used by the kernel.

closed 2021-01-29 by paroj

vinicius-daros, 2025-01-27:
> see [#159 (comment)](https://github.com/paroj/xpad/issues/159#issuecomment-756734217)
> 
> actually this is synced about once a year with mainline. Its just a PITA to push patches through the mailing list workflow used by the kernel.

Would be possible for someone to upstream it for you?