paroj/xpad #141: Usb PS3 clone is working with annoying messages in the log (same as #114)

mvvvv, 2020-01-16:
Hi
My PS3 clone works well but I have this message every 5 seconds, with an increasing device number :+1: 
kernel: usb 1-4: USB disconnect, device number 17
kernel: xpad 1-4:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
kernel: usb 1-4: new full-speed USB device number 18 using xhci_hcd
kernel: usb 1-4: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.10
kernel: usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
kernel: usb 1-4: Product: XBOX 360 For Windows
kernel: usb 1-4: Manufacturer: CHIPUNION Co.,Ltd
kernel: input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/input/input28
mtp-probe[4595]: checking bus 1, device 18: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4"
mtp-probe[4595]: bus: 1, device: 18 was not an MTP device

I'm using Ubuntu 19.10 with an Intel i5 6400.

mvvvv, 2020-01-16:
When I'm connected to supertuxkart this message do not appear anymore until I leave the game.

Sometime jstest-gtk do not see the gamepad and I have to relaunch it.

jschwartzenberg, 2026-08-16:
Does your controller work at all with the latest driver version? It seems our controllers behave similarly, but I have to revert a change to let it work again: #370

mvvvv, 2026-08-17:
I can't tell I don't use it anymore.